RfidResearchGroup/proxmark3 #358: Linking fail on OSX

Heat-Miser, 2019-08-26:
**Describe the bug**
When compiling the client, the following error occurs

```
$>make

===================================================================
Platform name:     Proxmark3 Generic target
PLATFORM:          PM3OTHER
Platform extras:   No extra selected
Included options:  LF HITAG ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF
Standalone mode:   LF_SAMYRUN
===================================================================
[*] MAKE client/all
[=] LD proxmark3
Undefined symbols for architecture x86_64:
  "_rl", referenced from:
      _hash1 in elite_crack.o
      _rk in elite_crack.o
  "_rr", referenced from:
      _hash1 in elite_crack.o
  "_swap", referenced from:
      _hash1 in elite_crack.o
     (maybe you meant: _mbedtls_mpi_safe_cond_swap, _wiegand_add_parity_swapped , _mbedtls_mpi_swap )
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [proxmark3] Error 1
make: *** [client/all] Error 2```

I triple checked but it looks like it's different from issue 357.

**To Reproduce**
Using MacOs 10.14.6 (18G87)
using this commit

```
commit 1628ac8e6217a3dcf2690e885202bba077ca7023 (HEAD -> master, origin/master, origin/HEAD)
Author: Philippe Teuwen <phil@teuwen.org>
Date:   Mon Aug 26 13:55:42 2019 +0200

    revert temporarily hf iclass chk/lookup 2a1e6dff
```

**Expected behavior**

Succesfully linking the client :)

**Desktop (please complete the following information):**
 - OS:  MacOs 10.14.6 (18G87)

Heat-Miser, 2019-08-26:
Full compilation (make clean all)

```
$> make clean all

===================================================================
Platform name:     Proxmark3 Generic target
PLATFORM:          PM3OTHER
Platform extras:   No extra selected
Included options:  LF HITAG ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF
Standalone mode:   LF_SAMYRUN
===================================================================
[*] MAKE client/clean
[*] MAKE bootrom/clean
[*] MAKE fpga_compress/clean
[*] MAKE armsrc/clean
[*] MAKE recovery/clean
[*] MAKE mfkey/clean
[*] MAKE nonce2key/clean
[*] MAKE client/all
[-] CC proxmark3.c
[-] CC uart/uart_posix.c
[-] CC uart/uart_win32.c
[-] CC ui.c
[-] CC ../common/commonutil.c
[-] CC util.c
[-] CC ../common/util_posix.c
[-] CC scandir.c
[-] CC ../common/crc16.c
[-] CC comms.c
[-] CC ../common/crapto1/crapto1.c
[-] CC ../common/crapto1/crypto1.c
[-] CC mifare/mfkey.c
[-] CC tea.c
[-] CC fido/additional_ca.c
[-] CC fido/cose.c
[-] CC fido/cbortools.c
[-] CC fido/fidocore.c
[-] CC crypto/asn1dump.c
[-] CC crypto/libpcrypto.c
[-] CC crypto/asn1utils.c
[-] CC cliparser/argtable3.c
[-] CC cliparser/cliparser.c
[-] CC loclass/cipher.c
[-] CC loclass/cipherutils.c
[-] CC loclass/ikeys.c
[-] CC loclass/elite_crack.c
[-] CC fileutils.c
[-] CC whereami.c
[-] CC mifare/mifarehost.c
[-] CC ../common/parity.c
[-] CC ../common/crc.c
[-] CC ../common/crc64.c
[-] CC ../common/legic_prng.c
[-] CC ../common/iso15693tools.c
[-] CC prng.c
[-] CC graph.c
[-] CC cmddata.c
[-] CC ../common/lfdemod.c
[-] CC emv/crypto_polarssl.c
[-] CC emv/crypto.c
[-] CC emv/emv_pk.c
[-] CC emv/emv_pki.c
[-] CC emv/emv_pki_priv.c
[-] CC emv/test/cryptotest.c
[-] CC emv/apduinfo.c
[-] CC emv/dump.c
[-] CC emv/tlv.c
[-] CC emv/emv_tags.c
[-] CC emv/dol.c
[... 206 lines cut ...]
[-] CC deflate.c
[-] CC adler32.c
[-] CC trees.c
[-] CC zutil.c
[-] CC inflate.c
[-] CC inffast.c
[-] CC inftrees.c
[=] AR libz.a
[=] GEN lualibs/pm3_cmd.lua
[=] GEN lualibs/mfc_default_keys.lua
[=] LD proxmark3
Undefined symbols for architecture x86_64:
  "_rl", referenced from:
      _hash1 in elite_crack.o
      _rk in elite_crack.o
  "_rr", referenced from:
      _hash1 in elite_crack.o
  "_swap", referenced from:
      _hash1 in elite_crack.o
     (maybe you meant: _mbedtls_mpi_safe_cond_swap, _wiegand_add_parity_swapped , _mbedtls_mpi_swap )
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [proxmark3] Error 1
make: *** [client/all] Error 2
```

Heat-Miser, 2019-08-26:
Any idea ? I'm probably doing something stupid... 

Thanks for your help.

iceman1001, 2019-08-26:
which compiler?

doegox, 2019-08-26:
can you show with make V=1 ?
how did you setup the environment? cf https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Mac-OS-X-Homebrew-Installation-Instructions.md#homebrew-mac-os-x-developer-installation

Heat-Miser, 2019-08-26:
Wokay forget it, the build failed due to a LDflag I used in my env... Sorry for the noise, I must have checked it before !

closed 2019-08-26 by Heat-Miser

doegox, 2019-08-26:
OSX ppl have hard time with Mondays apparently... ;)

Heat-Miser, 2019-08-26:
yeah... harsh monday... n33d moar C0FFEE !

mikemykhaylov, 2023-12-15:
I'm still having this issue on the latest commit, what exactly did you change to resolve the issue?