RfidResearchGroup/proxmark3 #365: Android app have a bug.

xianglin1998, 2019-08-28:
How can we solve the problem that the CommandReceived () function executes many times and causes multiple tasks to execute at the same time so that timeout occurs?

ps: some task i can't get executing status, such as lf t55xx dump, some task app thinks it is finished,  but not, it's still running .     :(

Wishing you a pleasant job ：）

iceman1001, 2019-09-16:
It depends how you solved sending commands to the proxmark client in Android.
All commands should have an exit status code to look for.
Second problem is simulation or device operations that runs but leaves the client back to normal.  You press button on device to break the operation.   Client still can do things meanwhile.  

You need to identify all commands that you have an issue with on Android.

iceman1001, 2020-04-25:
Closed because of inactivity from original poster

closed 2020-04-25 by iceman1001