RfidResearchGroup/proxmark3 #367: autopwn vs. hardnested:

doegox, 2019-08-28:
There is an issue with autopwn when hardnested is on a bad slope, no way to interrupt it and restart without losing what's already found.
E.g. on one run, when reaching the 10th sector my hardnested foresees 17min of BF while it was way faster on other sectors and probably on that sector if aborted and restarted as well.
It was not much an issue when hardnested could be restarted manually but now that it's automated we should be more cautious to handle these ones in a better way, e.g. with an increasing limit for the foreseen bf time.
E.g.:
* limit: 1 min, start hardnested
* when reaching BF phase, proceed only if BT foreseen time < 1 min
* else restart hardnested with limit * 2

iceman1001, 2019-08-28:
You can force hardnested to collect more nonces, instead of going for BF...  
I adapted it to go for BF earlier...  but on hard keys,  that might not be the optimal solution

doegox, 2019-08-28:
I'm not sure more nonce would help, in this case it was clear since the beginning that it was going large BF and collecting more nonces didn't improve much the situation:
```
       0 |       0 | Start using 8 threads and AVX2 SIMD core                |                 |          
       0 |       0 | Brute force benchmark: 512 million (2^28.9) keys/s      | 140737488355328 |    3d          
       1 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |    3d          
       4 |     112 | Apply bit flip properties                               |   3192336416768 |    2h          
       5 |     224 | Apply bit flip properties                               |   2146056667136 | 70min          
       6 |     335 | Apply bit flip properties                               |   2135126573056 | 70min          
       7 |     447 | Apply bit flip properties                               |   2127205761024 | 69min          
       8 |     559 | Apply bit flip properties                               |   2127205761024 | 69min          
       9 |     670 | Apply bit flip properties                               |   2127205761024 | 69min          
      10 |     779 | Apply bit flip properties                               |   2127205761024 | 69min          
      10 |     888 | Apply bit flip properties                               |   2127205761024 | 69min          
      11 |     996 | Apply bit flip properties                               |   2127205761024 | 69min          
      12 |    1108 | Apply bit flip properties                               |   2127205761024 | 69min          
      13 |    1216 | Apply bit flip properties                               |   2127205761024 | 69min          
      13 |    1327 | Apply bit flip properties                               |   2127205761024 | 69min          
      14 |    1439 | Apply bit flip properties                               |   2127205761024 | 69min          
      15 |    1550 | Apply bit flip properties                               |   2127205761024 | 69min          
      16 |    1659 | Apply bit flip properties                               |   2127205761024 | 69min          
      17 |    1767 | Apply bit flip properties                               |   2127205761024 | 69min          
      20 |    1874 | Apply Sum property. Sum(a0) = 128                       |    539105394688 | 18min          
      20 |    1984 | Apply bit flip properties                               |    515078258688 | 17min          
      21 |    2094 | Apply bit flip properties                               |    526535983104 | 17min          
      22 |    2205 | Apply bit flip properties                               |    526535983104 | 17min          
      23 |    2315 | Apply bit flip properties                               |    526535983104 | 17min          
      23 |    2421 | Apply bit flip properties                               |    526535983104 | 17min          
      24 |    2421 | (Ignoring Sum(a8) properties)                           |    526535983104 | 17min          
```

iceman1001, 2019-08-28:
it should try gather even more nonces.
Interesting card.  First time I see 2400+ nonces for the improved hardnested.  But in general,  still,  collecting more nonces helps to deduct keyspace. Here you see it tries to BF even it its 17min

doegox, 2019-08-28:
yeay, repeating the attack always leads to predicted large times on that specific key (but still BF phase significantly faster than predicted).
so more nonces is probably the way to go.
I'm surprised there is no parameter for that in hardnested usage

doegox, 2019-08-28:
e.g.
```
      18 |    1550 | Apply Sum property. Sum(a0) = 128                       |    120793292800 |  3min          
      18 |    1659 | Apply bit flip properties                               |    204949192704 |  5min          
      19 |    1768 | Apply bit flip properties                               |    204949192704 |  5min          
      20 |    1878 | Apply bit flip properties                               |    204949192704 |  5min          
      20 |    1988 | Apply bit flip properties                               |    204949192704 |  5min          
      21 |    1988 | (1. guess: Sum(a8) = 256)                               |    204949192704 |  5min          
      21 |    1988 | Apply Sum(a8) and all bytes bitflip properties          |    204949192704 |  5min          
      21 |    1988 | (2. guess: Sum(a8) = 192)                               |    859048968192 | 22min          
      22 |    1988 | Apply Sum(a8) and all bytes bitflip properties          |    845032849408 | 21min          
      22 |    1988 | (3. guess: Sum(a8) = 224)                               |   1360115007488 | 34min          
      22 |    1988 | Apply Sum(a8) and all bytes bitflip properties          |   1348240146432 | 34min          
      22 |    1988 | (4. guess: Sum(a8) = 160)                               |   1979259682816 | 50min          
      26 |    1988 | Apply Sum(a8) and all bytes bitflip properties          |   1668455596032 | 42min          
      30 |    1988 | Brute force phase completed. Key found: 45d388cdba81    |               0 |    0s          
```
while the BF took only seconds, not 42min!

iceman1001, 2019-08-28:
the estimation is how long the whole leftover keyspace will take.  You might find the key in beginning...

closed 2019-08-29 by doegox