RfidResearchGroup/proxmark3 #368: cmdhw.c:629:9: error: ‘__builtin_memcpy’ writing 47 bytes into a region of size 37 overflows the destination [-Werror=stringop-overflow=]

strasharo, 2019-08-28:
**Describe the bug**
Trying to build the latest HEAD on Fedora 30 with PM3OTHER platform fails with the following exception:
```
[-] CC cmdlfcotag.c
cmdhw.c: In function ‘pm3_version’:
cmdhw.c:629:9: error: ‘__builtin_memcpy’ writing 47 bytes into a region of size 37 overflows the destination [-Werror=stringop-overflow=]
  629 |         strcat(msg + strlen(msg), _YELLOW_("GCC "__VERSION__));
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[1]: *** [Makefile:418: obj/cmdhw.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:55: client/all] Error 2
```
**To Reproduce**
Steps to reproduce the behavior on Fedora 30 x86_64:
1. git clone https://github.com/RfidResearchGroup/proxmark3.git
2. cd proxmark3
3. cp Makefile.platform.sample Makefile.platform
4. edit Makefile.platform and set PLATFORM=PM3OTHER
5. make clean
6. make -j8
7. Get the folowing exception:
```
[-] CC cmdlfcotag.c
cmdhw.c: In function ‘pm3_version’:
cmdhw.c:629:9: error: ‘__builtin_memcpy’ writing 47 bytes into a region of size 37 overflows the destination [-Werror=stringop-overflow=]
  629 |         strcat(msg + strlen(msg), _YELLOW_("GCC "__VERSION__));
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[1]: *** [Makefile:418: obj/cmdhw.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:55: client/all] Error 2

```

**Expected behavior**
It should be as it did before.

**Desktop (please complete the following information):**
 - OS: Fedora 30 x86_64

**Additional context**
gcc (GCC) 9.1.1 20190503 (Red Hat 9.1.1-1)
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

doegox, 2019-08-28:
Thanks, should be fixed by https://github.com/RfidResearchGroup/proxmark3/commit/aa77f7298c18c22dda15673d518093b6f0f0e55e

strasharo, 2019-08-28:
Thanks for the fast response, retried with the latest HEAD and got this:
```

[-] CC cmdhw.c
cmdhw.c: In function ‘pm3_version’:
cmdhw.c:633:9: error: ‘__builtin_memcpy’ writing 47 bytes into a region of size 39 overflows the destination [-Werror=stringop-overflow=]
  633 |         strcat(msg + strlen(msg), _YELLOW_(PM3CLIENTCOMPILER __VERSION__));
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[1]: *** [Makefile:418: obj/cmdhw.o] Error 1
make: *** [Makefile:55: client/all] Error 2

```

doegox, 2019-08-28:
sigh... :) let me check

doegox, 2019-08-28:
try again, I simplified the code

strasharo, 2019-08-28:
Thanks a lot, builds and flashes fine now. :)

iceman1001, 2019-08-29:
Excellent @doegox 
I think the ascii colors might have made the string longer.  We calculated string length before adding it.

strasharo, 2019-08-29:
@iceman1001 , with the latest commits now I get this:

```
[-] CC cmdhw.c
cmdhw.c: In function ‘pm3_version’:
cmdhw.c:676:66: error: expected ‘)’ before ‘PM3HOSTOS’
  676 |         PrintAndLogEx(NORMAL, "Client: RRG/Iceman compiled with " _YELLOW_(PM3CLIENTCOMPILER __VERSION__) " on " _YELLOW_(PM3HOSTOS));
      |                                                                  ^
      |                                                                  )
cmdhw.c:692:49: error: expected ‘)’ before ‘PM3HOSTOS’
  692 |         PrintAndLogEx(NORMAL, "  compiled with " PM3CLIENTCOMPILER __VERSION__ " on " PM3HOSTOS);
      |                                                 ^                                     ~~~~~~~~~
      |                                                 )
make[1]: *** [Makefile:418: obj/cmdhw.o] Error 1
make: *** [Makefile:55: client/all] Error 2

```

iceman1001, 2019-08-29:
ha,  on mingw (ps32) it compiles,  switched over to wsl,  and it fails.

iceman1001, 2019-08-29:
try again

strasharo, 2019-08-29:
Compiles fine now, thanks a lot. :)

closed 2019-08-29 by strasharo