RfidResearchGroup/proxmark3 #375: HELP_EXAMPLE_STANDALONE causes confusing output

ZeroChaos-, 2019-08-29:
The HELP_EXAMPLE bits are a bit mismatched:
```
armsrc/Standalone/Makefile.hal:HELP_EXAMPLE_STANDALONE=HF_COLIN
common_arm/Makefile.hal:make PLATFORM=PM3OTHER STANDALONE=$(HELP_EXAMPLE_STANDALONE)
```
Producing this:
```
zero@theprophet proxmark3 % make PLATFORM=                                                                                                                                    (git)-[master] 
common_arm/Makefile.hal:64: *** Invalid or empty PLATFORM: . 
Known definitions:

+============================================+
| PLATFORM        | DESCRIPTION              |
+============================================+
| PM3RDV4 (def)   | Proxmark3 rdv4           |
...
snip
...
Options to define platform, platform extras and/or standalone mode:
(1) Run make with your PLATFORM, PLATFORM_EXTRAS and/or STANDALONE choices as follows:
make PLATFORM=PM3OTHER STANDALONE=HF_COLIN
...
snip
```

ZeroChaos-, 2019-08-29:
I'd suggest changing HELP_EXAMPLE_STANDALONE to something that works for PM3OTHER just to avoid the confusion of the sample not working

iceman1001, 2019-08-29:
Good thing the makefile stopped :)  
Fixed it

closed 2019-08-29 by iceman1001

doegox, 2019-08-29:
good catch!