RfidResearchGroup/proxmark3 #377: Making T5577 Downlink mode more transparent

mwalker33, 2019-08-29:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Not a problem.
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
I will do the work (see below for query)
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
Seeking feedback.
**Additional context**
Add any other context or screenshots about the feature request here.

Now that the current downlink mode support (r <mode>) has had time to settle in, I am working on making this a bit more transparent.
Short version : the lf t55 detect will add the downlink mode to the "config" detected.  Then as needed the default will use the downlink mode in the config (i.e. use would note need to supply the r <mode> unless the wish to override).
The lf t55 detect will, by default try all modes until one is found (or all have been tried)

So, the new behavior will be:
lf t55 detect
(config and dl-mode detected)
lf t55 read ... (the downlink mode found in the detect will be used unless r option supplied)
lf t55 write ... (the downlink mode found in the detect will be used unless r option supplied)

So, the result should be, unless you need to do something more advanced it should be automatic and transparent.

While i was looking at the above, I had an idea,  this is what I am seeking feed back on.
Since we should/need to run an lf t55 detect prior to working on a t5577 card AND whenever we make a change to the config, I thought why don't I add in an auto lf t55 detect IF we see a write to block 0 page 0 (normal config) OR block 3 page 1 (where we have extend config like the downlink modes)

This should have the result of:
lf t55 detect
lf t55 write b 0 d 12345678 
.... auto new detect to load the new config values ....
lf t55 read b 1  <- this should just work as we already know the new config from the auto detect.

My question:  Does this sound like a good idea and can anyone see any issues if we do an auto detect after a write to b 0/0 or b3/1
(or anything else I may not have considered that may cause an issue)

Thanks

iceman1001, 2019-08-30:
Yes and no.
I like to see those commands as building blocks.
We could have built in a call to `lf t55xx detect` long time ago.   We opted not to because its quite slow and doesn't have 100% success rate.

Since it is a guessing game with t55x7 cards,  its better to have these command separated.

We can however print a message when we see that a user wrote to block 0,  that they now must run detect again.

mwalker33, 2019-08-30:
@iceman no problem that's why we ask :)
when you get 5, can you have a look at the email I sent to ensure we are on the same page for the rest.
Thanks.

iceman1001, 2019-08-30:
Remember we also have T5555 / termic ,  and all T55x7 (57,67,77) cards that isn't revision C  to make things work with.

That was a long email ;)

mwalker33, 2019-08-30:
It should all work as this is just a cleaner way.  All the heavy lifting code will not change. Eg. If the card does not support the downlink mode then the first check will work as it does now, so all good.

iceman1001, 2019-08-30:
We had an other idea.

_See if data read from block0 matches the auto config detection_

Meaning we do detect and then try reading block0 and see if they match up.
This would ofcourse be a new command.   lf t55xx validate?    Are you up to implement it?

mwalker33, 2019-08-30:
Should be able to work on it.  With not too much thought, the simplest would be save the "demodulate 32 bits" used for the config from the detect (maybe in the config struct), then when we read b 0 we just need to compare 2 uint32_t values.

I was thinking of adding something to set the trace data (to a fixed value).  e.g. If you have erased or over written the trace data (b 1 and 2 page 1) then lf t55 trace or the p1 detect wont work. 
e.g. 
lf t55xx write b 1 1 d E0150A48 
lf t55xx write b 2 1 d 2D782308 

As a side note, something I thought was a nice result of my new round of changes was the lf t55 wipe now wipes page 0 and leaves the d/l mode as set (as the wipe does not write to page 1).
Prior you would have to reset the d/l mode first (back to the default) then use the wipe - now transparent and leaving the d/l mode as is.

iceman1001, 2019-08-30:
Yes,   almost,  our config isn't the same as a complete block0.  So its more bit compares.

all data on a t55xx card should be included in the dump file.
and the dump file should support json/eml/bin...  
That way you never loose data

mwalker33, 2019-08-30:
mmm I think I am missing something here.
Where does it write to a file ? or is that what you would like me to add as well ?

The current lf t55 detect already saves block 0 (so unless I am missing something), it should just be compare that to a read of block 0 and check if they match. 
```
[usb] pm3 --> lf t55 det
Chip Type  : T55x7
Modulation : ASK
Bit Rate   : 2 - RF/32
Inverted   : No
Offset     : 32
Seq. Term. : Yes
Block0     : 0x000880E0
DL Mode    : Leading Zero 

[usb] pm3 --> lf t55 read b 0
Reading Page 0:
blk | hex data | binary                           | ascii
----+----------+----------------------------------+-------
 00 | 000880E0 | 00000000000010001000000011100000 | ....
```

iceman1001, 2019-08-30:
My bad,   detect has a block,  then its just comparing with a read b 0 :)

next one is
lf t55xx dump   ->  where my comments is about what more it should save and which formats.

mwalker33, 2019-08-31:
Ignoring the printf output (I will fix that), is this what you where thinking for the validate ?
Note: the first function T55xx_Validate does the read and check (no user i/o) so can be called as needed. The 2nd function is the user interface.

```
#define T55xx_Validate_Return_OK          0
#define T55xx_Validate_Return_Invalid     1
#define T55xx_Validate_Return_FailAquire  2
#define T55xx_Validate_Return_FailDecode  3
#define T55xx_Validate_Return_FailGetData 4

uint8_t T55xx_Validate (uint32_t *block0_data, bool usepwd, uint32_t password, uint8_t downlink_mode) {

    if (!AquireData(T55x7_PAGE0, T55x7_CONFIGURATION_BLOCK, usepwd, password, downlink_mode)) 
        return T55xx_Validate_Return_FailAquire;

    if (!DecodeT55xxBlock()) {
        return T55xx_Validate_Return_FailDecode;
    }

    if (GetT55xxBlockData(block0_data) == false)
        return T55xx_Validate_Return_FailGetData;

    if (*block0_data == config.block0)
        return T55xx_Validate_Return_OK;
    else
        return T55xx_Validate_Return_Invalid;
}

static int  CmdT55xxValidate (const char *Cmd)
{
    uint8_t  validateResponse;
    bool     usepwd        = false;
    bool     override      = false;
    uint32_t password      = 0;
    uint8_t  downlink_mode = config.downlink_mode;
    uint32_t blockData     = 0;
    bool     errors        = false;
    uint8_t  cmdp          = 0;
    
    while (param_getchar(Cmd, cmdp) != 0x00 && !errors) {
        switch (tolower(param_getchar(Cmd, cmdp))) {
            case 'h':
                return usage_t55xx_validate();
            case 'o':
                override = true;
                cmdp++;
                break;
            case 'p':
                password = param_get32ex(Cmd, cmdp + 1, 0, 16);
                usepwd = true;
                cmdp += 2;
                break;
            case 'r':
                downlink_mode = param_getchar(Cmd, cmdp + 1) - '0';
                if (downlink_mode > 3) downlink_mode = 0;
                cmdp += 2;
                break;
            default:
                PrintAndLogEx(WARNING, "Unknown parameter '%c'", param_getchar(Cmd, cmdp));
                errors = true;
                break;
        }
    }
    if (errors) return usage_t55xx_validate();
    
    if (usepwd && !override) {
        printf ("Password supplied, need override\n");
        return PM3_SUCCESS;
    }
    
    validateResponse = T55xx_Validate (&blockData,usepwd,password,downlink_mode);
    switch (validateResponse) {
        case T55xx_Validate_Return_OK : 
                printf ("Valid : Config. Block 0 : %08X = Read Block 0 : %08X \n",config.block0,blockData);
                break;
        case T55xx_Validate_Return_Invalid : 
                printf ("Invalid : Config. Block 0 : %08X <> Read Block 0 : %08X \n",config.block0,blockData);
                break;
        case T55xx_Validate_Return_FailAquire : 
                printf ("Failed to aquire data\n");
        case T55xx_Validate_Return_FailDecode : 
                printf ("Failed to decode data.  try running lf t55xx detect first\n");
                break;
        case T55xx_Validate_Return_FailGetData : 
                printf ("Failed to retreive decoded data\n");
                break;
        default:
                printf ("Unknown Error\n");
    }
    
    return PM3_SUCCESS;
}
```

mwalker33, 2019-08-31:
Sample "auto downlink mode" use - for comment

Detect the config
```
[usb] pm3 --> lf t55 det
Chip Type  : T55x7
Modulation : ASK
Bit Rate   : 5 - RF/64
Inverted   : No
Offset     : 32
Seq. Term. : Yes
Block0     : 0x00148040
DL Mode    : leading zero reference
```
Read block 6, change the data and dump all blocks (no downlink mode supplied)
```
[usb] pm3 --> lf t55 read b 6
Reading Page 0:
blk | hex data | binary                           | ascii
----+----------+----------------------------------+-------
 06 | 11223344 | 00010001001000100011001101000100 | ."3D

[usb] pm3 --> lf t55 write b 6 d aabbccdd
[=] Writing page 0  block: 06  data: 0xAABBCCDD
[usb] pm3 --> lf t55 read b 6
Reading Page 0:
blk | hex data | binary                           | ascii
----+----------+----------------------------------+-------
 06 | AABBCCDD | 10101010101110111100110011011101 | ....
[usb] pm3 --> lf t55 dump
Reading Page 0:
blk | hex data | binary                           | ascii
----+----------+----------------------------------+-------
 00 | 00148040 | 00000000000101001000000001000000 | ...@
 01 | FF8C6529 | 11111111100011000110010100101001 | ..e)
 02 | 8C94A940 | 10001100100101001010100101000000 | ...@
 03 | FFFFFFFF | 11111111111111111111111111111111 | ....
 04 | FFFFFFFF | 11111111111111111111111111111111 | ....
 05 | FFFFFFFF | 11111111111111111111111111111111 | ....
 06 | AABBCCDD | 10101010101110111100110011011101 | ....
 07 | FEEDBEEF | 11111110111011011011111011101111 | ....
Reading Page 1:
blk | hex data | binary                           | ascii
----+----------+----------------------------------+-------
 00 | 00148040 | 00000000000101001000000001000000 | ...@
 01 | E0150A48 | 11100000000101010000101001001000 | ...H
 02 | 2D782308 | 00101101011110000010001100001000 | -x#.
 03 | 90000800 | 10010000000000000000100000000000 | ....
```
Change the downlink mode to 1 of 4 and retest read, write and dump
```
[usb] pm3 --> lf t55 write b 3 1 d 90000C00
[=] Writing page 1  block: 03  data: 0x90000C00
[usb] pm3 --> lf t55 det
Chip Type  : T55x7
Modulation : ASK
Bit Rate   : 5 - RF/64
Inverted   : No
Offset     : 32
Seq. Term. : Yes
Block0     : 0x00148040
DL Mode    : 1 of 4 coding reference

[usb] pm3 --> lf t55 read b 6
Reading Page 0:
blk | hex data | binary                           | ascii
----+----------+----------------------------------+-------
 06 | AABBCCDD | 10101010101110111100110011011101 | ....
[usb] pm3 --> lf t55 write b 6 d 12345678
[=] Writing page 0  block: 06  data: 0x12345678
[usb] pm3 --> lf t55 read b 6
Reading Page 0:
blk | hex data | binary                           | ascii
----+----------+----------------------------------+-------
 06 | 12345678 | 00010010001101000101011001111000 | .4Vx
[usb] pm3 --> lf t55 dump
Reading Page 0:
blk | hex data | binary                           | ascii
----+----------+----------------------------------+-------
 00 | 00148040 | 00000000000101001000000001000000 | ...@
 01 | FF8C6529 | 11111111100011000110010100101001 | ..e)
 02 | 8C94A940 | 10001100100101001010100101000000 | ...@
 03 | FFFFFFFF | 11111111111111111111111111111111 | ....
 04 | FFFFFFFF | 11111111111111111111111111111111 | ....
 05 | FFFFFFFF | 11111111111111111111111111111111 | ....
 06 | 12345678 | 00010010001101000101011001111000 | .4Vx
 07 | FEEDBEEF | 11111110111011011011111011101111 | ....
Reading Page 1:
blk | hex data | binary                           | ascii
----+----------+----------------------------------+-------
 00 | 00148040 | 00000000000101001000000001000000 | ...@
 01 | E0150A48 | 11100000000101010000101001001000 | ...H
 02 | 2D782308 | 00101101011110000010001100001000 | -x#.
 03 | 90000C00 | 10010000000000000000110000000000 | ....
```

As can be seen we can use the different downlink modes as long as the lf t55 detect has been completed.
you can still supply the r mode if you need to overwrite for any reason.

Comments ?

iceman1001, 2019-08-31:
For future reference,  learn how to **git branch**
usually ppl make a branch and reference it,  so other ppl can download and test the code.  
That branch will be the base for your PR later on. 

Besides from introducting new styled return codes  *don't*, 
its about the right idea. I thought on having the block0, to be optional for aquiring,  meaning you can have a block0 .   lf t55 vali 00088e0 
or from somewhere else in code,  when you down a clone and you updated block0.

You can see "validate block0" as a subset of "validate".   Since there is no validation of writes to tag,  so this subset has its use for the sensitive nature of block0.   And in the long run a validate command should take a blockdata,  try read same block, and return PM3_SUCCESS  or true,   there is not direct need to see the exact reason for failure.   In that case, use a verbose flag, making optional prints instead.

mwalker33, 2019-09-01:
Withdrawn

closed 2019-09-01 by mwalker33

iceman1001, 2019-09-01:
Well,  only close if you are not going to implement it.  We can keep this issue to have documentation about the PR :)