RfidResearchGroup/proxmark3 #384: Compilation error on gcc 9.2.0

unsubtleguy, 2019-09-01:
**Describe the bug**
Compiling on Fedora 29/30 using gcc 9.2.0 (arm-none-eabi-gcc (Fedora 9.2.0-1.fc30) 9.2.0) fails with the following error:

```
appmain.c: In function 'UsbPacketReceived':
appmain.c:1001:40: error: taking address of packed member of 'struct <anonymous>' may result in an unaligned pointer value [-Werror=address-of-packed-member]
 1001 |    setSamplingConfig((sample_config *) c->d.asBytes);
```

**To Reproduce**
Steps to reproduce the behavior:
1. Checkout latest proxmark3
2. Use fedora 29/30 - with gcc 9.2.1
3. ( install prereqs - 'readline-devel', 'arm-none-eabi-gcc-cs', 'arm-none-eabi-newlib', 'avr-libc', 'pcsc-lite', 'pcsc-lite-devel')
4. make

**Expected behavior**
Compilation to succeed

**Desktop (please complete the following information):**
 - OS: Fedora 30 
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version (not relevant)
 - hw status (not relevant)
 - data tune (not relevant)

**Additional context**

Seems very similar to Very similar to https://github.com/RfidResearchGroup/proxmark3/issues/192  - I've modified `common/Makefile.common`  to include `-Wno-address-of-packed-member` under CFLAGs, and compilation succeeded - program seems to work normally after compilation too.

For brevity, the patch i'm describing is as follows:
```
diff --git a/common/Makefile.common b/common/Makefile.common
index 952a5b9..fab1e49 100644
--- a/common/Makefile.common
+++ b/common/Makefile.common
@@ -67,7 +67,7 @@ VPATH = . ../common ../common/crapto1 ../common/mbedtls ../fpga ../zlib
 
 INCLUDES = ../include/proxmark3.h ../include/at91sam7s512.h ../include/config_gpio.h ../include/usb_cmd.h $(APP_INCLUDES)
 
-CFLAGS =  -c $(INCLUDE) -Wall -Werror -pedantic -std=c99 -Os $(APP_CFLAGS)
+CFLAGS =  -c $(INCLUDE) -Wall -Werror -pedantic -std=c99 -Os $(APP_CFLAGS) -Wno-address-of-packed-member 
 LDFLAGS = -nostartfiles -nodefaultlibs -Wl,-gc-sections -n
 
```

doegox, 2019-09-01:
Hmm that's strange, I'm using gcc 9.2.1 under Debian without any issue, investigating...

doegox, 2019-09-01:
I'm sorry but I tried:

```
docker run -it fedora:30
dnf install git make gcc gcc-c++ arm-none-eabi-gcc-cs arm-none-eabi-newlib readline-devel qt5-qtbase-devel libatomic
git clone https://github.com/RfidResearchGroup/proxmark3.git
cd proxmark3
make -j
```

and I didn't get any error.
* gcc (GCC) 9.2.1 20190827 (Red Hat 9.2.1-1)
* arm-none-eabi-gcc gcc version 9.2.0 (Fedora 9.2.0-1.fc30)

unsubtleguy, 2019-09-03:
Embarrassingly - it appears that I've raised this against the wrong repo, as it appears to be a problem in compiling https://github.com/Proxmark/proxmark3.git  - not https://github.com/RfidResearchGroup/proxmark3

Curiously, this repo compiles fine (yet the official repo does not) - I'm not too interested in determining why, as I also have an RDV4 and would benefit from swapping to this project instead.

I'll see myself out, please mark this as an PEBCAC issue :)

closed 2019-09-03 by unsubtleguy

iceman1001, 2019-09-03:
Since you do have a RDV4,  I do recommend use this repo.

doegox, 2019-09-03:
No prob @unsubtleguy and welcome here.
At least your ticket had the side-effect that we've added Fedora installation instructions to our docs ;)