RfidResearchGroup/proxmark3 #385: Questions about Felica support

anson338, 2019-09-02:
On my research on google, iceman should have already WIP in Felica support in months or years ago. 
However in the documentation. I can't find any about it?.

Can I get the status about it? Still, WIP or already finished? and docs? :)

iceman1001, 2019-09-02:
It was actually  @Satsuoni who implemented dump/sim for FeliCA lite tags.  
It isnt a super userfriendly experience to use,  and the device crashed sometimes. 
I tried to fix it but in the end I got stuck not having a FeliCa enabled reader to test with.

You find the source code easily :) armsrc/felica.c  client/cmdhffelica.c    Feel free to fix 'em  :)

Satsuoni, 2019-09-02:
Indeed, and it has been quite a while ago, so my memory is mostly gone. IIRC, it should be able to read iso18092 tags (212khz, and probably 424, but no automatic detection), dump Felica Lite (not full Felica) and emulate NFCID (for basic spoofing), and provide basic reading framework that can be used to add more codes from specification for both Lite and full variants.  I can try answering questions, if you have any.

anson338, 2019-09-02:
Thanks for the Infos. I'll look through and first. And I need to get Felica lite as @Satsuoni mentioned.

Thanks @iceman1001 and @Satsuoni 's reply.

iceman1001, 2019-09-02:
or you start adding full FeliCa support if you have that tag/reader :)

anson338, 2019-09-02:
Yeah, I got full felica tag (octopus card). Hope I could get it to work for full felica :)

Satsuoni, 2019-09-02:
Current version should detect full Felica, but won't dump it, or do much to it. The frame/packet structure is the same, though, so TransmitFor18092_AsReader and WaitForFelicaReply *should* work, though the one Octopus that I had access to had SNR too low for proxmark to detect (it was quite old). It did react to local Suica, though,

iceman1001, 2019-09-02:
Yeah,  well,  could be that I mess it all up :)  so..  anyway.  needs fixin'

closed 2019-09-04 by anson338