RfidResearchGroup/proxmark3 #386: Terminal shows repeated gdk_pixbuf errors at hw tune on kali

jonshorter, 2019-09-03:
**Describe the bug**
On a clean install of Kali under VMware (VMWare tools installed), and using the latest iceman fork, running hw tune brings up the graph but terminal is full of errors, especially when changing the size of the Overlays window.

**To Reproduce**
Steps to reproduce the behavior:
1. set up the environment as described
2. start pm3
3. hw tune

**Expected behavior**
A clear view of the output from hw tune

**Screenshots**

** (proxmark3:12507): WARNING **: 12:13:37.654: Invalid borders specified for theme pixmap:
        /usr/share/themes/Kali-X/gtk-2.0/widgets/Scale/trough-horizontal.png,
borders don't fit within the image

(proxmark3:12507): GdkPixbuf-CRITICAL **: 12:13:37.654: gdk_pixbuf_get_n_channels: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(proxmark3:12507): GdkPixbuf-CRITICAL **: 12:13:37.654: gdk_pixbuf_get_pixels_with_length: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(proxmark3:12507): GdkPixbuf-CRITICAL **: 12:13:37.655: gdk_pixbuf_get_rowstride: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(proxmark3:12507): GdkPixbuf-CRITICAL **: 12:13:37.655: gdk_pixbuf_get_n_channels: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

.... this then perpetually repeats.

**Desktop (please complete the following information):**
 - OS: 

No LSB modules are available.
Distributor ID:	Kali
Description:	Kali GNU/Linux Rolling
Release:	2019.3
Codename:	kali-rolling

VMWare: Fusion Professional Version 11.1.1 
VMWare Tools: 10.3.10.10540 (build-12406962)
Running on macOS Mojave v10.14.6
Hardware Overview:
  Model Name:	MacBook Pro
  Model Identifier:	MacBookPro15,3
  Processor Name:	Intel Core i9
  Processor Speed:	2.9 GHz
  Number of Processors:	1
  Total Number of Cores:	6
  L2 Cache (per Core):	256 KB
  L3 Cache:	12 MB
  Hyper-Threading Technology:	Enabled
  Memory:	32 GB
  Boot ROM Version:	220.270.99.0.0 (iBridge: 16.16.6568.0.0,0)

 - inside proxmark3 client run the following commands and paste the output here.
 [ Proxmark3 RFID instrument ] 
          
 [ CLIENT ]          
  client: RRG/Iceman          
  compiled with GCC 9.2.1 20190821 OS:Linux ARCH:x86_64          

 [ PROXMARK RDV4 ]          
  external flash:                  present           
  smartcard reader:                present           

 [ PROXMARK RDV4 Extras ]          
  FPC USART for BT add-on support: present           
          
 [ ARM ]
  bootrom: RRG/Iceman// 2019-08-25 22:30:53
       os: RRG/Iceman// 2019-08-25 22:31:27
  compiled with GCC 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

 [ FPGA ]
  LF image built for 2s30vq100 on 2019-07-31 at 15:57:16
  HF image built for 2s30vq100 on 2018-09-03 at 21:40:23          

 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev A          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 272207 bytes (52%) Free: 252081 bytes (48%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory          

          
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ] 
          

 [ CLIENT ]          
  client: RRG/Iceman          
  compiled with GCC 9.2.1 20190821 OS:Linux ARCH:x86_64          

 [ PROXMARK RDV4 ]          
  external flash:                  present           
  smartcard reader:                present           

 [ PROXMARK RDV4 Extras ]          
  FPC USART for BT add-on support: present           
          
 [ ARM ]
  bootrom: RRG/Iceman// 2019-08-25 22:30:53
       os: RRG/Iceman// 2019-08-25 22:31:27
  compiled with GCC 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

 [ FPGA ]
  LF image built for 2s30vq100 on 2019-07-31 at 15:57:16
  HF image built for 2s30vq100 on 2018-09-03 at 21:40:23          

 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev A          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 272207 bytes (52%) Free: 252081 bytes (48%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory          

          
[usb] pm3 --> hw status
#db# Memory           
#db#   BIGBUF_SIZE.............40000          
#db#   Available memory........40000          
#db# Tracing           
#db#   tracing ................1          
#db#   traceLen ...............30          
#db# Currently loaded FPGA image           
#db#   mode.................... HF image built for 2s30vq100 on 2018-09-03 at 21:40:23          
#db# Flash memory           
#db#   Baudrate................24 MHz           
#db#   Init....................OK           
#db#   Memory size.............2 mbits / 256 kb           
#db#   Unique ID...............0xD567A882A775A925          
#db# Smart card module (ISO 7816)           
#db#   version.................v3.11           
#db# LF Sampling config           
#db#   [q] divisor.............95 ( 125 kHz )          
#db#   [b] bps.................8          
#db#   [d] decimation..........1          
#db#   [a] averaging...........Yes          
#db#   [t] trigger threshold...0          
#db# LF T55XX config           
#db#            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]          
#db#            mode            |start|write|write|write| read|write|write          
#db#                            | gap | gap |  0  |  1  | gap |  2  |  3          
#db# ---------------------------+-----+-----+-----+-----+-----+-----+------          
#db# fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A |           
#db#     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |           
#db#               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |           
#db#    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 |           
#db#           
#db# Transfer Speed           
#db#   Sending packets to client...          
#db#   Time elapsed............500ms          
#db#   Bytes transferred.......147456          
#db#   Transfer Speed PM3 -> Client = 294912 bytes/s          
#db# Various           
#db#   DBGLEVEL................1          
#db#   ToSendMax...............24          
#db#   ToSendBit...............8          
#db#   ToSend BUFFERSIZE.......2308          
#db#   Slow clock..............30966 Hz          
#db# Installed StandAlone Mode           
#db#   LF HID26 standalone - aka SamyRun (Samy Kamkar)          
#db# Flash memory dictionary loaded           
#db#   Mifare..................820 keys          
#db#   T55x7...................109 keys          
#db#   iClass..................7 keys          
[usb] pm3 --> data tune
          
[=] Measuring antenna characteristics, please wait...
          
[=] You can cancel this operation by pressing the pm3 button          
..
          
[+] LF antenna: 31.30 V - 125.00 kHz          
[+] LF antenna: 39.37 V - 134.00 kHz          
[+] LF optimal: 39.37 V - 133.33 kHz          
[+] LF antenna is OK  
          
[+] HF antenna: 49.00 V - 13.56 MHz          
[+] HF antenna is OK           
          
[+] Displaying LF tuning graph. Divisor 89 is 134kHz, 95 is 125kHz.
          
[usb] pm3 --> 
** (proxmark3:12646): WARNING **: 12:47:45.860: Pixbuf theme: Cannot load pixmap file /usr/share/themes/Kali-X/gtk-2.0/widgets/Scale/trough-horizontal.png: Failed to load image “/usr/share/themes/Kali-X/gtk-2.0/widgets/Scale/trough-horizontal.png”: Fatal error in PNG image file: IDAT: fixed block coding not supported

(proxmark3:12646): GdkPixbuf-CRITICAL **: 12:47:45.860: gdk_pixbuf_get_width: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(proxmark3:12646): GdkPixbuf-CRITICAL **: 12:47:45.860: gdk_pixbuf_get_height: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

** (proxmark3:12646): WARNING **: 12:47:45.860: Invalid borders specified for theme pixmap:
        /usr/share/themes/Kali-X/gtk-2.0/widgets/Scale/trough-horizontal.png,
borders don't fit within the image

(proxmark3:12646): GdkPixbuf-CRITICAL **: 12:47:45.860: gdk_pixbuf_get_n_channels: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(proxmark3:12646): GdkPixbuf-CRITICAL **: 12:47:45.860: gdk_pixbuf_get_pixels_with_length: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(proxmark3:12646): GdkPixbuf-CRITICAL **: 12:47:45.860: gdk_pixbuf_get_rowstride: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(proxmark3:12646): GdkPixbuf-CRITICAL **: 12:47:45.860: gdk_pixbuf_get_n_channels: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(proxmark3:12646): GdkPixbuf-CRITICAL **: 12:47:45.860: gdk_pixbuf_get_pixels_with_length: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

etc....

**Additional context**

Appears to be the same issue as documented here (wrong theme)...
https://ubuntuforums.org/showthread.php?t=1639829

gtpy, 2019-09-03:
@jonshorter same was on Ubuntu -see #354
It is a problem with the theme.
You could try to fix the theme or change it. In Kali I have the same pixbuf warning.
Or you change the theme (Applications: Adwaita, Blackbird,...) as a workaround.

iceman1001, 2019-09-03:
Sounds like Theme has issues on both Ubuntu and Kali.  
Maybe open an issue at their bugtrackers?

Some fast googling, and stackover has some ideas. 
https://stackoverflow.com/questions/11869705/gdkpixbuf-critical-gdk-pixbuf-get-width-assertion-gdk-is-pixbuf-pixbuf

jonshorter, 2019-09-03:
As simple as that! Man I feel foolish heh .. switched from Kali-X to Kali-X Dark theme and all is good! Sorry about that .. I was googling way too deep into things.

closed 2019-09-03 by jonshorter