RfidResearchGroup/proxmark3 #389: Crash on reading World TAG Titan, parallax 125khz

bosb, 2019-09-07:
**Describe the bug**
When reading the tag, proxmark.sh exists
Tag: https://www.digikey.de/product-detail/de/parallax-inc/32399/32399-ND/2666919

**To Reproduce**
Steps to reproduce the behavior:
1. lf search
with the tag on the proxmark 

**Expected behavior**
showing the UID of the tag / at least not crashing

**Screenshots**
```
[usb] pm3 --> lf search
[=] NOTE: some demods output possible binary
[=] if it finds something that looks like a tag
[=] False Positives ARE possible
[=]
[=] Checking for known tags...

Process 52057 stopped
* thread #6, name = 'WorkerThread', stop reason = EXC_BAD_ACCESS (code=2, address=0x700009890000)
    frame #0: 0x00007fff67e54969 libsystem_platform.dylib`_platform_bzero$VARIANT$Haswell + 41
libsystem_platform.dylib`_platform_bzero$VARIANT$Haswell:
->  0x7fff67e54969 <+41>: rep    stosb	%al, %es:(%rdi)
    0x7fff67e5496b <+43>: movq   %rdx, %rax
    0x7fff67e5496e <+46>: popq   %rbp
    0x7fff67e5496f <+47>: retq
Target 0: (proxmark3) stopped.
(lldb) bt
* thread #6, name = 'WorkerThread', stop reason = EXC_BAD_ACCESS (code=2, address=0x700009890000)
  * frame #0: 0x00007fff67e54969 libsystem_platform.dylib`_platform_bzero$VARIANT$Haswell + 41
    frame #1: 0x0000000100031810 proxmark3`ASKDemod_ext(Cmd="64 0 1000 3008", verbose=false, emSearch=false, askType='\x01', stCheck=0x00007000098bd9cf) at cmddata.c:489 [opt]
    frame #2: 0x0000000100031de5 proxmark3`ASKDemod(Cmd=<unavailable>, verbose=<unavailable>, emSearch=<unavailable>, askType=<unavailable>) at cmddata.c:576 [opt]
    frame #3: 0x00000001000a55a5 proxmark3`EM4x50Read(Cmd=<unavailable>, verbose=<unavailable>) at cmdlfem4x.c:938 [opt]
    frame #4: 0x00000001000a2435 proxmark3`CmdLFfind(Cmd=<unavailable>) at cmdlf.c:1057 [opt]
    frame #5: 0x00000001000cbf00 proxmark3`CmdsParse(Commands=0x00000001001d7140, Cmd="search") at cmdparser.c:212 [opt]
    frame #6: 0x00000001000cbf00 proxmark3`CmdsParse(Commands=0x00000001001d9130, Cmd="lf search") at cmdparser.c:212 [opt]
    frame #7: 0x00000001000019bc proxmark3`main_loop(script_cmds_file=<unavailable>, script_cmd=0x0000000000000000, stayInCommandLoop=false) at proxmark3.c:206 [opt]
    frame #8: 0x00000001015e8183 QtCore`___lldb_unnamed_symbol228$$QtCore + 323
    frame #9: 0x00007fff67e602eb libsystem_pthread.dylib`_pthread_body + 126
    frame #10: 0x00007fff67e63249 libsystem_pthread.dylib`_pthread_start + 66
    frame #11: 0x00007fff67e5f40d libsystem_pthread.dylib`thread_start + 13

(lldb) register read
General Purpose Registers:
       rax = 0x0000000000000000
       rbx = 0x0000700009910ab0
       rcx = 0x000000000002d980
       rdx = 0x000070000986f780
       rdi = 0x0000700009890000
       rsi = 0x0000000000000000
       rbp = 0x000070000986f720
       rsp = 0x000070000986f720
        r8 = 0x00007000098bd9cf
        r9 = 0x0000000000000001
       r10 = 0x0000000000000000
       r11 = 0x0000000000000000
       r12 = 0x0000000000000000
       r13 = 0x00007000098bd9cf
       r14 = 0x0000000000000000
       r15 = 0x0000000000000001
       rip = 0x00007fff67e54969  libsystem_platform.dylib`_platform_bzero$VARIANT$Haswell + 41
    rflags = 0x0000000000010206
        cs = 0x000000000000002b
        fs = 0x0000000000000000
        gs = 0x0000000000000000
```
**Desktop (please complete the following information):**
 - OS: MacOS 10.14
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version 
```
 [ Proxmark3 RFID instrument ]
 [ CLIENT ]
  client: RRG/Iceman
  compiled with Clang/LLVM 4.2.1 Compatible Apple LLVM 10.0.0 (clang-1000.11.45.5) OS:OSX ARCH:x86_64

 [ PROXMARK RDV4 ]
  external flash:                  present
  smartcard reader:                present

 [ PROXMARK RDV4 Extras ]
  FPC USART for BT add-on support: present

 [ ARM ]
  bootrom: RRG/Iceman/master/3cf64f9f 2019-08-21 18:19:36
       os: RRG/Iceman/master/5dc63fa2 2019-09-05 11:02:59
  compiled with GCC 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]

 [ FPGA ]
  LF image built for 2s30vq100 on 2019-07-31 at 15:57:16
  HF image built for 2s30vq100 on 2018-09-03 at 21:40:23

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 276633 bytes (53%) Free: 247655 bytes (47%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory
```

 - hw status 
```
#db# Memory
#db#   BIGBUF_SIZE.............40000
#db#   Available memory........40000
#db# Tracing
#db#   tracing ................0
#db#   traceLen ...............23
#db# Currently loaded FPGA image
#db#   mode.................... LF image built for 2s30vq100 on 2019-07-31 at 15:57:16
#db# Flash memory
#db#   Baudrate................24 MHz
#db#   Init....................OK
#db#   Memory size.............2 mbits / 256 kb
#db#   Unique ID...............0xD567A882A71CA926
#db# Smart card module (ISO 7816)
#db#   version.................v3.10
#db# LF Sampling config
#db#   [q] divisor.............95 ( 125 kHz )
#db#   [b] bps.................8
#db#   [d] decimation..........1
#db#   [a] averaging...........Yes
#db#   [t] trigger threshold...0
#db# LF T55XX config
#db#            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
#db#            mode            |start|write|write|write| read|write|write
#db#                            | gap | gap |  0  |  1  | gap |  2  |  3
#db# ---------------------------+-----+-----+-----+-----+-----+-----+------
#db# fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A |
#db#     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
#db#               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |
#db#    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 |
#db#
#db# Transfer Speed
#db#   Sending packets to client...
#db#   Time elapsed............500ms
#db#   Bytes transferred.......1102848
#db#   Transfer Speed PM3 -> Client = 2205696 bytes/s
#db# Various
#db#   DBGLEVEL................1
#db#   ToSendMax...............-1
#db#   ToSendBit...............0
#db#   ToSend BUFFERSIZE.......2308
#db#   Slow clock..............105393 Hz
#db# Installed StandAlone Mode
#db#   No standalone mode present
#db# Flash memory dictionary loaded
 ```
- data tune 
```
[=] Measuring antenna characteristics, please wait...

..

[+] LF antenna: 37.25 V - 125.00 kHz
[+] LF antenna: 29.32 V - 134.00 kHz
[+] LF optimal: 36.96 V - 123.71 kHz
[+] LF antenna is OK

[+] HF antenna: 48.86 V - 13.56 MHz
[+] HF antenna is OK

[+] Displaying LF tuning graph. Divisor 89 is 134kHz, 95 is 125kHz.
```

**Additional context**

iceman1001, 2019-09-07:
thats bad.
How about you make a trace file and share with us, so we can debug properly.
```
lf read
data save titan389.pm3
```

bosb, 2019-09-07:
[titan389.pm3.log](https://github.com/RfidResearchGroup/proxmark3/files/3586789/titan389.pm3.log)

iceman1001, 2019-09-13:
Its a EM4x50 based tag.     Try   **lf em 4x50...**  commands 

ref
https://www.parallax.com/product/32399

bosb, 2019-09-16:
[usb] pm3 --> lf em 4x50_read
./pm3: line 82: 21075 Floating point exception: 8   $CLIENT "$@"

iceman1001, 2019-09-19:
hm..  since 4x50_read actually doesn't read.. I suspect it doesn't check if the graphbuffer has enough data.  I pushed a simple fix.   Compile and test again

```
pm3-> lf read
pm3-> lf em 4x50_demod
pm3-> data setdebug 1
pm3-> lf em 4x05_demod
```

bosb, 2019-09-23:
still not much luck, an empty buffer is catched:
```
[usb] pm3 --> lf em 4x50_read
[!!] Error: EM4x50 - Too little data in Graphbuffer
```
else crashes still exist:
```
lf search
[=] NOTE: some demods output possible binary
[=] if it finds something that looks like a tag
[=] False Positives ARE possible
[=]
[=] Checking for known tags...

./pm3: line 77: 32642 Bus error: 10           $CLIENT "$@"
```

```
[usb] pm3 --> lf read
#db# LF Sampling config
#db#   [q] divisor.............95 ( 125 kHz )
#db#   [b] bps.................8
#db#   [d] decimation..........1
#db#   [a] averaging...........Yes
#db#   [t] trigger threshold...0
#db# Done, saved 40000 out of 40000 seen samples at 8 bits/sample
#db# buffer samples: 00 00 00 00 00 00 00 00 ...
Reading 39999 bytes from device memory

Data fetched
Samples @ 8 bits/smpl, decimation 1:1
[usb] pm3 --> data setdebug 1
[usb] pm3 --> lf em 4x50_demod
[#] LF signal properties:
[#]   high..........254
[#]   low...........1
[#]   mean..........129
[#]   amplitude.....125
[#]   is Noise......No
[#]   THRESHOLD noise amplitude......8

Note: one block = 50 bits (32 data, 12 parity, 6 marker)

Block 0:
./pm3: line 77: 33438 Bus error: 10           $CLIENT "$@"
```

lf read looks modulated:
[image]

iceman1001, 2019-09-23:
Ping @TomHarkness ...

iceman1001, 2019-09-23:
and use   ./client/proxmark3  instead of ./pm3  since the latter is a shell script and the first is the actual executable

bosb, 2019-09-23:
Does this help?
```
[usb] pm3 --> lf read
#db# LF Sampling config
#db#   [q] divisor.............95 ( 125 kHz )
#db#   [b] bps.................8
#db#   [d] decimation..........1
#db#   [a] averaging...........Yes
#db#   [t] trigger threshold...0
#db# Done, saved 40000 out of 40000 seen samples at 8 bits/sample
#db# buffer samples: 00 00 00 00 00 00 00 00 ...
Reading 39999 bytes from device memory

Data fetched
Samples @ 8 bits/smpl, decimation 1:1
[#] LF signal properties:
[#]   high..........255
[#]   low...........1
[#]   mean..........128
[#]   amplitude.....127
[#]   is Noise......No
[#]   THRESHOLD noise amplitude......8
[#] DEBUG: (setClockGrid) clear settings
[usb] pm3 --> lf em 4x50_demod
[#] LF signal properties:
[#]   high..........255
[#]   low...........1
[#]   mean..........128
[#]   amplitude.....127
[#]   is Noise......No
[#]   THRESHOLD noise amplitude......8

Note: one block = 50 bits (32 data, 12 parity, 6 marker)

Block 0:
Process 49225 stopped
* thread #6, name = 'WorkerThread', stop reason = EXC_BAD_ACCESS (code=2, address=0x700000b83000)
    frame #0: 0x00007fff5a752969 libsystem_platform.dylib`_platform_bzero$VARIANT$Haswell + 41
libsystem_platform.dylib`_platform_bzero$VARIANT$Haswell:
->  0x7fff5a752969 <+41>: rep    stosb	%al, %es:(%rdi)
    0x7fff5a75296b <+43>: movq   %rdx, %rax
    0x7fff5a75296e <+46>: popq   %rbp
    0x7fff5a75296f <+47>: retq
Target 0: (proxmark3) stopped.
(lldb) bt
* thread #6, name = 'WorkerThread', stop reason = EXC_BAD_ACCESS (code=2, address=0x700000b83000)
  * frame #0: 0x00007fff5a752969 libsystem_platform.dylib`_platform_bzero$VARIANT$Haswell + 41
    frame #1: 0x0000000100032470 proxmark3`ASKDemod_ext(Cmd="64 0 1000 3008", verbose=false, emSearch=false, askType='\x01', stCheck=0x0000700000bb092f) at cmddata.c:521 [opt]
    frame #2: 0x0000000100032a45 proxmark3`ASKDemod(Cmd=<unavailable>, verbose=<unavailable>, emSearch=<unavailable>, askType=<unavailable>) at cmddata.c:608 [opt]
    frame #3: 0x00000001000a70a7 proxmark3`EM4x50Read(Cmd=<unavailable>, verbose=<unavailable>) at cmdlfem4x.c:952 [opt]
    frame #4: 0x00000001000d05c0 proxmark3`CmdsParse(Commands=0x00000001001e4aa0, Cmd="4x50_demod") at cmdparser.c:212 [opt]
    frame #5: 0x00000001000d05c0 proxmark3`CmdsParse(Commands=0x00000001001e4480, Cmd="em 4x50_demod") at cmdparser.c:212 [opt]
    frame #6: 0x00000001000d05c0 proxmark3`CmdsParse(Commands=0x00000001001e6630, Cmd="lf em 4x50_demod") at cmdparser.c:212 [opt]
    frame #7: 0x0000000100001bcc proxmark3`main_loop(script_cmds_file=<unavailable>, script_cmd=0x0000000000000000, stayInCommandLoop=false) at proxmark3.c:248 [opt]
    frame #8: 0x00000001015f9183 QtCore`___lldb_unnamed_symbol228$$QtCore + 323
    frame #9: 0x00007fff5a75e2eb libsystem_pthread.dylib`_pthread_body + 126
    frame #10: 0x00007fff5a761249 libsystem_pthread.dylib`_pthread_start + 66
    frame #11: 0x00007fff5a75d40d libsystem_pthread.dylib`thread_start + 13
```

iceman1001, 2019-09-23:
That help a bit.  Now I need the debug statments to go with what you just did.  gdb and run the following

```
lf read
data setd 2
lf em 4x50_demod
```

bosb, 2019-09-23:
this doesn't look different?
```
[usb] pm3 --> lf read
#db# LF Sampling config
#db#   [q] divisor.............95 ( 125 kHz )
#db#   [b] bps.................8
#db#   [d] decimation..........1
#db#   [a] averaging...........Yes
#db#   [t] trigger threshold...0
#db# Done, saved 40000 out of 40000 seen samples at 8 bits/sample
#db# buffer samples: 00 00 00 00 00 00 00 00 ...
Reading 39999 bytes from device memory

Data fetched
Samples @ 8 bits/smpl, decimation 1:1
[usb] pm3 --> data setd 2
[usb] pm3 --> lf em 4x50_demod
[#] LF signal properties:
[#]   high..........255
[#]   low...........1
[#]   mean..........128
[#]   amplitude.....127
[#]   is Noise......No
[#]   THRESHOLD noise amplitude......8

Note: one block = 50 bits (32 data, 12 parity, 6 marker)

Block 0:
Process 56362 stopped
* thread #6, name = 'WorkerThread', stop reason = EXC_BAD_ACCESS (code=2, address=0x70000a258000)
    frame #0: 0x00007fff5a752969 libsystem_platform.dylib`_platform_bzero$VARIANT$Haswell + 41
libsystem_platform.dylib`_platform_bzero$VARIANT$Haswell:
->  0x7fff5a752969 <+41>: rep    stosb	%al, %es:(%rdi)
    0x7fff5a75296b <+43>: movq   %rdx, %rax
    0x7fff5a75296e <+46>: popq   %rbp
    0x7fff5a75296f <+47>: retq
Target 0: (proxmark3) stopped.
(lldb) bt
* thread #6, name = 'WorkerThread', stop reason = EXC_BAD_ACCESS (code=2, address=0x70000a258000)
  * frame #0: 0x00007fff5a752969 libsystem_platform.dylib`_platform_bzero$VARIANT$Haswell + 41
    frame #1: 0x0000000100032470 proxmark3`ASKDemod_ext(Cmd="64 0 1000 3008", verbose=false, emSearch=false, askType='\x01', stCheck=0x000070000a28592f) at cmddata.c:521 [opt]
    frame #2: 0x0000000100032a45 proxmark3`ASKDemod(Cmd=<unavailable>, verbose=<unavailable>, emSearch=<unavailable>, askType=<unavailable>) at cmddata.c:608 [opt]
    frame #3: 0x00000001000a70a7 proxmark3`EM4x50Read(Cmd=<unavailable>, verbose=<unavailable>) at cmdlfem4x.c:952 [opt]
    frame #4: 0x00000001000d05c0 proxmark3`CmdsParse(Commands=0x00000001001e4aa0, Cmd="4x50_demod") at cmdparser.c:212 [opt]
    frame #5: 0x00000001000d05c0 proxmark3`CmdsParse(Commands=0x00000001001e4480, Cmd="em 4x50_demod") at cmdparser.c:212 [opt]
    frame #6: 0x00000001000d05c0 proxmark3`CmdsParse(Commands=0x00000001001e6630, Cmd="lf em 4x50_demod") at cmdparser.c:212 [opt]
    frame #7: 0x0000000100001bcc proxmark3`main_loop(script_cmds_file=<unavailable>, script_cmd=0x0000000000000000, stayInCommandLoop=false) at proxmark3.c:248 [opt]
    frame #8: 0x00000001015f9183 QtCore`___lldb_unnamed_symbol228$$QtCore + 323
    frame #9: 0x00007fff5a75e2eb libsystem_pthread.dylib`_pthread_body + 126
    frame #10: 0x00007fff5a761249 libsystem_pthread.dylib`_pthread_start + 66
    frame #11: 0x00007fff5a75d40d libsystem_pthread.dylib`thread_start + 13
```

iceman1001, 2019-09-23:
Looking at your messages,  it looks like it breaks line 521 in cmddata.c   But thats just a array...
Have you pulled latest code and can trigger the bug again,  since cmddata.c was update two days ago.

```
    frame #1: proxmark3`ASKDemod_ext(Cmd="64 0 1000 3008", verbose=false, emSearch=false, askType='\x01',) at cmddata.c:521
```

https://github.com/RfidResearchGroup/proxmark3/blob/master/client/cmddata.c#L521

bosb, 2019-09-23:
Actually that is the line:  521     uint8_t bits[MAX_GRAPH_TRACE_LEN] = {0};
will pull and  make clean, and make again....
and it stays like this:
```
[usb] pm3 --> lf read
#db# LF Sampling config
#db#   [q] divisor.............95 ( 125 kHz )
#db#   [b] bps.................8
#db#   [d] decimation..........1
#db#   [a] averaging...........Yes
#db#   [t] trigger threshold...0
#db# Done, saved 40000 out of 40000 seen samples at 8 bits/sample
#db# buffer samples: 00 00 00 00 00 00 00 00 ...
Reading 39999 bytes from device memory

Data fetched
Samples @ 8 bits/smpl, decimation 1:1
[usb] pm3 --> data plot
[usb] pm3 --> data setd 2
[usb] pm3 --> lf em 4x50_demod
[#] LF signal properties:
[#]   high..........255
[#]   low...........1
[#]   mean..........129
[#]   amplitude.....126
[#]   is Noise......No
[#]   THRESHOLD noise amplitude......8

Note: one block = 50 bits (32 data, 12 parity, 6 marker)

Block 0:
Process 79064 stopped
* thread #6, name = 'WorkerThread', stop reason = EXC_BAD_ACCESS (code=2, address=0x700003972000)
    frame #0: 0x00007fff5a752969 libsystem_platform.dylib`_platform_bzero$VARIANT$Haswell + 41
libsystem_platform.dylib`_platform_bzero$VARIANT$Haswell:
->  0x7fff5a752969 <+41>: rep    stosb	%al, %es:(%rdi)
    0x7fff5a75296b <+43>: movq   %rdx, %rax
    0x7fff5a75296e <+46>: popq   %rbp
    0x7fff5a75296f <+47>: retq
Target 0: (proxmark3) stopped.
(lldb) bt
* thread #6, name = 'WorkerThread', stop reason = EXC_BAD_ACCESS (code=2, address=0x700003972000)
  * frame #0: 0x00007fff5a752969 libsystem_platform.dylib`_platform_bzero$VARIANT$Haswell + 41
    frame #1: 0x00000001000328f0 proxmark3`ASKDemod_ext(Cmd="64 0 1000 3008", verbose=false, emSearch=false, askType='\x01', stCheck=0x000070000399f92f) at cmddata.c:521 [opt]
    frame #2: 0x0000000100032ec5 proxmark3`ASKDemod(Cmd=<unavailable>, verbose=<unavailable>, emSearch=<unavailable>, askType=<unavailable>) at cmddata.c:608 [opt]
    frame #3: 0x00000001000a7b67 proxmark3`EM4x50Read(Cmd=<unavailable>, verbose=<unavailable>) at cmdlfem4x.c:952 [opt]
    frame #4: 0x00000001000d1100 proxmark3`CmdsParse(Commands=0x00000001001e5ac0, Cmd="4x50_demod") at cmdparser.c:212 [opt]
    frame #5: 0x00000001000d1100 proxmark3`CmdsParse(Commands=0x00000001001e5480, Cmd="em 4x50_demod") at cmdparser.c:212 [opt]
    frame #6: 0x00000001000d1100 proxmark3`CmdsParse(Commands=0x00000001001e7670, Cmd="lf em 4x50_demod") at cmdparser.c:212 [opt]
    frame #7: 0x0000000100001f8c proxmark3`main_loop(script_cmds_file=<unavailable>, script_cmd=0x0000000000000000, stayInCommandLoop=false) at proxmark3.c:248 [opt]
    frame #8: 0x00000001015fa183 QtCore`___lldb_unnamed_symbol228$$QtCore + 323
    frame #9: 0x00007fff5a75e2eb libsystem_pthread.dylib`_pthread_body + 126
    frame #10: 0x00007fff5a761249 libsystem_pthread.dylib`_pthread_start + 66
    frame #11: 0x00007fff5a75d40d libsystem_pthread.dylib`thread_start + 13
```

iceman1001, 2019-09-23:
hm.. did you get any warnings when compiling?  and how much RAM memory do your computer have?

line 521 is still just an array allocation....

iceman1001, 2019-09-23:
```uname -a```

bosb, 2019-09-23:
uname -a
Darwin nb-tbosbach7 18.6.0 Darwin Kernel Version 18.6.0: Sun Apr 28 18:06:45 PDT 2019; root:xnu-4903.261.4~6/RELEASE_X86_64 x86_64

memory: 16 GB

only 2 times smth. like this:
[=] AR libreveng.a
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: libreveng.a(bmpbit.o) has no symbols

iceman1001, 2019-09-23:
try pulling and testing again.

bosb, 2019-09-23:
not crashing anymore 👍 

```
[usb] pm3 --> lf read
#db# LF Sampling config
#db#   [q] divisor.............95 ( 125 kHz )
#db#   [b] bps.................8
#db#   [d] decimation..........1
#db#   [a] averaging...........Yes
#db#   [t] trigger threshold...0
#db# Done, saved 40000 out of 40000 seen samples at 8 bits/sample
#db# buffer samples: 00 00 00 00 00 00 00 00 ...
Reading 39999 bytes from device memory

Data fetched
Samples @ 8 bits/smpl, decimation 1:1
[usb] pm3 --> data plot
[usb] pm3 --> lf em 4x50_demod

Note: one block = 50 bits (32 data, 12 parity, 6 marker)

Block 0:
11111111 1 -> 0xff
11111111 1 -> 0xff
11111111 1 -> 0xff
11111111 1 -> 0xff

11111111 1 -> 0xff
[+] Parity checks | Fail
Found data at sample: 3580 - using clock: 64
Block 0: ffffffff
Parities checks | Fail
Try cleaning the read samples with 'data askedge'
```

and also lf search finishes nicely.

iceman1001, 2019-09-23:
Sounds like your OS has a limit for your stack...  You can temporary unlimit it.
```unlimit -a```

closed 2019-09-23 by iceman1001