RfidResearchGroup/proxmark3 #392: $HOME and windows 10 command line.

mwalker33, 2019-09-11:
**Describe the bug**
Minor bug on windows 10, there is no environment variable HOME, so when running the client on a windows command line  "proxmark3.exe <com port>" 

Result:
Could not retrieve $HOME from the environment
[!!] No history will be recorded

If you set the HOME environment variable, it will fail to create the folder
Could not create user directory \Users\mwalker/.proxmark3/

a "mkdir \Users\mwalker/.proxmark3/" from the command line also fails.
Changing the slash to backslash works.
 
[!!] No history will be recorded

**Expected behavior**
Maybe default to proxmark3.exe folder if no $HOME is found (my preferred) ?

**Desktop (please complete the following information):**
 - OS: Windows 10
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman
  compiled with MinGW-w64 8.2.0 OS:Windows (64b) ARCH:x86_64

 [ PROXMARK RDV4 ]
  external flash:                  present
  smartcard reader:                present

 [ PROXMARK RDV4 Extras ]
  FPC USART for BT add-on support: absent

 [ ARM ]
  bootrom: RRG/Iceman/master/b16f2fda-dirty-unclean 2019-07-24 08:25:08
       os: RRG/Iceman/master/a384774b-dirty-unclean 2019-09-11 20:12:07
  compiled with GCC 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

 [ FPGA ]
  LF image built for 2s30vq100 on 2019-07-31 at 15:57:16
  HF image built for 2s30vq100 on 2018-09-03 at 21:40:23

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 270485 bytes (52%) Free: 253803 bytes (48%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

 - hw status
#db# Memory
#db#   BIGBUF_SIZE.............40000
#db#   Available memory........40000
#db# Tracing
#db#   tracing ................0
#db#   traceLen ...............115
#db# Currently loaded FPGA image
#db#   mode.................... HF image built for 2s30vq100 on 2018-09-03 at 21:40:23
#db# Flash memory
#db#   Baudrate................24 MHz
#db#   Init....................OK
#db#   Memory size.............2 mbits / 256 kb
#db#   Unique ID...............0xD567A882A734BD26
#db# Smart card module (ISO 7816)
#db#   version.................v3.10
#db# LF Sampling config
#db#   [q] divisor.............95 ( 125 kHz )
#db#   [b] bps.................8
#db#   [d] decimation..........1
#db#   [a] averaging...........Yes
#db#   [t] trigger threshold...0
#db# LF T55XX config
#db#            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
#db#            mode            |start|write|write|write| read|write|write
#db#                            | gap | gap |  0  |  1  | gap |  2  |  3
#db# ---------------------------+-----+-----+-----+-----+-----+-----+------
#db# fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A |
#db#     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
#db#               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |
#db#    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 |
#db#
#db# Transfer Speed
#db#   Sending packets to client...
#db#   Time elapsed............500ms
#db#   Bytes transferred.......293888
#db#   Transfer Speed PM3 -> Client = 587776 bytes/s
#db# Various
#db#   DBGLEVEL................1
#db#   ToSendMax...............39
#db#   ToSendBit...............8
#db#   ToSend BUFFERSIZE.......2308
#db#   Slow clock..............29547 Hz
#db# Installed StandAlone Mode
#db#   LF HID26 standalone - aka SamyRun (Samy Kamkar)
#db# Flash memory dictionary loaded
#db#   T55x7...................79 keys

 - data tune
[=] Measuring antenna characteristics, please wait...

[=] You can cancel this operation by pressing the pm3 button
..

[+] LF antenna: 37.25 V - 125.00 kHz
[+] LF antenna: 28.75 V - 134.00 kHz
[+] LF optimal: 37.11 V - 123.71 kHz
[+] LF antenna is OK

[+] HF antenna: 48.44 V - 13.56 MHz
[+] HF antenna is OK

[+] Displaying LF tuning graph. Divisor 89 is 134kHz, 95 is 125kHz.

**Additional context**
Minor issue:
Tested commands seem to work ok.  It found the mfc_default_keys.dic in the client folder OK.

doegox, 2019-09-11:
Are you using ProxSpace3.2 or is it a manual setup? (I think it works under PS3.2)
If it's a manual setup, could you share instructions how to reproduce your setup?
Thanks.

mwalker33, 2019-09-11:
If I run it under ProxSpace it works ok/as expected.

The issue is if you run direct from the command-line, no ProxSpace
Up until now, I have used ProxSpace to build, but run direct (and have been referencing other folders outside of proxspace e.g. d:\pm3\traces or to use custom dic files.

If I create the HOME environment variable (under windows) and create the .proxmark3 folder then it will work and save the files into that folder. 

So 2 extra steps for direct windows running
1. set HOME=c:\users\mwalker
2. mkdir %HOME%\.proxmark3
then run
proxmark3.exe com4
All seems to work.

Since that will be lost on that cmd.exe session close, Windows users would need to create a batch file to re-map the HOME or add it to the windows setup. (step 3)

So to have it fail back to the client folder (if compiled for windows), would then allow either way to work.

Note: this is not a high priority for me, just noticed it and the history function is very useful :) But my work-around works for me.

Thanks