RfidResearchGroup/proxmark3 #394: ic

thesle3p, 2019-09-11:


thesle3p, 2019-09-11:
not sure why but whenever i try and use iclass functionality i get this 
```
          
[usb] pm3 --> hf iclass chk f dictionaries/iclass_default_keys.dic
[+] loaded  7 keys from dictionary file dictionaries/iclass_default_keys.dic           
[+] Reading tag CSN          
[+] Generating diversified keys          
[+] Searching for DEBIT key          
[+] Tag info          
[+] CSN     | 60 F8 A9 01 F8 FF 12 E0           
[+] CCNR    | FF FF FF FF FF FF FF FF 00 00 00 00           
#db# [-] Handshake status | 0 (fail 10)          
          
[+] Time in iclass checkkeys: 2 seconds
```

iceman1001, 2019-09-11:
that is me playing.

thesle3p, 2019-09-11:
I beg your pardon?

iceman1001, 2019-09-11:
I am testing new timeout values :)  and it got into the repo.  Hold on.

thesle3p, 2019-09-11:
ah ok, why push experimental code to the public repo though?

iceman1001, 2019-09-11:
bleeding edge.

thesle3p, 2019-09-11:
fair enough

thesle3p, 2019-09-11:
pulled down changes, bug still presists

thesle3p, 2019-09-11:
Any ideas?

iceman1001, 2019-09-11:
pull again.

thesle3p, 2019-09-11:
chk works but now dump and rdbl don't They both return a failed to authenticate error

iceman1001, 2019-09-11:
try again. Should be fixed :)

thesle3p, 2019-09-13:
yep working fine now. mifare and LF modules are more stable too

closed 2019-09-13 by thesle3p