RfidResearchGroup/proxmark3 #401: [SOLVED] - flashing image on device does not complete, by virtual Debian machine on Mac host

giovannicandotti, 2019-09-15:
**Describe the bug**
Hard time to update flash image on device.

**To Reproduce**
Steps to reproduce the behavior:
1. Environment: Mac host, VirtualBox, virtual machine with Debian, device assigned on USB port
2. follow the procedure from Github
3. device is found on /dev/ttyACM0, switches changes to 2 red on
4. the flash step never begins, as the device is no longer "Found"

**Expected behaviour**
Flash of image should start with no delay.

**SOLUTION**
The command to turn the device to bootloader update makes the device to be detached from Linux host.
--> while starting the flashing command, after the first "Found", quickly check assigned "Devices-->USB" (top of the screen, Virtualbox option) and re-assign the proxmark device to the virtual machine.
THIS assignment will make the procedure to end successfully

**Additional context**
I hope this will help someone to avoid loosing time, checking logs and trying to understand why the flash procedure does not complete

iceman1001, 2019-09-15:
Yeah,  virtual machines,  not an issue of the repo but nice to have it documented.
Maybe add a page in Wiki about **when on virtual machines..** ?

closed 2019-09-15 by iceman1001

giovannicandotti, 2019-09-15:
This is Definitely not an issue of the repo, I hope wiki is the right place to share similar findings.
A section on Wiki, could be relevant if there are other similar issues, frequently.

doegox, 2019-09-15:
Easier trick is to enable automatic USB mapping, cf "USB device filters" on Virtualbox VM settings so when device appears, it gets automatically hooked to the VM

giovannicandotti, 2019-09-15:
> Easier trick is to enable automatic USB mapping, cf "USB device filters" on Virtualbox VM settings so when device appears, it gets automatically hooked to the VM

That was in place already, but it didn't work properly.
The first time, hooking was successful, but after changing the status by the procedure it was not assigned again. VIrtualbox 6.0, may be a problem with that?