RfidResearchGroup/proxmark3 #402: endless loop after "hf search" ... Again :(

thunder0666, 2019-09-16:
like the other issue: https://github.com/RfidResearchGroup/proxmark3/issues/397

**proxmark not responding**
after a hf search command the leds "A and D" ligth up.

**for reproduce**
promark3 -c "hf search"

```
pm3 -c "hf search"
[+] execute command from commandline: hf search

[=] Using UART port /dev/ttyACM0
[=] Communicating with PM3 over USB-CDC
pm3 --> hf search
[=] Checking for known tags...

[/] Searching for iClass / PicoPass tag...[=] You can cancel this operation by pressing the pm3 button

[-] No known/supported 13.56 MHz tags found
```
and the proxmark goes to endless loop ( it's comming hot :) )
the only way to restart it's to disconnect the device.
( no tag is normal i try with nothing )

**HW and SOFT Versions**

```
pm3 -c "hw version"
[=] Session log /home/vincent/.proxmark3/log_20190912.txt
[+] execute command from commandline: hw version

[=] Using UART port /dev/ttyACM0
[=] Communicating with PM3 over USB-CDC
pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman
  compiled with GCC 8.3.0 OS:Linux ARCH:x86_64

 [ PROXMARK RDV4 ]
  external flash:                  present
  smartcard reader:                present

 [ PROXMARK RDV4 Extras ]
  FPC USART for BT add-on support: absent

 [ ARM ]
  bootrom: RRG/Iceman/master/8ba2f5b0 2019-09-16 10:02:56
       os: RRG/Iceman/master/8ba2f5b0 2019-09-16 10:02:59
  compiled with GCC 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

 [ FPGA ]
  LF image built for 2s30vq100 on 2019-07-31 at 15:57:16
  HF image built for 2s30vq100 on 2018-09-03 at 21:40:23

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 270323 bytes (52%) Free: 253965 bytes (48%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory
```

iceman1001, 2019-09-16:
my bad,  yet again did one of my unreleased code snuck into the commit.
Try again.

thunder0666, 2019-09-16:
Unable to compiling. under linux Rev: 3726df62
------------------------------------------------------------------------------
[-] CC cmdlfjablotron.c
cmdlfindala.c: In function ‘CmdIndalaClone’:
cmdlfindala.c:487:54: error: ‘datawords’ undeclared (first use in this function)
         SendCommandOLD(CMD_LF_INDALA_CLONE, 0, 0, 0, datawords, sizeof(datawords));
                                                      ^~~~~~~~~
cmdlfindala.c:487:54: note: each undeclared identifier is reported only once for each function it appears in
cmdlfindala.c:483:18: error: variable ‘blocks’ set but not used [-Werror=unused-but-set-variable]
         uint32_t blocks[2] = {0};
                  ^~~~~~
cc1: all warnings being treated as errors
make[1]: *** [Makefile:426: obj/cmdlfindala.o] Error 1
===================================================================
Platform name:     Proxmark3 rdv4
PLATFORM:          PM3RDV4
Platform extras:   No extra selected
Included options:  SMARTCARD FLASH LF HITAG ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF
Standalone mode:   LF_SAMYRUN
===================================================================

Sorry for this...
can we have a right to become a collaborator ? ( for directly modify this per example. i work only on liinux )

iceman1001, 2019-09-16:
pull again.

iceman1001, 2019-09-16:
And you don't need to be a collaborator in order to do a PR.

thunder0666, 2019-09-16:
ok all right ! :)  thx again.

closed 2019-09-16 by thunder0666