RfidResearchGroup/proxmark3 #406: Unable to correctly interact with some EM-Marin tags

S1yk3r, 2019-09-19:
I met a EM-Marin TAG ID which is impossible to read/write on iceman proxmark3 firmware, example: **24001AC532**. Maybe its value violates some standard range rules, but i met it on are real intercom, and official PM3 firmware works well with it.
(I use precompiled PM3 iceman bulids from here: https://drive.google.com/open?id=1mMgdfnSEgFvA77xvbG_VnaLrZJHnvcSP/)

iceman1001, 2019-09-19:
Did you test latest source from here? 

The precompiled build links found in forum post http://proxmark.org/forum/viewtopic.php?id=3975
say RDV40,  but you can flash and run it on other Proxmark3 devices aswell.

S1yk3r, 2019-09-19:
I can't use this (rdv40-64-20190918) build on my device:
https://aliexpress.com/item//32808141763.html
USB\VID_9AC4&PID_4B8F&REV_0001

After successful flashing "bootrom.elf" device starting correctly, but flashing "fullimage.elf" will brick device ("Device Descriptor Request Failed" etc.)

doegox, 2019-09-19:
you have to compile it with PLATFORM=PM3_OTHER

iceman1001, 2019-09-19:
If you don't use this repo,  then there is little to be done here.   Iceman repo is deprecated.
As mentioned, you can compile and use this repo on your non-rdv4 device.   If you do,  I think your tag will work again.

However until you do,   you can do one thing more.   Save a read trace and upload here.

```
lf read
data save  s1yk3r_em_24001AC532.pm3
```

iceman1001, 2019-09-24:
ping!

S1yk3r, 2019-09-26:
Reading (on iceman builds) T5577 flashed as "**lf em 410x_write 24001AC532 1**" will show the next output in console:

pm3 --> lf search
NOTE: some demods output possible binary
  if it finds something that looks like a tag
False Positives ARE possible

Checking for known tags:

NEDAP ID Found - Raw: ff805fb0009d4794ff805fb0009d4794
 - UID: FD01FF
 - i: 1300
 - Checksum2 CA51

[+] Valid NEDAP ID Found!

[s1yk3r_em_24001AC532.zip](https://github.com/RfidResearchGroup/proxmark3/files/3656543/s1yk3r_em_24001AC532.zip)

On command "**lf em 410x_read**" it displays reliable information for the tag. 
Looks like claims are only addressed to the (**lf search**) automatic detection mechanism.

It is a sad that there are no precompiled RDV4 builds for my device (i wanted to respond when i compile the RDV4 firmware - but i was busy). Sorry for such delay in answer.

iceman1001, 2019-09-26:
You can test the improved nedap commands instead of writing hex.
```
lf nedap clone
lf nedap read
```

Glad to see your issue is resolved.

closed 2019-09-26 by iceman1001