RfidResearchGroup/proxmark3 #407: Incorrect identification of magic commands (GEN 1b)

b4ndit, 2019-09-20:
**Describe the bug**
Just started playing with this fork today and I noticed that all of the cards I have are being incorrectly identified as Chinese magic cards (GEN 1b). These cards are from random hotels I've stayed at in the past couple years, and I've scanned them before, so I know they're not Chinese magic cards.

**To Reproduce**
Steps to reproduce the behavior:
1. Place any legit Mifare card on the Proxmark
2. hf search
4. See "[+] Answers to magic commands (GEN 1b): YES"

**Expected behavior**
Should be "[+] Answers to magic commands (GEN 1b): NO"

**Screenshots**
```
[usb] pm3 --> hf search
[=] Checking for known tags...
          
[|] Searching for ISO14443-A tag... UID : 79 66 21 14           
ATQA : 00 04          
 SAK : 08 [2]          
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1 | 1k Ev1          
[=] proprietary non iso14443-4 card found, RATS not supported          
[+] Answers to magic commands (GEN 1b): YES           
[+] Prng detection: WEAK           
          
[+] Valid ISO14443-A tag  found
```

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04.3 LTS
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
```
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ] 
          

 [ CLIENT ]          
  client: RRG/Iceman          
  compiled with GCC 7.4.0 OS:Linux ARCH:x86_64          

 [ PROXMARK RDV4 ]          
  external flash:                  absent           
  smartcard reader:                absent           

 [ PROXMARK RDV4 Extras ]          
  FPC USART for BT add-on support: absent           
          
 [ ARM ]
  bootrom: RRG/Iceman/master/e5ffcfd5 2019-09-19 15:51:50
       os: RRG/Iceman/master/e5ffcfd5 2019-09-19 15:52:01
  compiled with GCC 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]

 [ FPGA ]
  LF image built for 2s30vq100 on 2019-07-31 at 15:57:16
  HF image built for 2s30vq100 on 2018-09-03 at 21:40:23          

 [ Hardware ]           
  --= uC: AT91SAM7S256 Rev D          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 256K bytes, Used: 242404 bytes (92%) Free: 19740 bytes ( 8%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory          
```

 - hw status
```
[usb] pm3 --> hw status
#db# Memory           
#db#   BIGBUF_SIZE.............40000          
#db#   Available memory........40000          
#db# Tracing           
#db#   tracing ................0          
#db#   traceLen ...............103          
#db# Currently loaded FPGA image           
#db#   mode.................... HF image built for 2s30vq100 on 2018-09-03 at 21:40:23          
#db# LF Sampling config           
#db#   [q] divisor.............95 ( 125 kHz )          
#db#   [b] bps.................8          
#db#   [d] decimation..........1          
#db#   [a] averaging...........Yes          
#db#   [t] trigger threshold...0          
#db# LF T55XX config           
#db#            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]          
#db#            mode            |start|write|write|write| read|write|write          
#db#                            | gap | gap |  0  |  1  | gap |  2  |  3          
#db# ---------------------------+-----+-----+-----+-----+-----+-----+------          
#db# fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A |           
#db#     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |           
#db#               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |           
#db#    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 |           
#db#           
#db# Transfer Speed           
#db#   Sending packets to client...          
#db#   Time elapsed............500ms          
#db#   Bytes transferred.......256000          
#db#   Transfer Speed PM3 -> Client = 512000 bytes/s          
#db# Various           
#db#   DBGLEVEL................1          
#db#   ToSendMax...............39          
#db#   ToSendBit...............8          
#db#   ToSend BUFFERSIZE.......2308          
#db#   Slow clock..............30985 Hz          
#db# Installed StandAlone Mode           
#db#   LF HID26 standalone - aka SamyRun (Samy Kamkar)
```

 - data tune
```
[usb] pm3 --> data tune
          
[=] Measuring antenna characteristics, please wait...
          
[=] You can cancel this operation by pressing the pm3 button          
..
          
[+] LF antenna: 21.24 V - 125.00 kHz          
[+] LF antenna: 18.98 V - 134.00 kHz          
[+] LF optimal: 21.67 V - 126.32 kHz          
[+] LF antenna is OK  
          
[+] HF antenna: 22.93 V - 13.56 MHz          
[+] HF antenna is OK           
          
[+] Displaying LF tuning graph. Divisor 89 is 134kHz, 95 is 125kHz.
```

**Additional context**
Chinese magic cards (GEN 1a) are still correctly identified. I don't think I have any GEN 1b cards to test with to see what it'll do.

iceman1001, 2019-09-20:
cool, 
It could be the MFC Ev1 which has 0x43 command messing with detection.
can you run hf 14a list afterwards see if we can look at the actual transaction.

b4ndit, 2019-09-20:
Hey iceman,

Here's what I ran:
```
[usb] pm3 --> hf search
[=] Checking for known tags...

[|] Searching for ISO14443-A tag... UID : 9D 72 EE 50
ATQA : 00 04
 SAK : 08 [2]
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1 | 1k Ev1
[=] proprietary non iso14443-4 card found, RATS not supported
[+] Answers to magic commands (GEN 1b): YES
[+] Prng detection: WEAK

[+] Valid ISO14443-A tag  found

[usb] pm3 --> hf 14a list
[+] Recorded Activity (TraceLen = 103 bytes)
[=]
Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
ISO14443A - All times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |        992 | Rdr |52                                                                       |     | WUPA
       2244 |       4612 | Tag |04  00                                                                   |     |
       7040 |       9504 | Rdr |93  20                                                                   |     | ANTICOLL
      10692 |      16580 | Tag |9d  72  ee  50  51                                                       |     |
      19072 |      29536 | Rdr |93  70  9d  72  ee  50  51  8f  71                                       |  ok | SELECT_UID
      30788 |      34308 | Tag |08  b6  dd                                                               |     |
      44032 |      48736 | Rdr |60  00  f5  7b                                                           |  ok | AUTH-A(0)
      50372 |      55044 | Tag |ac  48  d7  c9                                                           |     |
[usb] pm3 -->
```

Is that what you needed?

iceman1001, 2019-09-21:
Nop,  that didn't do it.   Can you run the following?
```
hf 14a raw -p -a -b 7 40
hf 14a raw -p -a 43
hf 14a list
``

doegox, 2019-09-21:
Should be fixed now by https://github.com/RfidResearchGroup/proxmark3/commit/6422cc6d132c37d1cfdb9717fd31e9c5ae266bcc
Bug was introduced at https://github.com/RfidResearchGroup/proxmark3/commit/74cd1bee3513b8970ad9765cc8b5f9851ec7b406

closed 2019-09-21 by doegox