RfidResearchGroup/proxmark3 #408: Mifare card causes "UART:: write time-out"

b4ndit, 2019-09-20:
**Describe the bug**
When running a nested attack on a specific Mifare card, it causes the Proxmark3 Easy to lock up and the client application errors with "UART:: write time-out." This does not happen on any of the other Mifare cards I've used, it's only this card out of the 30+ I've tested.

I'm running Iceman's fork compiled yesterday on the Proxmark3 Easy. Before compiling, I copied Makefile.platform.sample to Makefile.platform and its contents are:
```
➜  proxmark3 git:(master) cat Makefile.platform
# If you want to use it, copy this file as Makefile.platform and adjust it to your needs
# Run 'make PLATFORM=' to get an exhaustive list of possible parameters for this file.

PLATFORM=PM3OTHER
# If you want more than one PLATFORM_EXTRAS option, separate them by spaces:
#PLATFORM_EXTRAS=BTADDON
#STANDALONE=LF_SAMYRUN
➜  proxmark3 git:(master)
```
I then ran "make" like normal. I did miss the Makefile.platform the first time I compiled, but noticed the banner stating it was compiling for the PM3RD4. To fix it, I created the file mentioned above and ran "make clean" before compiling again.

The Proxmark was flashed using "pm3-flash-fullimage," but it complained about needing the bootrom updated, so I ran "pm3-flash-bootrom" and then "pm3-flash-fullimage" again to make sure everything was flashed correctly.

**To Reproduce**
Steps to reproduce the behavior:
1. Place this specific hotel Mifare card on the Proxmark3.
2. Run a nested attack against it.
3. Proxmark software returns "UART:: write time-out" and lights A, C, and D stay lit.

**Expected behavior**
The nested attack should function normally, or at least, shouldn't crash the Proxmark.

**Screenshots**
The first half of this log is a run-through of the nested attack with a different card that does not cause the issue (to prove normal functionality). The second half is the card that crashes the Proxmark.

See the log here: https://pastebin.com/raw/r0vZCgSe

**Desktop (please complete the following information):**
 - OS: Mac OS X and/or Ubuntu 18.04.3 LTS
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
```
[usb] pm3 --> hw ver

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman
  compiled with Clang/LLVM 4.2.1 Compatible Apple LLVM 10.0.1 (clang-1001.0.46.4) OS:OSX ARCH:x86_64

 [ PROXMARK RDV4 ]
  external flash:                  absent
  smartcard reader:                absent

 [ PROXMARK RDV4 Extras ]
  FPC USART for BT add-on support: absent

 [ ARM ]
  bootrom: RRG/Iceman/master/e5ffcfd5 2019-09-19 15:51:50
       os: RRG/Iceman/master/e5ffcfd5 2019-09-19 15:52:01
  compiled with GCC 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]

 [ FPGA ]
  LF image built for 2s30vq100 on 2019-07-31 at 15:57:16
  HF image built for 2s30vq100 on 2018-09-03 at 21:40:23

 [ Hardware ]
  --= uC: AT91SAM7S256 Rev D
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 256K bytes, Used: 242404 bytes (92%) Free: 19740 bytes ( 8%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

[usb] pm3 -->
```
 - hw status
```
[usb] pm3 --> hw status
#db# Memory
#db#   BIGBUF_SIZE.............40000
#db#   Available memory........40000
#db# Tracing
#db#   tracing ................1
#db#   traceLen ...............0
#db# Currently loaded FPGA image
#db#   mode.................... HF image built for 2s30vq100 on 2018-09-03 at 21:40:23
#db# LF Sampling config
#db#   [q] divisor.............95 ( 125 kHz )
#db#   [b] bps.................8
#db#   [d] decimation..........1
#db#   [a] averaging...........Yes
#db#   [t] trigger threshold...0
#db# LF T55XX config
#db#            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
#db#            mode            |start|write|write|write| read|write|write
#db#                            | gap | gap |  0  |  1  | gap |  2  |  3
#db# ---------------------------+-----+-----+-----+-----+-----+-----+------
#db# fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A |
#db#     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
#db#               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |
#db#    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 |
#db#
#db# Transfer Speed
#db#   Sending packets to client...
#db#   Time elapsed............500ms
#db#   Bytes transferred.......299008
#db#   Transfer Speed PM3 -> Client = 598016 bytes/s
#db# Various
#db#   DBGLEVEL................1
#db#   ToSendMax...............-1
#db#   ToSendBit...............0
#db#   ToSend BUFFERSIZE.......2308
#db#   Slow clock..............30962 Hz
#db# Installed StandAlone Mode
#db#   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[usb] pm3 -->
```
 - data tune
```
[usb] pm3 --> data tune

[=] Measuring antenna characteristics, please wait...

[=] You can cancel this operation by pressing the pm3 button
..

[+] LF antenna: 21.24 V - 125.00 kHz
[+] LF antenna: 19.12 V - 134.00 kHz
[+] LF optimal: 21.67 V - 126.32 kHz
[+] LF antenna is OK

[+] HF antenna: 23.08 V - 13.56 MHz
[+] HF antenna is OK

[+] Displaying LF tuning graph. Divisor 89 is 134kHz, 95 is 125kHz.

[usb] pm3 -->
```
**Additional context**
This error only happens with this one specific card out of 30+. Unfortunately, this is the only card from this specific hotel I have available to test with.

b4ndit, 2019-09-20:
I flashed my firmware back to the official repo (HEAD-d8ecc98) just now for testing purposes.

Here's what happens with the official firmware:
```
proxmark3> hf mf nested 1 8 A FFFFFFFFFFFF t
--nested. sectors:16, block no:  8, key type:A, eml:y, dmp=n checktimeout=471 us
Testing known keys. Sector count=16
nested...
-----------------------------------------------
Error: No response from Proxmark.

proxmark3>
```

The proxmark3 Easy has A, B, C, and D lights lit up after it crashes with the official firmware. B was not lit when it crashed on the Iceman fork.

iceman1001, 2019-09-21:
So,  fw ok,  antennas ok,  good.  
Now where is the output from your info and nested command?

```
hf 14a info
hf mf nested ....
hf 14a list
```

doegox, 2019-09-21:
hmm interesting but it will be hard to debug if we can't reproduce...

b4ndit, 2019-09-21:
I can run the commands Iceman suggested when I get back home today. Will "hf 14a list" work after the Proxmark crashes?

I can also connect to the Proxmark over JTAG if that'll help debug the crash at all.

How would you normally debug a crash on the Proxmark itself?

doegox, 2019-09-21:
I'm afraid `hf 14a list` won't help much but you can try.
To debug, first try to locate the position of the crash then the conditions of the crash.
For position, the simplest is the "printf" way, here adding a few "Dbprintf" statements in armsrc/mifarecmd.c MifareNested() and possibly some printf in client/mifare/mifarehost.c mfnested()

iceman1001, 2019-09-22:
If all issues was this good documented ..  I just found your log file. 
These cards,  which system are they for?  I noticed you found some keys.  Maybe a PR to the default keys dictionary? 

Back to issue,
- check keys -  it find the default keys,   transfers to emulator mem.
- nested - finds default keys,   enters nested,  get stuck.

Not running check keys before? ...  just speculation.
try targeting one specific block ?  
```
pm3 -->hf mf nestedt 1 8 a ffffffffffff 0 a
```

On what kind of computer are you running the pm3 client?

b4ndit, 2019-09-23:
The cards I have are from different hotels. I travel a lot, so I collect a lot of cards. I can certainly add all the found keys from the stack of cards I ran through autopwn the other day to a PR.

Here's some more info that may be helpful. I set the debug level to 3 before running the attack that fails and it gets stuck in an infinite loop on Nonce#2:
https://pastebin.com/raw/T4mKKUyw

Here is a log where I ran check keys before. Didn't turn on debug for this one, but same issue happens:
https://pastebin.com/raw/Ha0tBJsW

I'm running the same version of the client on both Mac OS X and Ubuntu. I switch between them depending on if I'm at work or at home. Both experience the same exact issue.

iceman1001, 2019-09-23:
with debug on 3,  no attacks will be able to perform since they are very time critical.

b4ndit, 2019-09-23:
I figured that may influence the results. However, the same commands worked fine on my other card with debug on 3, so I didn't think it would be an issue.

iceman1001, 2019-09-23:
if you only experienced it on OS X,  then it could have been the app nap funtionality..
But you also tested on Ubuntu...
```
disableAppNap("Proxmark3 polling UART");
```

iceman1001, 2019-09-23:
you could always send that one card to me,  if you could spare it...

b4ndit, 2019-09-23:
Yeah, that's definitely an option. I'd just have to figure out what the cost to ship would be. 

Another option I thought of was to hook the Proxmark up to a Raspberry Pi and give you remote access to tinker with it.

iceman1001, 2019-09-23:
remote access and the card on the antenna (distance 1cm about) would also work :)

b4ndit, 2019-09-25:
I should have it together by this weekend. I'll update here when I get it all setup and working.

iceman1001, 2019-10-02:
lets see.  @b4ndit  you got mail

iceman1001, 2019-10-03:
...chinese clone of a card with fixed nonce...  nested should fail.   the timeout message is however interesting.  I think it gets stuck in nonce collecting loop and because it has a fixed nonce it ```gets stuck```

b4ndit, 2019-10-03:
That makes sense to me. After looking at the code, it looks like it will never break out if it can't find a second nonce. Is there a way to detect fixed nonce cards?

A "hacky" fix would be to put a max loop counter in there to break out in case of this condition (or any other that triggers an infinite loop), but I doubt that's an optimal solution.

@iceman1001 what did you run to determine it was a fixed nonce card?

iceman1001, 2019-10-03:
just run two or more authentications and look at the trace data.... shouldnt repeat...:)

iceman1001, 2019-10-11:
Hm,  only time I can get this message, is if I have two terminals open.
A - runs proxmark3 client.
B - I forgot I have a client running against a device and trying to start the client against same device.

Which is kind of normal.  We don't support dual clients against one device over one serial.  

We do however support dual clients to one device but then use blueshark and usb.

We pushed a lot of fixes,  some was OSX related.  How about you pull latest, compile, flash and test if you get the same error still?

iceman1001, 2019-10-16:
Might aswell link to the other issue about fixed nonces.
https://github.com/RfidResearchGroup/proxmark3/issues/133

iceman1001, 2019-12-25:
I have pushed a detection for static / fixed nonces.

closed 2019-12-25 by iceman1001