RfidResearchGroup/proxmark3 #414: [idea] Integrate with LGTM Continuous security analysis

ViRb3, 2019-09-25:
I know there's Coverity integration already, but there's another new kid in the block called [LGTM](https://lgtm.com/). UI and setup seems much easier, could provide some interesting insights.

iceman1001, 2019-09-25:
The problem right now isn't more automated test tools... but people who can fix the already found bugs/errors/warnings...   
We can set it up for the fun of it.  Why not.

RfidResearchGroup, 2019-09-25:
Up and trying atleast...   It fails on ARM compilation.  Needs to be configured properly.  @merlokk ???
[image]

RfidResearchGroup, 2019-09-25:
https://lgtm.com/projects/g/RfidResearchGroup/proxmark3/

RfidResearchGroup, 2019-09-25:
I guess we need a LGTM yml configuration file
https://lgtm.com/help/lgtm/lgtm.yml-configuration-file

RfidResearchGroup, 2019-09-25:
Build env is docker based Ubuntu.   
https://lgtm.com/help/lgtm/analysis-faqs#whats-the-default-build-environment-for-cpp-projects

merlokk, 2019-09-26:
looks like it cant compile arm-none-eabi standard library.

iceman1001, 2019-10-02:
If @ViRb3  or @merlokk  is not able to configure it properly, I fear that we might just aswell drop this idea.   Whats your comment?

doegox, 2019-10-04:
if the problem is the arm-none-eabi, it's no different than Coverity which also scans only the client part.
I'm still very interested to see how both compare on the client part.

iceman1001, 2019-10-04:
No,  Coverity does scan arm.
[image]

doegox, 2019-10-04:
> No, Coverity does scan arm.

With zero line of code scanned, this is fine :D 
See on the dashboard "Files in latest snapshot", there are only /client/ files

[image]

iceman1001, 2019-10-04:
Coverity did something this spring,  they changed system or was bought up,  and since then nothing works.  Will look in to it when I get back from work.

iceman1001, 2019-10-04:
This LGTM did run on the files it found. ie python files.  It didn't like it.

iceman1001, 2020-01-14:
closed due to lack of activity

closed 2020-01-14 by iceman1001