RfidResearchGroup/proxmark3 #417: Allow Writing of Kastle HID Prox Cards

swg0101, 2019-10-02:
The official firmware allows a format to be specified when cloning HID to a T55xx card. However, no such format exists for Kastle cards on the RRG fork. While Kastle cards are recognized on an lf search, there is no way to write it back onto the card (unless perhaps the data is manually written on the card via lf t55xx write).

I am referring to the equivalent of:
lf hid write Kastle f x c xxxxx i xx

Thanks!

iceman1001, 2019-10-02:
Feel free to add it!

swg0101, 2019-10-02:
Today is the first day I had my hands on my Proxmark3, so I would appreciate if someone can chime in and make this happen... :)

swg0101, 2019-10-07:
Created PR here: https://github.com/RfidResearchGroup/proxmark3/pull/423
Looks like the only thing missing with the high bit, and lf hid clone worked perfectly after that.

iceman1001, 2019-10-08:
Good catch!  And your first PR :)

closed 2019-10-08 by iceman1001