RfidResearchGroup/proxmark3 #418: Password checking for T55xx detect and writes

swg0101, 2019-10-02:
Currently, it looks like the T55xx password is only checked on reads, but not detects or writes. However, manually specifying a password for these commands will still perma-brick a card or tag if no password is currently set.

Would it make sense for the client to disallow these commands with a password (unless forced) if a "t55xx detect" command succeeds without a password before conducting the actual commands?

This should stop more people from bricking their T55xx cards (which I learned the hard way).

Thanks!

mwalker33, 2019-10-02:
Not sure how a password protected write (when no password is set) can result in a "bricked" card.
At best it would set the lock bit (and stop you from changing that block), even then i cant see how it would do that as a password protected write can only be seen as a password protected write; which should result in a command sequence error (too many bits).
If you make a password protected read when the password is not set, this results in the command looking like a valid non-password write, which can set the lock bit (as the bit count is the same).

Do you have an example of a command that bricked the card and the exact card/chip type (e.g.. t5555, t5557, t5577) ?  I would like to re-read the specific data sheet.
 
Note: On many test runs of have sent password protected writes to a non password protected t5577  If the command was to be interpreted as a non password write it should have locked the block (as the Lock bit would have been set (1) based on the password i use in testing.  This has never happened.

swg0101, 2019-10-02:
The cards I had were T5577, and if I remember correctly I did:
lf hid clone 2
lf t5 wr b 0 d 0 p feedbeef

and the card bricked after that and won't respond to any commands.

iceman1001, 2019-10-02:
pwd checks for write is as @mwalker33 says,  not needed.
For t55xxDetect is makes sense to add the same checks as t55xxread has.

swg0101, 2019-10-02:
> That would write an invalid config to the card. Try the lf t55 wipe command and see if it comes back to life.

Nope (usually that's the first thing I try - along with writing something easily searchable like lf hid clone 2)... even a lf search u returns just noise. It could be that I have also mixed in a detect with password in my testing that broke the tags.

mwalker33, 2019-10-02:
Just did some tests on a t5577
the write with password when no password was set worked as a normal write command.  
i.e. the data was written and the password ignored (as expected)

The lf t55 detect  with password will have some undesired results depending on the password used.
e.g. if the most significant bit is set 0x80000000 then that will set the lock bit on block 0 and the card will not detect.  this can normally be recovered (for the t5577) with test mode.
    lf t55 write b 0 d 000880E0 t

In theory the correct pattern with the lock bit set and bit 15 set would then disable test mode and then brick the card
e.g. I would expect this to be really bad if used as a password on a read of block 0
0xB0010000 (1 0110000 00000001 00000000 00000000 x)
i.e. set the lock bit then config of 0x60020000

from the data sheet
Note: 1. If Master Key is 6 and bit15 is set, the test mode access is disabled and the extended mode is active.

if the most significant bit is not set, this results in an invalid config that can be recovered with 
lf t55 wipe
or
lf t55 write b 0 d 000880E0

Key note:  some clone t5577 are not t5577 (I am calling them t5200) and do NOT support the test mode.  While these look the same, the don't support block 3 page 1 config either.  With my quick test these do become locked/bricked.

In summary
Try the test mode to see if you can recover.
Yes, some checks on the detect could help stop or slow down this issue.

Some feedback needed here.
Idea:
regardless of if a password is supplied or not, 
Try to detect without a password - If that works, exit without using the password.
Then, try to detect with the password IF supplied with the "o" option (? not sure if we would still need the o option if we have already tried and failed without the password).
(follow on from this), we could set a flag if the detect needed a password or not, then (ignore) the password for any read while the "don't need password" flag is set.

swg0101, 2019-10-02:
@mwalker33 - Thanks for being so thorough and the explanations. I will try to fetch the cards from the trash and see if I can recover them.

swg0101, 2019-10-03:
Just tried both
lf t55 write b 0 d 000880E0 t
and
lf t55 write b 0 d 000880E0

Both did not work to revive the card and will not accept writes. On lg sea u, I can see something like:

> [-] no repeating pattern found, try increasing window size
> 01100110011001100110011001100110
> 01100111010101010101010101010101
> 01010101010101010101010101010101
> 01010101010101010101010101010101
> 010101010..101111001100110011001
> 10011001100110011001100110011001
> 10011001100110011001100110011001
> 10011001110101010101010101010101
> 01010101010101010101010101010101
> 01010101.011010010

> [-] no repeating pattern found, try increasing window size
> 01100110011001100110011001100110
> 01100111010101010101010101010101
> 01010101010101010101010101010101
> 01010101010101010101010101010101
> 010101010..101111001100110011001
> 10011001100110011001100110011001
> 10011001100110011001100110011001
> 10011001110101010101010101010101
> 01010101010101010101010101010101
> 010101010.110100101

> [-] no repeating pattern found, try increasing window size
> ................................
> ................................
> ................................
> ................................
> ................................
> ................................
> ................................
> ..........0101010

mwalker33, 2019-10-03:
I will try to make this the short version.
The t55xx will spit out the data from block 1 to block x encoded as per the config in block 0 page 0
So the fact that you got some data out indicates that the block 0 config is at least partly valid (even if you cant change it)

Can you test something for me.
On a blank card (that is known to be working and you can write to).
lf t55 wipe
lf t55 det
.. should show a clean detect
lf t55 write b 3 1 d 90000800
lf t55 det
(it should now show the card in "leading 0 reference" for the downlink mode.)
if not in leading 0 reference, 
    lf t55 dump
  
lf t55 wipe
(should put it back to defaults)

If it does show the leading 0, then its should be a real t5577
if it does not show leading 0 mode then its not a real t5577
(most of the cards and fobs I got from the cheap Chinese sells were NOT real t5577 cards)
As per my prev. post, the test command only works for a valid card T5577 and not the cheaper clones.

Can you post the output of the commands.

Thanks

swg0101, 2019-10-03:
This was the blank white tag (without holes) I was using:

> [usb] pm3 --> lf t55 wipe
> 
> [=] Begin wiping T55x7 tag
> 
> [=] Default configation block 000880E0
> [=] Writing page 0  block: 00  data: 0x000880E0
> [=] Writing page 0  block: 01  data: 0x00000000
> [=] Writing page 0  block: 02  data: 0x00000000
> [=] Writing page 0  block: 03  data: 0x00000000
> [=] Writing page 0  block: 04  data: 0x00000000
> [=] Writing page 0  block: 05  data: 0x00000000
> [=] Writing page 0  block: 06  data: 0x00000000
> [=] Writing page 0  block: 07  data: 0x00000000
> [usb] pm3 --> lf t55 det
>     Chip Type      : T55x7
>     Modulation     : ASK
>     Bit Rate       : 2 - RF/32
>     Inverted       : No
>     Offset         : 33
>     Seq. Term.     : Yes
>     Block0         : 0x000880E0
>     Downlink Mode  : default/fixed bit length
> 
> [usb] pm3 --> lf t55 write b 3 1 d 90000800
> [=] Writing page 1  block: 03  data: 0x90000800
> [usb] pm3 --> lf t55 det
>     Chip Type      : T55x7
>     Modulation     : ASK
>     Bit Rate       : 2 - RF/32
>     Inverted       : No
>     Offset         : 33
>     Seq. Term.     : Yes
>     Block0         : 0x000880E0
>     Downlink Mode  : default/fixed bit length
> 
> [usb] pm3 --> lf t55 dump
> [+] Reading Page 0:
> [+] blk | hex data | binary                           | ascii
> [+] ----+----------+----------------------------------+-------
> [+]  00 | 000880E0 | 00000000000010001000000011100000 | ....
> [+]  01 | FFFFFFFF | 11111111111111111111111111111111 | ....
> [+]  02 | FFFFFFFF | 11111111111111111111111111111111 | ....
> [+]  03 | FFFFFFFF | 11111111111111111111111111111111 | ....
> [+]  04 | FFFFFFFF | 11111111111111111111111111111111 | ....
> [+]  05 | FFFFFFFF | 11111111111111111111111111111111 | ....
> [+]  06 | FFFFFFFF | 11111111111111111111111111111111 | ....
> [+]  07 | FFFFFFFF | 11111111111111111111111111111111 | ....
> [+] Reading Page 1:
> [+] blk | hex data | binary                           | ascii
> [+] ----+----------+----------------------------------+-------
> [+]  00 | 000880E0 | 00000000000010001000000011100000 | ....
> [+]  01 | C02A4807 | 11000000001010100100100000000111 | .*H.
> [+]  02 | 2A387C79 | 00101010001110000111110001111001 | *8|y
> [+]  03 | 20001001 | 00100000000000000001000000000001 |  ...
> [usb] pm3 -->
> [usb] pm3 --> lf t55 wipe
> 
> [=] Begin wiping T55x7 tag
> 
> [=] Default configation block 000880E0
> [=] Writing page 0  block: 00  data: 0x000880E0
> [=] Writing page 0  block: 01  data: 0x00000000
> [=] Writing page 0  block: 02  data: 0x00000000
> [=] Writing page 0  block: 03  data: 0x00000000
> [=] Writing page 0  block: 04  data: 0x00000000
> [=] Writing page 0  block: 05  data: 0x00000000
> [=] Writing page 0  block: 06  data: 0x00000000
> [=] Writing page 0  block: 07  data: 0x00000000
[... 123 lines cut ...]
> [+] Reading Page 1:
> [+] blk | hex data | binary                           | ascii
> [+] ----+----------+----------------------------------+-------
> [+]  00 | 000880E0 | 00000000000010001000000011100000 | ....
> [+]  01 | FFFFFFFF | 11111111111111111111111111111111 | ....
> [+]  02 | FFFFFFFF | 11111111111111111111111111111111 | ....
> [+]  03 | 90000800 | 10010000000000000000100000000000 | ....
> [usb] pm3 -->
> [usb] pm3 --> lf t55 wipe
> 
> [=] Begin wiping T55x7 tag
> 
> [=] Default configation block 000880E0
> [=] Writing page 0  block: 00  data: 0x000880E0
> [=] Writing page 0  block: 01  data: 0x00000000
> [=] Writing page 0  block: 02  data: 0x00000000
> [=] Writing page 0  block: 03  data: 0x00000000
> [=] Writing page 0  block: 04  data: 0x00000000
> [=] Writing page 0  block: 05  data: 0x00000000
> [=] Writing page 0  block: 06  data: 0x00000000
> [=] Writing page 0  block: 07  data: 0x00000000
> [=] Writing page 1  block: 03  data: 0x00000000
> [usb] pm3 -->

It looks like your theory may be correct that the counterfeits may not have T5577 chips in them but a copycat chip instead.

iceman1001, 2019-10-03:
Good thing OP recovered his fobs/tags. Keysy sets password on their tags,  annoying.
Good thing it can be reset with ```t testmode```

@mwalker33 
didn't you say you will do a printout for Page1 Block3 ... like the info command
```lf t55 write b 3 1 d 90000800```

swg0101, 2019-10-03:
@iceman1001 I was only able to recover the Keysy tag. All the other tags/cards are supposedly Chinese ripoffs and they are perma-bricked. It looks like the Keysy tags are the only ones I have seen so far with genuine T5577 chips inside.

iceman1001, 2019-10-03:
If that is your latest output, I don't understand why you say they are perma-bricked.  They look like a empty tag after a wipe.

mwalker33, 2019-10-03:
I think that was the output from the tests to see if the working cards were real t5577 thus support test mode or the clone that looks the same but does not support test mode or block 3 page 1 config.  I was getting him to use the downlink mode to test them.

iceman1001, 2019-10-03:
aha.. working cards.   I see.

swg0101, 2019-10-06:
@mwalker33 Did you find any reputable Chinese sellers selling real T5577s?

iceman1001, 2019-10-06:
This the the repo where we talk about the issue at hand, 
If you have questions about sellers, I suggest you go look at the trade section on the pm3 forum.

swg0101, 2019-10-06:
Ok, thanks.

mwalker33, 2019-10-06:
Back to the challenge at hand.

Using a password on detect when no password was set.

My idea here was to have the detect try without a password first, then if a password was supplied AND card was not detected without one, try with the password.

This way the lf t55xx detect should work without risk and without needing the o option.
It also means there should be little to no risk with the password read when no password is set (as it tried without a password first).  

We could also add the o option to FORCE the password use for performance reasons, but this will mean if a user uses the history of the last detect that needed a password, we still a have a risk.

i.e. 
lf t55xx detect 
check all 4 downlink modes without a password (current behavior).

lf t55xx detect p 12345678
check all 4 downlink modes without a password, then if still not found, check all 4 downlink modes with the password supplied (low risk password detect).

lf t55xx detect p 12345678 o 
Check all 4 downlink modes with the password supplied (higher risk, but user knows best ) .

Note: with things like the brute force/chk calls, we could add a detect at the start and only proceed with the password checks if the detect failed.

thoughts ?

iceman1001, 2019-10-07:
The normal case 1 and 2 is good easy end user experience.  I agree that case2 should work as @mwalker33  suggest.  I say, skip the "o" option (case3).  No need to make it more complicated.  
We can not make the proxmark3 idiotsafe.  If ppl use the device not knowing what they are doing and doesn't read warning message printed,  then that is that. 

We can add more warning messages and use colors to enhance the message.

The idea with bruteforce / chk is that the tag is pwd protected which means it doesn't answer without a pwd...  Pointless to add a detect.

doegox, 2019-10-07:
> The idea with bruteforce / chk is that the tag is pwd protected which means it doesn't answer without a pwd... Pointless to add a detect.

imho there is currently a risk because ppl run commands also just to see or get help and so far, `lf t55 brute` immediately starts bruteforcing.
I'd prefer this command to run only if start and end are give, starting from 00000000 to FFFFFFFF is pointless anyway, so ppl running without arg would see the help, preferably with a big warning to not run against unprotected cards.

`lf t55 chk` requires at least `m` to start so it's fine, would be better to display help instead of doing nothing.

doegox, 2019-10-07:
also please document in the help which default value for `r` when `r` is not explicitly set

swg0101, 2019-10-07:
@doegox That's exactly how I broke one of my cards... 😆

mwalker33, 2019-10-07:
> also please document in the help which default value for r when r is not explicitly set
Yep.  it got a little tricky as the default is not always fixed.
ie: detect r = 4 is the default
chk/brute  r = 0 is the default
other commands the default is the value found in the detect (e.g. if  1 or 4 downlink was found then r = 3 becomes the default for read/write operations.
I will tweak the display function to reflect this.

doegox, 2019-10-07:
about detecting clone vs atmel, a quick way is also to try to modify page1 block1 or block2 (traceability data).
they're RW on the clone and RO on Atmel.

doegox, 2019-10-07:
btw when executing on a clone
```
lf t55 write b 3 1 d 90000800
lf t55 det # no  "leading 0 reference" , it's a clone
lf t55 wipe
lf t55 dump
```
block3 p1 is still 90000800 and not reset.
But on Atmel tag, the wipe resets properly block3 p1

Why lf t55 wipe does not always reset the block3 p1 ?

doegox, 2019-10-07:
sidenote: block3 on my clones was always set to 00A00003 which doesn't make sense according to the datasheet, but maybe nothing is taken into account anyway regarding analog frontend on these clones

iceman1001, 2019-10-07:
,... I think this is another issue / subject to have in a seperate issue.

mwalker33, 2019-10-07:
While I agree this is a bigger challenge, a quick response

As to the 00A00003, I think that may be the t5200 clone of the t5577.  i.e. try changing the downlink mode and test, if the mode does not change, then not a t5577.
lf t55 write b 3 1 d 90000800
lf t55 det
(it should now show the card in "leading 0 reference" for the downlink mode.)

On all my "original" T5577 I can write the block 1 and 2 on page 1.  
keep in mind that the test write will reset the lock bits, so even if the block 1 and 2 on page 1 are locked by atmel, the test write clears that and make them read/writable.  (my view was that if the chip supports ALL the functions of the T5577 then its as good as a T5577.  We only really need to know if we need to do something different with the Chip. 

With the addition to the wipe command, I only wrote to block 3 page 1 IF the detected downlink mode was NOT the chip default (fixed bit).  i.e. reset back to fixed bit.

iceman1001, 2019-10-07:
I think it would be a good test to have ``trace``` or ```info``` command this  P1 -Block3 checks.

iceman1001, 2019-10-08:
And with @mwalker33  last PR its time to close this issue.
 @doegox  I suggest starting another issue for the Page1 / Block3 ideas

closed 2019-10-08 by iceman1001