RfidResearchGroup/proxmark3 #419: Color on windows 10

mwalker33, 2019-10-05:
**Describe the solution you'd like** 
Enable ansi color if a users windows can support it.

**Additional context**
I have the latest version of windows : Microsoft Windows [Version 10.0.18362.388]

In the proxmark3.c file (around line 684)
If I make this change     : session.supports_colors = true; 

Then colors seem to work (i.e. on my windows ansi colors will work if enabled)
To enable, a registry key needs to be set (for Windows 10.0.16257 and later)

[HKEY_CURRENT_USER\Console]
"VirtualTerminalLevel"=dword:00000001

Note: if the above key entry is not in the registry or is set to 0, then the ANSI colors don't print and session.supports_colors will still need to be set to false to ensure a clean output.

I have some test code working that will check the registry to see if the key is set to 1, and if so ,set session.supports_colors = true, else leave at false.

If we think this is a good idea, I will submit a PR for review and comments.

doegox, 2019-10-05:
Nice ! 
If registry test can be done in a reliable way without hurting the other platforms, why not.

iceman1001, 2019-10-05:
if we can easily access the registry under windows.  Go go go!

mwalker33, 2019-10-06:
If your happy, we can close this now.

closed 2019-10-06 by doegox