RfidResearchGroup/proxmark3 #421: Issue with `hf mf autopwn`

bruno-, 2019-10-07:
**Describe the bug**

Hi,

thank you for all the good work on proxmark3 software.

I can't get `hf mf autopwn` to dump a particular mifare classic 1k card. These are the approaches I tried:

1. `hf mf autopwn * 1 f mfc_default_keys`, full command output: 
[tmux-proxmark-2-0-20191007T140338.log](https://github.com/RfidResearchGroup/proxmark3/files/3697155/tmux-proxmark-2-0-20191007T140338.log)

2. `hf mf autopwn k 0 A a0a1a2a3a4a5`, full command output: [tmux-proxmark-2-0-20191007T134132.log](https://github.com/RfidResearchGroup/proxmark3/files/3697145/tmux-proxmark-2-0-20191007T134132.log)

The result looks ok for a while, but after some time I'm just getting this forever `Apply bit flip properties                               |             nan |  nand`.

I was able to dump this card with official Proxmark3 repo, so I'm pretty confident the card can be dumped. These are the official pm3 repo commands I used:

- `hf mf chk *1 ? t`
- `hf mf hardnested ....`
- `hf mf ekeyprn`
- `hf mf dump`

**To Reproduce**

- `hf search` outputs mifare classic card 1k, hard
- `hf mf autopwn * 1 f mfc_default_keys` this works for a while
- `Apply bit flip properties                               |             nan |  nand` I start getting this and it seems there's no progress... I'm not sure if that's an error, but `nan` and `nand` are suspicious.

**Expected behavior**

I expected this command `hf mf autopwn * 1 f mfc_default_keys` to finish in reasonable amount of time like 10-20 minutes. I expected to get card dump and these files in the repo: `hf-mf-<uid>-data.bin`, `hf-mf-<uid>-data.eml`, `hf-mf-<uid>-data.json`, `hf-mf-<uid>-key.bin`. The autopwn command never finished and these files are not present in the repo.

**Screenshots**

I hope these log files from the terminal will be helpful (it's the same files pasted above):

1. `hf mf autopwn * 1 f mfc_default_keys`, full command output: 
[tmux-proxmark-2-0-20191007T140338.log](https://github.com/RfidResearchGroup/proxmark3/files/3697155/tmux-proxmark-2-0-20191007T140338.log)

2. `hf mf autopwn k 0 A a0a1a2a3a4a5`, full command output: [tmux-proxmark-2-0-20191007T134132.log](https://github.com/RfidResearchGroup/proxmark3/files/3697145/tmux-proxmark-2-0-20191007T134132.log)

**Desktop (please complete the following information):**
- MacOS High Sierra, Version 10.13.6.

 - hw version

```log

 [ Proxmark3 RFID instrument ] 
          

 [ CLIENT ]          
  client: RRG/Iceman          
  compiled with Clang/LLVM 4.2.1 Compatible Apple LLVM 10.0.0 (clang-1000.10.44.4) OS:OSX ARCH:x86_64          

 [ PROXMARK3 RDV4 ]          
  external flash:                  present           
  smartcard reader:                present           

 [ PROXMARK3 RDV4 Extras ]          
  FPC USART for BT add-on support: absent           
          
 [ ARM ]
  bootrom: RRG/Iceman/master/78c153fe 2019-10-06 01:13:55
       os: RRG/Iceman/master/78c153fe 2019-10-07 12:36:17
  compiled with GCC 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]

 [ FPGA ]
  LF image built for 2s30vq100 on 2019-07-31 at 15:57:16
  HF image built for 2s30vq100 on 2018-09-03 at 21:40:23          

 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev B          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 274819 bytes (52%) Free: 249469 bytes (48%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory          

          
[usb] pm3 --> 
```
 - hw status

```
[usb] pm3 --> hw status
#db# Memory           
#db#   BIGBUF_SIZE.............40000          
#db#   Available memory........40000          
#db# Tracing           
#db#   tracing ................0          
#db#   traceLen ...............0          
#db# Currently loaded FPGA image           
#db#   mode.................... HF image built for 2s30vq100 on 2018-09-03 at 21:40:23          
#db# Flash memory           
#db#   Baudrate................24 MHz           
#db#   Init....................OK           
#db#   Memory size.............2 mbits / 256 kb           
#db#   Unique ID...............0xD567A882A76AB225          
#db# Smart card module (ISO 7816)           
#db#   version.................v3.11           
#db# LF Sampling config           
#db#   [q] divisor.............95 ( 125.00 kHz )          
#db#   [b] bps.................8          
#db#   [d] decimation..........1          
#db#   [a] averaging...........Yes          
#db#   [t] trigger threshold...0          
#db#   [s] samples to skip.....0           
#db# LF T55XX config           
#db#            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]          
#db#            mode            |start|write|write|write| read|write|write          
#db#                            | gap | gap |  0  |  1  | gap |  2  |  3          
#db# ---------------------------+-----+-----+-----+-----+-----+-----+------          
#db# fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A |           
#db#     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |           
#db#               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |           
#db#    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 |           
#db#           
#db# Transfer Speed           
#db#   Sending packets to client...          
#db#   Time elapsed............500ms          
#db#   Bytes transferred.......341504          
#db#   Transfer Speed PM3 -> Client = 683008 bytes/s          
#db# Various           
#db#   DBGLEVEL................1          
#db#   ToSendMax...............12          
#db#   ToSendBit...............8          
#db#   ToSend BUFFERSIZE.......2308          
#db#   Slow clock..............31920 Hz          
#db# Installed StandAlone Mode           
#db#   No standalone mode present          
#db# Flash memory dictionary loaded           
#db#   Mifare..................859 keys          
#db#   T55x7...................109 keys          
#db#   iClass..................7 keys
```

 - data tune

```
[usb] pm3 --> data tune
          
[=] Measuring antenna characteristics, please wait...
          
[=] You can cancel this operation by pressing the pm3 button          
..
          
[+] LF antenna: 71.10 V - 125.00 kHz          
[+] LF antenna: 39.23 V - 133.33 kHz          
[+] LF optimal: 71.10 V - 125.00 kHz          
[+] LF antenna is OK  
          
[+] HF antenna: 44.04 V - 13.56 MHz          
[+] HF antenna is OK           
          
[+] Displaying LF tuning graph. Divisor 89 is 133.33 kHz, 95 is 125.00 kHz.
          
No GUI in this build!
[usb] pm3 --> 
```

**Additional context**

Thank you for checking this issue, and let me know if you need any additional info.

iceman1001, 2019-10-07:
Those logfiles didn't make it.  Try again to link them

bruno-, 2019-10-07:
I updated the first comment to include pastebin links to relevant files. Thank you for checking the issue this fast.

rdnxl, 2019-10-07:
I'm having the same issue. After compiling and testing with various commits I figured out that the issue was introduced in [this commit](https://github.com/RfidResearchGroup/proxmark3/commit/93915858eacf3bd8b31f87bab9180752d6751f5c). When I compile an older commit it goes back to normal.

iceman1001, 2019-10-08:
@rdnxl  that commit was overruled by later changes.  You would need to find a later point.

rdnxl, 2019-10-08:
I also tried many later points, and they also didn't work. I also noticed how I got "Using 0 precalculated bitflip state tables" on that commit and on newer ones I got "Using 13 precalculated bitflip state tables" while on the older where it does work I got more than 200 precalculated bitflip state tables. Maybe this has something to do with it? Any commit after [this one](https://github.com/RfidResearchGroup/proxmark3/commit/e7d67fc2b0d09fcd5a44aefa40c15dacdeb23453) gets me 13 in the output.
Anyway, thanks for taking a look at this, I've been scratching my head trying to get hardnested working last two weeks.

iceman1001, 2019-10-08:
Its the new detection of  directory vs filename thats messing.
ie,  the searchfile function shouldn't "search" if user entered a foldername instead of filename.

Problem is that you get part right like 13...

iceman1001, 2019-10-08:
@rdnxl   would you mind running
```
data setd 2
hf mf hardnested t
```

rdnxl, 2019-10-08:
I'd be happy to give you the output. I can get it done later today.

rdnxl, 2019-10-08:
@iceman1001 looks like your changes today resolved the issue for me, thanks!

I was not able to get output for `hf mf hardnested t` I'm not sure how the 'tests?' option is supposed to be used. Whenever I added that option I would get the help text as output or it would request for a key to be given.

iceman1001, 2019-10-08:
So it works on OS X again.  Good,  then I will close this issue.

closed 2019-10-08 by iceman1001