RfidResearchGroup/proxmark3 #427: Password validation fails when changing T5577 passwords

swg0101, 2019-10-08:
When changing a password using `lf t5 pr p [oldpass] n [newpass]` format, the password validation fails even though the new password is successfully written. Not sure if this is just more of a comestic thing, although the password seems to work just fine.

> [usb] pm3 --> lf t5 pr p 51243648 n DEADBEEF
> [=] Checking current configuration          
> [=] Safety check overridden - proceeding despite risk          
> [=] PWD bit is already set          
> [+] Wrote new password          
> [=] Safety check overridden - proceeding despite risk          
> [!] Failed to validate the password write. aborting.          
> [!] Command failed. Did you run `lf t55xx detect` before?          
> [usb] pm3 --> lf t5 det p DEADBEEF
>     Chip Type      : T55x7          
>     Modulation     : FSK2a          
>     Bit Rate       : 4 - RF/50           
>     Inverted       : Yes           
>     Offset         : 32          
>     Seq. Term.     : No          
>     Block0         : 0x00107070          
>     Downlink Mode  : default/fixed bit length          
>     Password Set   : Yes           
>     Password       : DEADBEEF

mwalker33, 2019-10-08:
did the lf t55 detect work just prior to the lf t55 protect call ?

swg0101, 2019-10-08:
Yep... I have always just disregarded that error since the password change seems to be successful.

mwalker33, 2019-10-08:
So it was happening before my detect update ?

swg0101, 2019-10-08:
Yep.

mwalker33, 2019-10-08:
Can we get a log of everything in sequence.  e.g. lf t55 det, changes, set password then change password.
I cant seem to make the "change password" reproduce it.
e.g.
on my test
```
[usb] pm3 --> lf t55 pro p 00001111 n 00002222
[=] Checking current configuration
[=] Safety check overridden - proceeding despite risk
[=] PWD bit is already set
[+] Wrote new password
[=] Safety check overridden - proceeding despite risk
[+] Validated new password
[+] Wrote modified configuration block
[!] Safety check: Could not detect if PWD bit is set in config block. Exits.
[=] Block0 write detected, running `detect` to see if validation is possible
    Chip Type      : T55x7
    Modulation     : FSK2a
    Bit Rate       : 4 - RF/50
    Inverted       : Yes
    Offset         : 33
    Seq. Term.     : No
    Block0         : 0x00107070
    Downlink Mode  : default/fixed bit length
    Password Set   : Yes
    Password       : 00002222

[+] New configuration block 00107070 password 00002222
[+] Success, tag is locked
```

swg0101, 2019-10-08:
Don't remember the last sequence, but this one does it:

```[usb] pm3 --> lf t5 chk i t55xx_default_pwds
[+] loaded 109 keys from dictionary file /usr/local/bin/../share/proxmark3/dictionaries/t55xx_default_pwds.dic           
[=] Testing 51243648          
    Chip Type      : T55x7          
    Modulation     : FSK2a          
    Bit Rate       : 4 - RF/50           
    Inverted       : Yes           
    Offset         : 32          
    Seq. Term.     : No          
    Block0         : 0x00107070          
    Downlink Mode  : default/fixed bit length          
    Password Set   : No           
          
[+] Found valid password: [ 51243648 ]          
          
[+] Time in check pwd: 0 seconds
          
[usb] pm3 --> lf t5 pr
This command set or unsets the pwd bit on T5577.          
Usage:  lf t55xx protect [r <mode>] [p <password>] [o] [n <new_password>]          
Options:          
     p <password>        - OPTIONAL password (8 hex characters)          
     o                   - OPTIONAL override safety check          
     n <new password>    - new password          
     r <mode>     - downlink encoding 0|1|2|3          
                       0 - fixed bit length (detected default)          
                       1 - long leading reference          
                       2 - leading zero          
                       3 - 1 of 4 coding reference          
          
Examples:          
      lf t55xx protect n 01020304         - sets new password to 01020304          
      lf t55xx protect p 11223344         - use pwd 11223344 to set newpwd to 00000000          
          
[usb] pm3 --> lf t5 pr p 51243648 n DEADBEEF
[=] Checking current configuration          
[=] Safety check overridden - proceeding despite risk          
[=] PWD bit is already set          
[+] Wrote new password          
[=] Safety check overridden - proceeding despite risk          
[!] Failed to validate the password write. aborting.          
[!] Command failed. Did you run `lf t55xx detect` before?          
[usb] pm3 --> lf t5 det p DEADBEEF
    Chip Type      : T55x7          
    Modulation     : FSK2a          
    Bit Rate       : 4 - RF/50           
    Inverted       : Yes           
    Offset         : 32          
    Seq. Term.     : No          
    Block0         : 0x00107070          
    Downlink Mode  : default/fixed bit length          
    Password Set   : Yes           
    Password       : DEADBEEF          
          
[usb] pm3 --> q```

iceman1001, 2019-10-08:
...but as @mwalker33  says,  you don't call detect between chk pwd and protect...

swg0101, 2019-10-08:
Let me try that again and see...

swg0101, 2019-10-09:
Looks like you might be right... I do think it's a bit strange for the user to have to run a detect against the old password to change to a new password since the protect command has both passwords... It probably would be better if the tool just did a detect by itself for user-friendliness.

As an aside, the tooltip says `This command set or unsets the pwd bit on T5577.`, but I am not quite sure how you would "unset" the password since doing a blank password simply gets the password to 0 rather than unprotecting the tag. Maybe a bug?

iceman1001, 2019-10-09:
and that would be because you don't know what you are doing or don't understand the concepts behind what you are trying to do.   Which leads to your failures. 

Read the T5577 datasheet and you start to get an understanding on what you are playing with.
Then I suggest you read on the Proxmark3 forum for more specific subjects relating to your troubles.

Since GH is where we discuss bugs regarding the source code,   and forum is where we find answers and discussions,  I will close this issue.

closed 2019-10-09 by iceman1001

swg0101, 2019-10-09:
What I was suggesting was that since both passwords are provided as arguments for the protect command, wouldn't it be easier for the code to run the detect against the old password given by the user now that it is safer against non-protected cards?

The second part I still think is a bug since providing p [pwd] and n with no arguments will just set the password to 0x00000000 with the PWD bit still set. I am not seeing any way to really "unset" the bit on the protect command. Instead, I have to do `lf t55xx write b 0 d 000880e0 p [pwd]` followed by a wipe if I want to clean off the card.

iceman1001, 2019-10-09:
See my previous comment.
The protect command doesn't do "unprotect".  As seen in the commit message.
The bug you are looking for is only textual. I removed the offending text of "unset".  

```
CHG: 'lf t55xx wipe c' - now accepts user provided configuration block to configure a wiped tag with.

ADD: 'lf t55xx protect' -  Sets password and enables password protection in configuration block.   It tries to validate its writes aswell.

Samples:
lf t55xx protect n 11223344            - set pwd to 11223344
lf t55xx protect n 01020304 p 11223344 - change a password protected tag from 11223344 to 01020304

To remove the password protection, you need to remember the password :)
lf t55xx wipe p 11223344               - wipes whole tag, removes password protection bits.
````

swg0101, 2019-10-09:
Thanks, I didn't even notice that wipe now supports password on it.
That said, now there is another textual change you may want to make from `This command set the` to `This command sets the`... just saying.. ;)

iceman1001, 2019-10-09:
feel free to contribute!