RfidResearchGroup/proxmark3 #428: hf iclass eload

sherhannn79, 2019-10-09:
**Describe the bug**
After executing the specified command, the proxmark3 interrupt the work.

[image]

iceman1001, 2019-10-09:
Thats a large iclass_tagdump.bin file.  Doesn't seem like a normal sized one.
How did you generate it?  Try  ```hf iclass dump``` to create a dump file and try it with eload.

iceman1001, 2019-10-09:
I tried with the dump file that we have in repo:
 /client/resources/iclass_dump.bin

Works very well still.  What is your output from 

```
hw status
hw version
```

[image]

sherhannn79, 2019-10-09:
> Thats a large iclass_tagdump.bin file. Doesn't seem like a normal sized one.
Why? My iclass tag has 256 blocks of 8 bytes. 256*8 = 2048. Is not it?

sherhannn79, 2019-10-09:
> What is your output from
> 
> ```
> hw status
> hw version
> ```
> 

I can provide tomorrow.

sherhannn79, 2019-10-10:
> hw status
> hw version

[usb] pm3 --> hw status
#db# Memory
#db#   BIGBUF_SIZE.............40000
#db#   Available memory........40000
#db# Tracing
#db#   tracing ................1
#db#   traceLen ...............0
#db# Currently loaded FPGA image
#db#   mode.................... HF image built for 2s30vq100 on 2018-09-03 at 21:40:23
#db# LF Sampling config
#db#   [q] divisor.............95 ( 125.00 kHz )
#db#   [b] bps.................8
#db#   [d] decimation..........1
#db#   [a] averaging...........Yes
#db#   [t] trigger threshold...0
#db#   [s] samples to skip.....0
#db# LF T55XX config
#db#            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
#db#            mode            |start|write|write|write| read|write|write
#db#                            | gap | gap |  0  |  1  | gap |  2  |  3
#db# ---------------------------+-----+-----+-----+-----+-----+-----+------
#db# fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A |
#db#     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
#db#               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |
#db#    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 |
#db#
#db# Transfer Speed
#db#   Sending packets to client...
#db#   Time elapsed............500ms
#db#   Bytes transferred.......193024
#db#   Transfer Speed PM3 -> Client = 386048 bytes/s
#db# Various
#db#   DBGLEVEL................1
#db#   ToSendMax...............-1
#db#   ToSendBit...............0
#db#   ToSend BUFFERSIZE.......2308
#db#   Slow clock..............31438 Hz
#db# Installed StandAlone Mode
#db#   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman
  compiled with MinGW-w64 9.2.0 OS:Windows (64b) ARCH:x86_64

 [ PROXMARK3 ]

 [ ARM ]
  bootrom: RRG/Iceman/master/release (git)
       os: RRG/Iceman/master/release (git)
  compiled with GCC 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

 [ FPGA ]
  LF image built for 2s30vq100 on 2019-07-31 at 15:57:16
  HF image built for 2s30vq100 on 2018-09-03 at 21:40:23

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 238284 bytes (45%) Free: 286004 bytes (55%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

[usb] pm3 -->

iceman1001, 2019-10-10:
Good.  a non rdv4 device.
64b mingw env.
We pushed a lot of fixes last days,  just make sure you compile and flash latest.

Lets try setting debug on your device and see  what it says when you eload.
```
hw dbg 4
hf iclass eload f ...
```

sherhannn79, 2019-10-10:
It’s strange. After entering 'dbg 4', the simulation worked. However, another problem appeared:
https://www.sendspace.com/file/6shvqs

iceman1001, 2019-10-10:
Strange,  try pulling latest, compile, flash.  Take notice if you got any compilation errors / flash warnings.
the dbg4 shouldn't changed anything regarding eload.
the sim 3 will work bad because of debugging taking too long,  but still doesn't explain why the SELECT fails.

iceman1001, 2020-01-04:
@sherhannn79  Is this still an issue?

sherhannn79, 2020-01-04:
I very sorry, but now I have absolutely no time for cooperation.

closed 2020-01-04 by sherhannn79