RfidResearchGroup/proxmark3 #429: need superuser rights to start client on manjaro

yakamoz147, 2019-10-09:
**Describe the bug**

When starting the client it freezes after displaying the banner but before showing the prompt for user input on my machine (see details below). The only way to get around that is to start the client with superuser rights on my machine.

It seems like the issue exists since commit 0449527e2aae99d483405fc18de346fb7613ff9c. It was still working fine without superuser rights on commit 43ffed1a261f7a5af1480ff0009b9961e63551bd (and the 3 commits in between seem to be broken (?) because they wont compile on my machine). to view the changes between these commits click [here](https://github.com/RfidResearchGroup/proxmark3/compare/43ffed1..0449527).

The thing is that there is no error message in any logfile whatsoever and also giving the binary every possible capability with `setcap` does not work. Also the CPU usage goes through the roof upon freeze (to 100% on the core executing the client) so maybe there is an endless loop here when the client is being executed in the context of my local user? I haven't found a reason for that though...

**To Reproduce**
Steps to reproduce the behavior:

0. Setup the Environment (OS, Kernel etc) like in the description below
1. Change to repo directory (after cloning it of course)
2. execute `client/proxmark3` (as user without superuser rights)
3. Bug: Banner shows up but it freezes before the prompt appears

**Expected behavior**
I expected the prompt of the proxmark3 client to show up as it did before (without having to execute it with `sudo`)

**Screenshots**
[image]

**Desktop:**
 - OS: Manjaro 18.1.0 Juhraya
 - Kernel: Linux 5.2.11-1-MANJARO
 - Architecture: x86_64

iceman1001, 2019-10-09:
Is it by choice you are trying to run the client in offline mode?  (ie no proxmark3 device connected)

yakamoz147, 2019-10-09:
yes, the behaviour is the same whether i start it in offline mode or with the connected proxmark3 device so i thought this example is simpler ;)

iceman1001, 2019-10-09:
Lets start with the basic.  Output from 
```
hw version
hw status
hw tune
```

doegox, 2019-10-09:
I added an option to skip Qt from the client, could you try to recompile the client and see if it helps?
```
make client/clean
make -j client SKIPQT=1
client/proxmark3
```

yakamoz147, 2019-10-09:
@iceman1001 
Unfortunately the problem is not with the hardware but with the client (apparently with the QT engine)

@doegox
Yes, it is working fine without QT. Thank you very much! So it seems like it's an issue with QT that somehow only gets triggered in the current commits of this repository and did not matter for the older commits.

iceman1001, 2019-10-09:
Those commands gives us insight to the software,  like which compiler, which version, etc,   that is why its in the bug template.   
Good thing removal of QT worked for you.

closed 2019-10-09 by iceman1001