RfidResearchGroup/proxmark3 #436: Proxmark doesn't boot after installing this fw

Knocks83, 2019-10-11:
Hi, I'm a newbie about Proxmark (I bought a chinese clone just4fun).
I'm trying to set it up but if I use your repo the Proxmark doesn't get recognized anymore by my Windows 10 PC (the Proxmark3 repo works just fine).
This is the compiling log, I didn't find any error but 4 eyes are better than 2
[output.txt](https://github.com/RfidResearchGroup/proxmark3/files/3719118/output.txt)

This is the flash output:
```
# ./pm3-flash-all
[=] Waiting for Proxmark3 to appear...
[=] Session log C:\Users\Knocks\Desktop\ProxSpace-master\pm3/.proxmark3/log_20191011.txt
[+] About to use the following files:
[+]     C:\Users\Knocks\Desktop\ProxSpace-master\pm3\client\../bootrom/obj/bootrom.elf
[+]     C:\Users\Knocks\Desktop\ProxSpace-master\pm3\client\../armsrc/obj/fullimage.elf
[+] Waiting for Proxmark3 to appear on COM6
.Found
[=] Available memory on this board: 512K bytes

[=] Permitted flash range: 0x00100000-0x00180000
[+] Loading ELF file C:\Users\Knocks\Desktop\ProxSpace-master\pm3\client\../bootrom/obj/bootrom.elf
[+] Loading usable ELF segments:
[+]    0 : V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
[+]    1 : V 0x00200000 P 0x00100200 (0x00000dd8->0x00000dd8) [R X] @0x298

[+] Loading ELF file C:\Users\Knocks\Desktop\ProxSpace-master\pm3\client\../armsrc/obj/fullimage.elf
[+] Loading usable ELF segments:
[+]    0 : V 0x00102000 P 0x00102000 (0x0003ee78->0x0003ee78) [R X] @0x94
[+]    1 : V 0x00200000 P 0x00140e78 (0x0000147c->0x0000147c) [RW ] @0x3ef0c
[=] Note: Extending previous segment from 0x3ee78 to 0x402f4 bytes

[+] Flashing...

[+] Writing segments for file: C:\Users\Knocks\Desktop\ProxSpace-master\pm3\client\../bootrom/obj/bootrom.elf
[+]  0x00100000..0x001001ff [0x200 / 1 blocks]
. OK
[+]  0x00100200..0x00100fd7 [0xdd8 / 7 blocks]
....... OK

[+] Writing segments for file: C:\Users\Knocks\Desktop\ProxSpace-master\pm3\client\../armsrc/obj/fullimage.elf
[+]  0x00102000..0x001422f3 [0x402f4 / 514 blocks]
.................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................. OK

[+] All done.

Have a nice day!
```

But when it starts instead of getting recognized as COM device Windows says "USB Device not recognized (device descriptor request failed)" (something like this, translated from Italian)
[image]

Knocks83, 2019-10-11:
I've just noticed that if I reboot the computer Windows can detect it as a COM port (noticed because I've tried reinstalling the drivers, it always says "Your driver is already up to date") but even if I can detect it I cannot use it.

iceman1001, 2019-10-11:
Lets see.  Compiling for RDV4 but you say you have a chinese clone.   How about you read and follow the instructions here
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/4_Advanced-compilation-parameters.md

Knocks83, 2019-10-12:
> Lets see. Compiling for RDV4 but you say you have a chinese clone. How about you read and follow the instructions here
> https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/4_Advanced-compilation-parameters.md

It worked, I had to set the platform to PM3OTHER.
Thanks a lot, I'm closing the issue.

closed 2019-10-12 by Knocks83