RfidResearchGroup/proxmark3 #437: 8 LEDs stay solid

joshpotter, 2019-10-15:
My proxmark3 RDv4 would not be recognized unless I held down the button when I plugged it into my computer. After some looking around I found a page for trouble shooting a possibly bricked proxmark. I tried "pm3-flash-bootrom" and it just seemed to hang for a few minutes. It (or maybe I slightly let go of the button by accident) disconnected. When I plugged it back in the LEDs stayed lit and now will not go off. Ive tried stand alone power and plugged into a computer. The proxmark is not seen at all now even when holding the button.

I clearly get this was my doing, perhaps I poke around more than needed, but this is a new issue for me and I have not found anything else similar yet.

doegox, 2019-10-15:
All LEDs stay on even if you press the button *while* you're plugging it in?
If yes, it means the bootloader is bricked too and you'll need to reflash via JTAG.
But first, if it happened, could it be because of https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/ModemManager-Must-Be-Discarded.md ?

joshpotter, 2019-10-15:
Thank you for the assistance. I just double checked and the LEDs stay on while the button is pressed. I had not removed the ModemManager so I am guessing that is my issue. I do not have a JTAG so I will have to get one and try to go down that road once I get back home.

iceman1001, 2019-10-16:
If you released the button ever so slightly during a bootrom flash,  you soft-bricked it.  As @doegox  say you will need to JTAG to recover it.   Read the docs about JTAG and your device be back in no time.  I will close this issue.

closed 2019-10-16 by iceman1001