RfidResearchGroup/proxmark3 #439: Invalid Character in Client on Windows

digitalentropy, 2019-10-16:
**Describe the bug**
Recent versions in the repo have introduced color coding in the console, but this functionality appears to result in nearly impossible to read text on some systems.

**To Reproduce**
Steps to reproduce the behavior:
1. Install ProxSpace v3.2 on Windows 10 x64
2. Compile latest git version
3. Run pm3 client and run "help" command.
**Expected behavior**
Text should readable by default, without the use of any registry keys.

**Screenshots**
Example Output:

`[usb] pm3 --> help
[32mhelp            [0m  This help. Use '<command> help' for details of a particular command.
[32manalyse         [0m  { Analyse utils... }
[32mdata            [0m  { Plot window / data buffer manipulation... }
[32memv             [0m  { EMV ISO-14443 / ISO-7816... }
[32mhf              [0m  { High frequency commands... }
[32mhw              [0m  { Hardware commands... }
[32mlf              [0m  { Low frequency commands... }
[32mmem             [0m  { Flash Memory manipulation... }
[32mreveng          [0m  { CRC calculations from RevEng software }
[32msc              [0m  { Smart card ISO-7816 commands... }
[32mscript          [0m  { Scripting commands }
[32mtrace           [0m  { Trace manipulation... }
[32musart           [0m  { USART commands... }
[32mwiegand         [0m  { Wiegand format manipulation... }
[32m                [0m
[32mmsleep          [0m  Add a pause in milliseconds
[32mrem             [0m  Add a text line in log file
[32mquit            [0m
[32mexit            [0m  Exit program
[usb] pm3 -->`

**Desktop (please complete the following information):**
 - OS: Windows 10 x64 [Version 10.0.18362.418]

**Additional context**
Add any other context about the problem here.

mwalker33, 2019-10-16:
Did you check the registey key as per windows 10 color section of 
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Windows-Installation-Instructions.md

iceman1001, 2019-10-16:
I have seen this on some ppl's screens now,  I think we can't rely on ppl to edit a windows registry key in order to sort it out.  @mwalker33 ,  Didn't you have a suggestion to set the registry if this value isn't set?  Or maybe we should add it to a windows.reg file and create a bat file to process it.

doegox, 2019-10-16:
+1 for the automatic check of the registry key. ref: https://github.com/RfidResearchGroup/proxmark3/issues/419

mwalker33, 2019-10-16:
I will have a look.  The code should not enable color if the key has not been set.  I.e. default should be no color, then if the user wanted to try, they could set the key.

So until I check the code I can only assume the reg is set, yet still windows is not supporting.

mwalker33, 2019-10-16:
I have completed a review and testing.

On my setup it works as expected. 
short version
session.supports_colors = false;
if reg key exists and = 1 then 
    session.supports_colors = true;

Tested with key present and 0, 1 and key not present.

What I did notice was that if you set the key, it only applies to a newly opened console window, else you do get the color "characters text" and not the color (the quick fix here is simply to start a new command prompt after setting the registry key).

If  others are seeing the characters then the registry key has been set (for some reason) and windows is not showing colors (windows not supporting for some reason).  The fix here would be as per instructions, set to 0 or delete.  BUT without knowing why it would be set if not to supportings colors 

So, I cant see a way to make it more automatic then currently set (open to ideas)

I can see a more manual ways.

How would you like to proceed ?
Would you like me to comment out the colors for windows until a better approach can be found ?

doegox, 2019-10-16:
Yeah maybe on the more manual way, reintroduce the idea of a JSON preferences file so ppl have to set the reg key and turn it on in the pref file...

mwalker33, 2019-10-16:
Will do.
I have commented out the win10 color reg key check.  PR submitted.
Will working on the preference file.

iceman1001, 2019-10-16:
a batch file which enables / disables it will also help out.

mwalker33, 2019-10-16:
I will have a think about the best way.
My current idea is the json settings file (that can be used for other things if/when needed)
Then in the client a command to "turn on windows color" that If selected will check the registry and if the flag is not set "offer" to set it.  
This way the user can just say "windows color on", then if it does not work "windows color off"
... And add the reminder to exit and restart the console for the changes to apply (windows thing)

iceman1001, 2019-10-16:
I rather not have direct commands for OS in the pm3 client. I rather go the batch file where the user can execute it right.  Running pm3 client with as low rights as possible.

mwalker33, 2019-10-16:
Looks like I might have missed the actual bug.
The changes I made were 100% undone and the color on most of the client has gone.
BUT the top level help still tries to show in green (which is what is showing in the bug report).
Looking for that now.

iceman1001, 2019-10-16:
show in green?  That is my change, but that should only be on ANSI enabled like all color macros is defined.

iceman1001, 2019-10-16:
@doegox  fixed it without knowing it...
https://github.com/RfidResearchGroup/proxmark3/commit/abf81bbbb7659af0d774f624ed633f7072f90c53

@mwalker33 pointed out correctly that its my printf call that doesn't filter out the ansi.  My bad.

mwalker33, 2019-10-16:
Maybe a simple "print and don't log" functions that calls the  

memcpy_filter_ansi(buffer3, buffer2, sizeof(buffer2), !session.supports_colors);

iceman1001, 2019-10-16:
The mixed used of  sprintf/snprintf/fprintf/printf vs our prefered PrintAndLogEx way should be addressed one day.  Since it doesn't really add value, its low prio. 

```
PrintAndLogEx( level, "", );

PrintEx(level, "", ..);
```

doegox, 2019-10-16:
There is already an available mechanism with `g_printAndLog` variable to tell if `PrintAndLogEx` should be printed and/or logged.

`g_printAndLog = PRINTANDLOG_PRINT;` to not log.
`g_printAndLog = PRINTANDLOG_LOG;` to not print.
`g_printAndLog = PRINTANDLOG_PRINT | PRINTANDLOG_LOG;` to restore normal behavior

Examples:
https://github.com/RfidResearchGroup/proxmark3/blob/master/client/proxmark3.c#L235-L238
https://github.com/RfidResearchGroup/proxmark3/blob/master/client/proxmark3.c#L540-L544

iceman1001, 2019-10-16:
that wasn't very obvious nor well documented but good thing it was implemented :)

Anyway,  I re-enabled @mwalker33 's Ansi test,  and trying @doegox global param for a test, to see if users still get the green ansi mess.   letting the users be testers :)  

It can all be refactored away when we get prefs.json settings file.

iceman1001, 2019-10-16:
@digitalentropy   pull latest and test if you get the ansi chars still in your client?

digitalentropy, 2019-10-26:
Just pulled latest.  Still an unreadable mess. :-(

Using ProxSpace v3.2 on Windows 10.

digitalentropy, 2019-10-26:
Clarification: After setting the dword value to 0 I can now use the client without having readability issues.

mwalker33, 2019-10-26:
Thanks for the feedback.

The characters you see are to display the text in color.   Some versions of windows 10 command (newer version) can display color, but only attempt to show the color if the reg dword has been set to 1.
The code will check that key, if "1", will output the color codes.
From the MS documentation, that dword should only be set to "1" to enable the enhanced terminal (that supports color) in command prompt.
One thing I did note was, setting the value to 1 will only apply to new cmd.exe prompts after it has been set. i.e. if you open cmd.exe, then set the dword it wont apply to the cmd.exe that is open, but will apply to all new ones.

That said, windows being windows does not support the color in ALL versions.
In the sort term you have it working.  I think longer term we will add a setting to output the  color codes (in windows 10).  That way the pm3 code wont attempt color unless you turn it on (and then set the dword to tell windows to attempt to use it).

I do have a followup question.
Did you create the key and set to "1" or was it already there?
In my 3 windows 10 computers, none had the key present, and I needed to add and set.

mwalker33, 2019-10-26:
Since you have the same version of windows as I do, in theory the color should work.
So, I was thinking something else needs to be set.
Can you open cmd.exe and check its properties.
[image]

I note that if I have "Use legacy console" checked, colors dont work (when dword is 1)
But if checked, they do.  NOTE: that you may need then check the "Terminal colors"
i.e. after unchecking the "use legacy console" start a new cmd.exe and the "terminal" tab should now show.
[image]

If you could have a play and give some feed back, that would be great.

digitalentropy, 2019-10-26:
Some follow-up:

1. I did have to create the key.  When I opened the ticket the key did not exist and I had to manually create it.  At least at the time, the absence of the key resulted in unexpected behavior.

2. Using the latest git version as of this comment, setting the dword to 1 still results in colors not working and text being unreadable.  "Use legacy console" was already unchecked an the Terminal tab is there.  I spent some time fiddling with settings and no combination of checkboxes resulted in color working.

I would note that like most users, I am running the ProxSpace environment and all of this is happening inside of msys2.  It is possible that behavior would be different on Windows Subsystem for Linux, but that should not be a requirement for having a readable display.

Speaking constructively, this really feels to me like a solution in search of a problem.  I like color coding just as much as the next person, but the default user experience needs to be as consistent as possible.  Manually creating registry keys, and looking for obscure checkboxes is not going to be an experience that does the proxmark3 platform any favors with its reputation.

My very strong suggestion is to have the color coding features, but require a compile flag to enable them rather than trying to automatically figure out if it's going to bork on the user's system.

I am happy to continue providing feedback regarding this issue, but the default behavior should change back to what it was.

mwalker33, 2019-10-27:
Since windows by default did not have the key set (or exist) then the default would have been NOT to attempt to use color.  The original issue was due to some output that bypassed the color filter (which was then fixed by iceman).
So, by setting the key back to 0, put it back to the original/default state. thus working without attempting to use color.

So while I agree that the user should not have to fiddle with things for a stable use, it was setup to work that way and then IF the user wanted to attempt to get color to work on windows, then they could try (i.e. it was not default to try to use color).  It is windows that states you need to create the registry key and the checkbox, so the notes/instructions where simply trying to provide pointers.

If it was still trying to use colors outside of the reg key(s) being set then we need to track that down.  
Side note:  i have always compiled in proxspace and can run the client in windows direct (as the output is a valid windows executable) i.e. no subsystem needed.

digitalentropy, 2019-10-27:
I think one of the confusing issues was that when I first reported the issue it was struggling even with no key set.  As you have mentioned, that is reportedly solved now.

I too used to be able to run the pm3 executable directly in the command prompt but the autobuild script in ProxSpace is not currently grabbing all of the required DLL’s to work right so rather than deal with that I just run everything from inside ProxSpace.

That said, I did spend some time to manually move some DLL’s around and run the command from a Windows command prompt, and the colors do indeed work as expected.  It just seems that MSYS2 has some issues when the key is enabled.  So…if there is a way to make it work nicely in both MSYS2/Cygwin and Windows Console, that would be super great.

mwalker33, 2019-10-27:
"...That said, I did spend some time to manually move some DLL’s around and run the command from a Windows command prompt, and the colors do indeed work as expected.  It just seems that MSYS2 has some issues when the key is enabled.  So…if there is a way to make it work nicely in both MSYS2/Cygwin and Windows Console, that would be super great...."

Yep, that now seems to be the current challenge.  

I just added the path to the dlls to my system path, but yes I do understand the challenge for those with less skills and just need things to work.

The reason I spent time getting it to run directly was it would start up faster and I can add shortcuts.

I am trying to find out why proxspace (migwin/msys2) is not passing the color codes correctly as it clearly supports some color, With any luck there might just be an option to support ansicolors.

I will see what @iceman thinks, but I have a few thoughts as we work to improve things.

digitalentropy, 2019-10-27:
I’m in full agreement.

Thank you for the comprehensive and thoughtful responses.

iceman1001, 2019-10-28:
I don't think we have a solution for mingw inside proxspace and ansi colors.
The idea of a settings.json file isn't new but still little bang for the buck to spend time on it.

I rather have @mwalker33 fixin' more new interesting LF stuff than adding a settings.json :)

If someone wants to add support for a settings.json file,  in which we let user say which windows to use when "data plot" or positionings, or default folder paths for saving etc,  I will merge it.

However,  the inoffical todo list is 80plus items,   on public roadmap its 20plus atleast.
No small changes either.  *just sayin'*

This issue is resolved as it was stated before and the idea of a settings.json file is logged already in other issue-lists. 

I am closing this one.

closed 2019-10-28 by iceman1001