RfidResearchGroup/proxmark3 #445: Compile Issue Felica

db69, 2019-10-18:
Latest pull from repo doesn't compile correctly (Proxspace V3.2) : 
:: Starting core system upgrade...
 there is nothing to do
:: Starting full system upgrade...
 there is nothing to do
warning: option --force is deprecated; use --overwrite instead

Cache directory: /var/cache/pacman/pkg/
:: Do you want to remove ALL files from cache? [Y/n]
removing all files from cache...

Database directory: /var/lib/pacman/
:: Do you want to remove unused repositories? [Y/n]
removing unused sync repositories...

pm3 ~$ git clone https://github.com/RfidResearchGroup/proxmark3.git
Cloning into 'proxmark3'...
remote: Enumerating objects: 102, done.                                                                                 remote: Counting objects: 100% (102/102), done.                                                                         remote: Compressing objects: 100% (82/82), done.                                                                        remote: Total 45111 (delta 48), reused 49 (delta 20), pack-reused 45009                                                 Receiving objects: 100% (45111/45111), 32.57 MiB | 7.55 MiB/s, done.                                                    Resolving deltas: 100% (35127/35127), done.                                                                             Updating files: 100% (1343/1343), done.                                                                                 pm3 ~$ cd proxmark3
pm3 ~/proxmark3$ make clean && make all

===================================================================
Platform name:     Proxmark3 rdv4
PLATFORM:          PM3RDV4
Platform extras:   No extra selected
Included options:  SMARTCARD FLASH LF HITAG ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF
Standalone mode:   LF_SAMYRUN
===================================================================
[*] MAKE client/clean
[!] Platform definitions changed, cleaning bootrom/armsrc/recovery first...
[*] MAKE bootrom/clean
[*] MAKE fpga_compress/clean
[*] MAKE armsrc/clean
[*] MAKE recovery/clean
[*] MAKE mfkey/clean
[*] MAKE nonce2key/clean

===================================================================
Platform name:     Proxmark3 rdv4
PLATFORM:          PM3RDV4
Platform extras:   No extra selected
Included options:  SMARTCARD FLASH LF HITAG ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF
Standalone mode:   LF_SAMYRUN
===================================================================
[*] MAKE client/all
[-] CC proxmark3.c
make[1]: *** wait: No child processes.  Stop.
make[1]: *** Waiting for unfinished jobs....
make[1]: *** wait: No child processes.  Stop.
make: *** [Makefile:98: client/all] Error 2

pm3 ~/proxmark3$ cd
pm3 ~$ git clone https://github.com/RfidResearchGroup/proxmark3.git
Cloning into 'proxmark3'...
remote: Enumerating objects: 102, done.
remote: Counting objects: 100% (102/102), done.
remote: Compressing objects: 100% (82/82), done.
remote: Total 45111 (delta 48), reused 49 (delta 20), pack-reused 45009
Receiving objects: 100% (45111/45111), 32.57 MiB | 5.90 MiB/s, done.
Resolving deltas: 100% (35127/35127), done.
Updating files: 100% (1343/1343), done.
pm3 ~$ cd proxmark3
pm3 ~/proxmark3$ make clean && make all

===================================================================
Platform name:     Proxmark3 rdv4
PLATFORM:          PM3RDV4
Platform extras:   BTADDON
Included options:  SMARTCARD FLASH FPC_USART_HOST LF HITAG ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF FPC_USART
Standalone mode:   LF_SAMYRUN
===================================================================
[... 360 lines cut ...]
[-] CC mifaresim.c
[-] CC iso14443b.c
[-] CC ../common/crapto1/crypto1.c
[-] CC des.c
[-] CC desfire_key.c
[-] CC mifaredesfire.c
[-] CC ../common/mbedtls/aes.c
[-] CC ../common/mbedtls/platform_util.c
[-] CC iclass.c
[-] CC optimized_cipher.c
[-] CC ../common/crc.c
[-] CC ../common/crc16.c
[-] CC ../common/crc32.c
[-] CC felica.c
felica.c: In function 'felica_send_request_service':
felica.c:777:43: error: division 'sizeof (uint8_t * {aka unsigned char *}) / sizeof (uint8_t {aka unsigned char})' does not compute the number of array elements [-Werror=sizeof-pointer-div]
     uint8_t len = sizeof(request_service) / sizeof((request_service)[0]);
                                           ^
felica.c:776:43: note: first 'sizeof' operand was declared here
 void felica_send_request_service(uint8_t *request_service) {
                                  ~~~~~~~~~^~~~~~~~~~~~~~~
cc1.exe: all warnings being treated as errors
make[1]: *** [../common_arm/Makefile.common:63: obj/felica.o] Error 1
make: *** [Makefile:95: armsrc/all] Error 2
pm3 ~/proxmark3$

doegox, 2019-10-18:
It has already been spotted, thanks.
Should be solved soon... Poking @7homasSutter 
https://github.com/RfidResearchGroup/proxmark3/commit/a5e94c5a1db88a5202eeecfb0f6e3dd77d2050fd#r35550616

doegox, 2019-10-18:
I pushed a temporary fix to un-break the repo

closed 2019-10-18 by doegox