RfidResearchGroup/proxmark3 #448: Broken "lf em 410x_sim"

temskiy, 2019-10-23:
The latest commit with worked function "lf em 410x_sim"  is 945fb82b48054d4096f82fda892abd21422a0025

**To Reproduce**
1. Build client & firmware at any commit after 945fb82b48054d4096f82fda892abd21422a0025
2. Flash firmwatre to PM
3. Start in client emulating em4100 tag with any ID

iceman1001, 2019-10-23:
...did you flash bootrom and fullimage?

temskiy, 2019-10-23:
Yes. After compiling i run "pm3-flash-all"

iceman1001, 2019-10-23:
And your output from 
```
hw status
hw tune
```

temskiy, 2019-10-23:
In attached file
[log.txt](https://github.com/RfidResearchGroup/proxmark3/files/3762176/log.txt)

iceman1001, 2019-10-23:
ok,  @mwalker33  confirmed the bug.  Its related to a coverity fix I tried.  He will have a look at it soon.

mwalker33, 2019-10-24:
This should now be fixed.  Can you update with the latest and confirm it's working for you.

temskiy, 2019-10-25:
Works fine
Thanks

iceman1001, 2019-10-27:
Thanks @temskiy  for reporting it in  and  thanks @mwalker33  for the fix!

closed 2019-10-27 by iceman1001

temskiy, 2020-01-11:
It`s broken again.