RfidResearchGroup/proxmark3 #450: hf mf autopwn #db# AcquireNonces: Auth1 error loop

Fl0-0, 2019-10-24:
hf mf autopwn does not work  and loop on `#db# AcquireNonces: Auth1 error`

**To Reproduce**
Steps to reproduce the behavior:
1. Try  hf mf autopwn on on mifare classic with HARD prng
2. The hardnested attack phase fails with #db# AcquireNonces: Auth1 error` loop:

```
[usb] pm3 --> hf mf autopwn
[!] no known key was supplied, key recovery might fail          
[+] loaded 23 keys from hardcoded default array          
[=] running strategy 1          
...          
[+] Chunk: 7.7s | found 14/32 keys (23)
          
[=] running strategy 2          
...          
[+] Chunk: 7.4s | found 14/32 keys (23)
          
[+] target sector:  0 key type: A -- found valid key [  A0 A1 A2 A3 A4 A5  ] (used for nested / hardnested attack)          
[+] target sector:  0 key type: B -- found valid key [  FF FF FF FF FF FF  ]          
[+] target sector:  3 key type: A -- found valid key [  FF FF FF FF FF FF  ]          
[+] target sector:  3 key type: B -- found valid key [  FF FF FF FF FF FF  ]          
[+] target sector:  9 key type: A -- found valid key [  FF FF FF FF FF FF  ]          
[+] target sector:  9 key type: B -- found valid key [  FF FF FF FF FF FF  ]          
[+] target sector: 10 key type: A -- found valid key [  FF FF FF FF FF FF  ]          
[+] target sector: 10 key type: B -- found valid key [  FF FF FF FF FF FF  ]          
[+] target sector: 11 key type: A -- found valid key [  FF FF FF FF FF FF  ]          
[+] target sector: 11 key type: B -- found valid key [  FF FF FF FF FF FF  ]          
[+] target sector: 12 key type: A -- found valid key [  A0 A1 A2 A3 A4 A5  ]          
[+] target sector: 13 key type: A -- found valid key [  A0 A1 A2 A3 A4 A5  ]          
[+] target sector: 14 key type: A -- found valid key [  A0 A1 A2 A3 A4 A5  ]          
[+] target sector: 15 key type: A -- found valid key [  A0 A1 A2 A3 A4 A5  ]          
[+] Using AVX2 SIMD core.          

          
 time    | #nonces | Activity                                                | expected to brute force          
         |         |                                                         | #states         | time           
------------------------------------------------------------------------------------------------------          
       0 |       0 | Start using 8 threads and AVX2 SIMD core                |                 |          
       0 |       0 | Brute force benchmark: 1036 million (2^29.9) keys/s     | 140737488355328 |    2d          
       1 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |    2d          
#db# AcquireNonces: Auth1 error          
#db# AcquireNonces: Auth1 error          
#db# AcquireNonces: Auth1 error          
#db# AcquireNonces: Auth1 error          
#db# AcquireNonces: Auth1 error
................
```

**Expected behavior**
It works well when i do it manually:

```
[usb] pm3 --> hf mf hardnested 0 A A0A1A2A3A4A5 4 A
--target block no:  4, target key type:A, known target key: 0x000000000000 (not set), file action: none, Slow: No, Tests: 0           
[+] Using AVX2 SIMD core.          

          
 time    | #nonces | Activity                                                | expected to brute force          
         |         |                                                         | #states         | time           
------------------------------------------------------------------------------------------------------          
       0 |       0 | Start using 8 threads and AVX2 SIMD core                |                 |          
       0 |       0 | Brute force benchmark: 1152 million (2^30.1) keys/s     | 140737488355328 |   34h          
       1 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |   34h          
       4 |     112 | Apply bit flip properties                               |     61067862016 |   53s          
       5 |     223 | Apply bit flip properties                               |      6988976640 |    6s          
       6 |     335 | Apply bit flip properties                               |      3215611904 |    3s          
       7 |     446 | Apply bit flip properties                               |      3163382528 |    3s          
       8 |     557 | Apply bit flip properties                               |      2735795712 |    2s          
[... 108 lines cut ...]
#db#   ToSendMax...............-1          
#db#   ToSendBit...............0          
#db#   ToSend BUFFERSIZE.......2308          
#db#   Slow clock..............31468 Hz          
#db# Installed StandAlone Mode           
#db#   HF Mifare sniff/simulation - (Craig Young)          
#db# Flash memory dictionary loaded           
#db#   Mifare..................859 keys          
#db#   T55x7...................109 keys          
#db#   iClass..................7 keys  
```

 - data tune

```
[+] LF antenna: 69.40 V - 125.00 kHz          
[+] LF antenna: 41.50 V - 133.33 kHz          
[+] LF optimal: 70.39 V - 126.32 kHz          
[+] LF antenna is OK  
          
[+] HF antenna: 48.58 V - 13.56 MHz          
[+] HF antenna is OK           
          
[+] Displaying LF tuning graph. Divisor 89 is 133.33 kHz, 95 is 125.00 kHz.
```

Fl0-0, 2019-10-25:
After playing with gdb and tracing the call to `mfnestedhard()`, i discovered that the key argument is NULL.

Call to mfnestedhard() by `hf mf hardnested 0 A A0A1A2A3A4A5 4 A`:

```
#0  mfnestedhard (blockNo=blockNo@entry=0x0, keyType=keyType@entry=0x0, key=key@entry=0x7fffea322224 "\240\241\242\243\244\245", trgBlockNo=trgBlockNo@entry=0x4, trgKeyType=trgKeyType@entry=0x0, 
    trgkey=trgkey@entry=0x0, nonce_file_read=0x0, nonce_file_write=0x0, slow=0x0, tests=0x0, foundkey=0x7fffea322230, filename=0x7fffea322620 "") at cmdhfmfhard.c:2161
```

Call to mfnestedhard() by `hf mf autopwn`:

```
#0  mfnestedhard (blockNo=blockNo@entry=0x0, keyType=keyType@entry=0x0, key=key@entry=0x7fffea32d39c "", trgBlockNo=trgBlockNo@entry=0x4, trgKeyType=trgKeyType@entry=0x0, trgkey=trgkey@entry=0x0, 
    nonce_file_read=0x0, nonce_file_write=0x0, slow=0x0, tests=0x0, foundkey=0x7fffea32d3b8, filename=0x0) at cmdhfmfhard.c:2161
```
Commit https://github.com/RfidResearchGroup/proxmark3/commit/abb011c179beb89f571288cf3770229bde17fbac change `key` for `tmp_key`, it works again with `key`.

ikarus23, 2019-10-26:
Can confirm. I noticed this behavior too. `hf mf autopwn` does not work while manual `hf mf hardnested` works.

iceman1001, 2019-10-28:
I need to verify the coverity issue related with this.

Fl0-0, 2019-11-04:
Any news on this issue ?

closed 2019-11-04 by doegox

doegox, 2019-11-04:
see my comment on the PR, thanks for your help @Fl0-0 !

iceman1001, 2019-11-04:
yup,  thanks @Fl0-0 !

PR #452 (2019-11-04): Fix for hf mf autopwn #db# AcquireNonces: Auth1 error loop, revert co…
…mmit abb011c

Fix #450