RfidResearchGroup/proxmark3 #455: not able to read block 19 of mifare ultralight ev1 (MF0UL1101)

zdpark, 2019-10-28:
I have tried to dump several MF0UL1101 via hf mfu dump k 'password '
but block 19 is zeroed all the time. 
hf mf rdbl b 19 'password' showed the same results.

Thanks.

iceman1001, 2019-10-28:
...I get the feeling this is not an issue about the code but more of a usage problem...

your output from,  where **aaaaaaaa** is to be replaced with your known key.

```
hf mfu info
hf mfu info k aaaaaaaa
hf 14a list
```

zdpark, 2019-10-28:
Nope, When I do the same job with old iceman build, 

I can get the block 19.

zdpark, 2019-10-28:
Here is the dump of the card I tried.

Block#   | Data        |lck| Ascii
---------+-------------+---+------
  0/0x00 | 04 9E 58 4A |   | ..XJ
  1/0x01 | 9A D0 59 80 |   | ..Y.
  2/0x02 | 93 48 08 00 |   | .H..
  3/0x03 | D8 05 DE 19 | 1 | ....
  4/0x04 | D9 C1 1E E2 | 0 | ....
  5/0x05 | A7 3D C6 F9 | 0 | .=..
  6/0x06 | 46 3F FC A7 | 0 | F?..
  7/0x07 | 5E 56 94 E0 | 0 | ^V..
  8/0x08 | 9A B2 B0 B7 | 0 | ....
  9/0x09 | 62 15 53 D8 | 0 | b.S.
 10/0x0A | 00 00 00 00 | 0 | ....
 11/0x0B | 00 00 00 00 | 0 | ....
 12/0x0C | 00 00 00 00 | 0 | ....
 13/0x0D | 00 00 00 00 | 0 | ....
 14/0x0E | 00 00 00 00 | 0 | ....
 15/0x0F | 00 00 00 00 | 0 | ....
 16/0x10 | 00 00 00 10 | 0 | ....
 17/0x11 | 80 05 00 00 | 0 | ....
 18/0x12 | 1C 90 D2 3C | 0 | ...<
 19/0x13 | E2 0A 00 00 | 0 | ....
---------------------------------
[=] Restoring data blocks.

zdpark, 2019-10-28:
the official 2.3.0 build has the same problem. tried w/ rdv2.0

iceman1001, 2019-10-28:
Normally block 18, 19 isn't readable.    Its only writeable.
Since block 18 is password block,   and block19 is pack.   When calling the commands with a key,  we put that key in block18 output and its pack response.   Since you don't give me the output to the commands,  this issue can't be resolved.   

I am closing because of not enough information given by OP

closed 2019-10-28 by iceman1001