RfidResearchGroup/proxmark3 #461: Script read_pwd_mem.lua problem

BreakSecurity, 2019-11-02:
**Describe the bug**
`script run read_pwd_mem -k 6`
print tons of empty passwords

**To Reproduce**
Steps to reproduce the behavior:
Set prox RDV4 in Bogito Standalone mode
With a second proxmark try:
`hf mfu dump k FFFFFFFF`
Then exit standalone mode (on rdv4) and try:
`script run read_pwd_mem -k 6`

**Desktop (please complete the following information):**
 - OS: Windows
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
`[ Proxmark3 RFID instrument ]                                                                                                                                                                                                                                                                                           [ CLIENT ]                                                                                               client: RRG/Iceman                                                                                      compiled with MinGW-w64 9.2.0 OS:Windows (64b) ARCH:x86_64                                                                                                                                                     [ PROXMARK RDV4 ]                                                                                        external flash:                  present                                                                smartcard reader:                present                                                                                                                                                                       [ PROXMARK RDV4 Extras ]                                                                                 FPC USART for BT add-on support: present                                                                                                                                                                       [ ARM ]                                                                                                  bootrom: RRG/Iceman/master/b4400bbc 2019-09-01 18:48:49                                                      os: RRG/Iceman/master/b4400bbc 2019-09-01 18:49:10                                                 compiled with GCC 8.3.1 20190703 (release) [gcc-8-branch revision 273027]                                                                                                                                      [ FPGA ]                                                                                                 LF image built for 2s30vq100 on 2019-07-31 at 15:57:16                                                  HF image built for 2s30vq100 on 2018-09-03 at 21:40:23                                                                                                                                                         [ Hardware ]                                                                                             --= uC: AT91SAM7S512 Rev A                                                                              --= Embedded Processor: ARM7TDMI                                                                        --= Nonvolatile Program Memory Size: 512K bytes, Used: 272414 bytes (52%) Free: 251874 bytes (48%)                                                                                                              --= Second Nonvolatile Program Memory Size: None                                                        --= Internal SRAM Size: 64K bytes                                                                       --= Architecture Identifier: AT91SAM7Sxx Series                                                         --= Nonvolatile Program Memory Type: Embedded Flash Memory   ` 
 - hw status
`#db# Memory                                                                                             #db#   BIGBUF_SIZE.............40000                                                                    #db#   Available memory........38912                                                                    #db# Tracing                                                                                            #db#   tracing ................0                                                                        #db#   traceLen ...............0                                                                        #db# Currently loaded FPGA image                                                                        #db#   mode.................... HF image built for 2s30vq100 on 2018-09-03 at 21:40:23                  #db# Flash memory                                                                                       #db#   Baudrate................24 MHz                                                                   #db#   Init....................OK                                                                       #db#   Memory size.............2 mbits / 256 kb                                                         #db#   Unique ID...............0xD567A882A71A9C26                                                       #db# Smart card module (ISO 7816)                                                                       #db#   version.................v3.11                                                                    #db# LF Sampling config                                                                                 #db#   [q] divisor.............95 ( 125 kHz )                                                           #db#   [b] bps.................8                                                                        #db#   [d] decimation..........1                                                                        #db#   [a] averaging...........Yes                                                                      #db#   [t] trigger threshold...0                                                                        #db# LF T55XX config                                                                                    #db#            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]                               #db#            mode            |start|write|write|write| read|write|write                              #db#                            | gap | gap |  0  |  1  | gap |  2  |  3                                #db# ---------------------------+-----+-----+-----+-----+-----+-----+------                             #db# fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A |                             #db#     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |                             #db#               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |                             #db#    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 |                             #db#                                                                                                    #db# Transfer Speed                                                                                     #db#   Sending packets to client...                                                                     #db#   Time elapsed............500ms                                                                    #db#   Bytes transferred.......282112                                                                   #db#   Transfer Speed PM3 -> Client = 564224 bytes/s                                                    #db# Various                                                                                            #db#   DBGLEVEL................1                                                                        #db#   ToSendMax...............-1                                                                       #db#   ToSendBit...............0                                                                        #db#   ToSend BUFFERSIZE.......2308                                                                     #db#   Slow clock..............29324 Hz                                                                 #db# Installed StandAlone Mode                                                                          #db#   HF 14a sniff standalone with ULC/ULEV1/NTAG auth storing in flashmem - aka BogitoRun (Bogito)                                                                                                            #db# Flash memory dictionary loaded            [... cut ...]                                      #db#   iClass..................7 keys  `
 - data tune
```
[=] Measuring antenna characteristics, please wait...

[=] You can cancel this operation by pressing the pm3 button
..

[+] LF antenna: 65.86 V - 125.00 kHz
[+] LF antenna: 41.92 V - 134.00 kHz
[+] LF optimal: 65.86 V - 125.00 kHz
[+] LF antenna is OK

[+] HF antenna: 35.38 V - 13.56 MHz
[+] HF antenna is OK

[+] Displaying LF tuning graph. Divisor 89 is 134kHz, 95 is 125kHz.
```
**Additional context**

iceman1001, 2019-11-02:
@bogiton  ,  maybe something for you?

bogiton, 2019-11-02:
PR #462 should fix it. Added a new lua script that uses SPIFFS instead (read_pwd_mem_spiffs.lua) that reads the hf_bog.log file where the passwords are saved now. Could be adapted for other files too.

iceman1001, 2019-11-02:
yeah, @BreakSecurity   test it,  I merged @bogiton 's PR

BreakSecurity, 2019-11-03:
`script run read_pwd_mem_spiffs.lua -f hf_bog.log `

Give the output:
`
---------------------------------------- 
Read passwords stored in memory (SPIFFS) 
----------------------------------------                                          
[01] 00000000 
[02] 00000000                            [03] 00000000                            [04] 00000000                            [05] 00000000                            [06] 00000000                            [07] 00000000                            [08] 00000000                            [09] 00000000                            [10] 00000000                            [11] 00000000                            [12] 00000000                            ---------------------------------------- [+] found 12 passwords                                                            [+] Finished read_pwd_mem_spiffs.lua     `

It should only give the ffffffff psw

bogiton, 2019-11-03:
The lua script stops when it finds a 0xFFFFFFFF value, just like the previous one. The logic behind this is that you normally won't have to sniff that default pwd but most importantly, at least in the previous version of the flash mem, the initial values (after wipe) is full of FFs. That is why it was chosen as the stop value. Of course, since we now use SPIFFS we could omit this restriction. @iceman1001 what do you think?

iceman1001, 2019-11-03:
if we download a file from spiffs,  we really don't need a stop value :)  the number of bytes read from spiffs should be the limit.
But why does it add 0x00 00 00 00 serveral times?  aha, you append regardsless of previous findings.

iceman1001, 2019-11-19:
Should be fixed with latest fixes from @bogiton

closed 2019-11-19 by iceman1001