RfidResearchGroup/proxmark3 #468: Would it be possible to create a python wrapper?

waleed-tariq, 2019-11-06:
I am very new to this RFID stuff, my only experience being with the ChameleonMini. I loved the Chameleon's Python interface to automate most, if not all of its commands. I was wondering if this same structure could be done with the Proxmark. If I could create a python wrapper to issue commands and retrieve the results from the device via the terminal. This is based of my assumptions that most commands like emulating and reading from the Proxmark can be done via the command line. I believe this method would be very helpful for me and the community. Please let me know if this would be possible.
Thanks!

doegox, 2019-11-06:
hi, there is no python wrapper, best you can do for now is call client from python, e.g.
```python
def pm3(cmd):
    try:
        r=subprocess.run(['proxmark3', '/dev/ttyACM0', '-c', cmd], stdout=subprocess.PIPE, stderr=subprocess.STDOUT, universal_newlines=True, timeout=5)
        out = r.stdout
    except subprocess.TimeoutExpired:
        lcd.clear()
        lcd.write_string("TIMEOUT!")
        time.sleep(0.5)
        out = ""
    return [x.rstrip() for x in out.split('\n')]

out=pm3("hf 14a reader")
```

waleed-tariq, 2019-11-06:
So would it be possible to emulate an ISO15693 card and set the NDEFdata using the proxmark through the subprocess/terminal to call the client? Basically, creating a tag that has its own specific NDEF data that I have manually set. Would that be possible?

doegox, 2019-11-06:
No idea, please use the forum for such questions.

closed 2019-11-06 by doegox