RfidResearchGroup/proxmark3 #469: iCLASS Simulation Inconsistent

digitalentropy, 2019-11-07:
**Describe the bug**
App issuer area block does not always simulate correctly from dumps.
**To Reproduce**
Steps to reproduce the behavior:
1. Create a dump with a unique App Issuer Block
2. Try to sim.
3. Cry.

**Expected behavior**
All blocks should be emulated correctly from dump file.

iceman1001, 2019-11-07:
Ok,  I see the problem.

iceman1001, 2019-11-07:
@digitalentropy  I pushed a fix.

ikarus23, 2019-11-08:
A little bit off topic and just out of curiosity: are @pwpiwi fixes for iCLASS also part of this repository? (e.g. https://github.com/Proxmark/proxmark3/commit/ece38ef311b28eefb1d716937139aad9ee00985c)

iceman1001, 2019-11-08:
Nop,

iceman1001, 2020-02-22:
@digitalentropy  Closing since no feedback from OP.

closed 2020-02-22 by iceman1001