RfidResearchGroup/proxmark3 #470: [idea] Implement hf mf cwipe

ikarus23, 2019-11-08:
I noticed that this repository does not have a `hf mf cwipe` command as the official repository. I'm not sure if `cwipe` is doing something special, like erasing a card even though the access bits are locked. 

It would be great to have this command here as well.

iceman1001, 2019-11-08:
....the chinese backdoor command for wipe is already inside the repo.
You find it as  a parameter to the following commands.
- hf mf csetuid w
- hf mf csetblk w 

The card implemention of wipe empties the card completely.  Not sectortrailers no block0,  just zeros.  So you will need to take that in consideration when using that specific function.

ikarus23, 2019-11-09:
Oh ok, thanks!

Not sure if I understand you correctly regarding the `cwipe`. I thought this command formats everything back to factory settings. So all data blocks are 0, the keys are F and the access conditions are default. Am I wrong?

merlokk, 2019-11-09:
`wipe` command of `gen1a` cards works very specific.
i have several cards that go nothing after this command.
but `cwipe` command is bigger:
https://github.com/merlokk/proxmark3/blob/master/armsrc/mifarecmd.c#L1285
it can fill the card with `default` data

merlokk, 2019-11-09:
the same, thanks @iceman1001 
https://github.com/RfidResearchGroup/proxmark3/blob/master/client/luascripts/remagic.lua

ikarus23, 2019-11-10:
Thanks for clearing that up! So `hf mf csetuid w` or `hf mf csetblk w` is close, but not exactly the same as `cwipe`. At least `cwipe` is more comfortable. But I guess the `remagic.lua` counts as an equivalent to `cwipe`. If not, it would be great to see `cwipe` in this repo as well. It's just so convenient. ;)

iceman1001, 2019-11-10:
hm,..   lets see,  in the name of unifying and since it is the famous @ikarus23  who is asking...
We currently have  
- lf t55xx wipe
which zeros and fills a t5577 card.
If we gonna have same style commands, a
- hf mf wipe
where we can take either a gen1a /gen1b,2 ,  or a original nxp mfc card and fill with zeros and proper st blocks and valid block0.

ikarus23, 2019-11-10:
> and since it is the famous @ikarus23 who is asking...

I'm blushing :) but don't do it just for me. If it's only me finding this feature great, I should implement it my self. ;)

The approaches you suggesting sound really good. This way it will be universal.

merlokk, 2019-12-06:
#487

merlokk, 2019-12-06:
@ikarus23 woks?

iceman1001, 2019-12-06:
@merlokk  if you read my previous post 

 > hm,.. lets see, in the name of unifying and since it is the famous @ikarus23 who is asking...
> We currently have
> 
>     lf t55xx wipe [which zeros and fills a t5577 card.´]
> 
> If we gonna have same style commands, a
> 
>     hf mf wipe
> 
> where we can take either a gen1a /gen1b,2 , or a original nxp mfc card and fill with zeros and proper ST blocks and valid block0.
> 
> And the command could take advantage of the special "wipe" command for Gen1a/1b tags.

We widened the scope to   **hf mf wipe**   and not just gen1a wipes....

merlokk, 2019-12-06:
ok, ill think. needs to combine check keys with write

iceman1001, 2020-06-09:
@merlokk  ....did you have some progress?

iceman1001, 2021-04-02:
implemented https://github.com/RfidResearchGroup/proxmark3/commit/f4694a4cd57bf22e2e3d5696b4fce53f9e9a8682

closed 2021-04-02 by iceman1001