RfidResearchGroup/proxmark3 #472: Pin Meassurments

duckie37, 2019-11-10:
So the first thing I did when I got my Proxsmark3 v4 was brick it,  At least I am sure I did. All the lights are solid and USB won't puck it up. I am ordering a jtag device but need to get a right angle breakaway pin for this. I noticed these are tiny compared to previous versions. Any ideas on what kind of right-angle pins will fit on this comfortably?

Thanks

merlokk, 2019-11-10:
I'd prefer pogo pins. As I remembered- it size like 50 or 75

iceman1001, 2019-11-10:
or get a adaptor like this  https://github.com/RfidResearchGroup/proxmark3/wiki/Tools

but before that,  have you tried the button trick?   Ie, are you sure you bricked your device?
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Troubleshooting.md

closed 2019-11-10 by iceman1001

duckie37, 2019-11-10:
Thank you for the link. Yes, I am sure it's bricked. I had modem management on when I was updating the flash and since then it's solid lights across even when I boot it holding button down.

iceman1001, 2019-11-11:
That sucks.  Luckily your way to recovery is JTAG.  You can read the docs here and look in the folder 
`https://github.com/RfidResearchGroup/proxmark3/tree/master/tools/jtag_openocd`