RfidResearchGroup/proxmark3 #473: Protecting T5577 fails after writing HID tags

swg0101, 2019-11-11:
I think this stopped working a couple of weeks back, and something like this doesn't work anymore:

> [usb] pm3 --> lf hid clone 2
> [=] Preparing to clone HID tag with ID 000000002
> [usb] pm3 --> lf sea
> [=] NOTE: some demods output possible binary
> [=] if it finds something that looks like a tag
> [=] False Positives ARE possible
> [=]
> [=] Checking for known tags...
> [=]
> [+] HID Prox TAG ID: 000000002 (1) - Format Len: 37bit - OEM: 000 - FC: 0 - Card: 1
> 
> [+] Valid HID Prox ID found!
> 
> [+] Chipset detection: T55xx
> [=] Hint: try `lf t55xx` commands
> [usb] pm3 --> lf t5 det
>     Chip Type      : T55x7
>     Modulation     : FSK2a
>     Bit Rate       : 4 - RF/50
>     Inverted       : Yes
>     Offset         : 32
>     Seq. Term.     : No
>     Block0         : 0x00107060
>     Downlink Mode  : default/fixed bit length
>     Password Set   : No
> 
> [usb] pm3 --> lf t5 pr n FEEDBEEF
> [=] Checking current configuration
> [+] Wrote new password
> [!] Failed to validate the password write. aborting.
> [!] Command failed. Did you run `lf t55xx detect` before?

Sometimes the command works if I do a search and then protect immediately, although sometimes that results in a corrupt card.

mwalker33, 2019-11-11:
While I have not looked it sounds like the write of the password bit failed, thus a password read when not password protected then causes the corrupt card config.
For testing I would use a password with the first byte of 00xxxxxx 

Is this to a card or fob?  
What proxmark are you using?
I am wondering  if it's a coupling issue?

swg0101, 2019-11-11:
Happens in both card or fob.
Proxmark3 Easy
If it was a coupling issue, I would think there could potentially be corruption simply by writing to the card. The issue only occurs while password protecting the card. When I say corruption, sometimes the password does get written to the card but the contents are no longer readable. For example, in the above scenario a lf search would result in an unknown card that's locked, but a lf t5 wip p FEEDBEEF can erase this back to a writable state.

mwalker33, 2019-11-11:
That's interesting.  Thanks for the extra information.  
Did you ever try an lf t55 det with the password in your last example? Sounds like the password and password bit got set but something else happened to the config block.

swg0101, 2019-11-11:
There are a few cases I see:
1. The protect command fails, lf search results in correct data --> lf t5 det succeeds without password.
2. The protect command succeeds, lf search results in unknown data --> lf t5 det works with the set password.
3. The protect command succeeds, lf search results in correct data --> lf t5 det works with the set password.

It is quite difficult to get case 3 to work. I generally have to retry it a couple times for it to work (and without moving or touching the card). And also, none of the write commands had any issues and I can read/write from/to the card just fine. It's just the password protection that seems wonky.

mwalker33, 2019-11-11:
OK, I have just complete some very quick tests.
The order should be.
1. lf t55 det    : This will ensure the pm3 client knows about the card setup.
2. lf hid clone 2  : Create the HID with value "2"
3. lf search  : confirm the card is now an HID card
4. lf t55 det   : IMPORTANT as it tells the pm3 client about any change from the HID clone.
5. lf t55 protect n 00112233   : set the password.

If i do the above, it works as expected.
if I dont do the 2nd detect (step 4) then I got the unknown card, the lf t55 det failed (with/without the password), but the lf t55 wipe worked (as per your findings).

Can you try with my steps and confirm if that works or fails.
Thanks

swg0101, 2019-11-11:
```
[usb] pm3 --> lf t5 det
    Chip Type      : T55x7
    Modulation     : ASK
    Bit Rate       : 2 - RF/32
    Inverted       : No
    Offset         : 32
    Seq. Term.     : Yes
    Block0         : 0x000880E0
    Downlink Mode  : default/fixed bit length
    Password Set   : No

[usb] pm3 --> lf hid cl 2
[=] Preparing to clone HID tag with ID 000000002
[usb] pm3 --> lf sea
[=] NOTE: some demods output possible binary
[=] if it finds something that looks like a tag
[=] False Positives ARE possible
[=]
[=] Checking for known tags...
[=]
[+] HID Prox TAG ID: 000000002 (1) - Format Len: 37bit - OEM: 000 - FC: 0 - Card: 1

[+] Valid HID Prox ID found!

[usb] pm3 --> lf t5 det
    Chip Type      : T55x7
    Modulation     : FSK2a
    Bit Rate       : 4 - RF/50
    Inverted       : Yes
    Offset         : 32
    Seq. Term.     : No
    Block0         : 0x00107060
    Downlink Mode  : default/fixed bit length
    Password Set   : No

[usb] pm3 --> lf t5 pr n FEEDBEEF
[=] Checking current configuration
[+] Wrote new password
[!] Failed to validate the password write. aborting.
[!] Command failed. Did you run `lf t55xx detect` before?
[usb] pm3 --> lf t5 pr n 00112233
[=] Checking current configuration
[+] Wrote new password
[+] Validated new password
[+] Wrote modified configuration block
[!] Safety check: Could not detect if PWD bit is set in config block. Exits.
[=] Block0 write detected, running `detect` to see if validation is possible
    Chip Type      : T55x7
    Modulation     : FSK2a
    Bit Rate       : 4 - RF/50
    Inverted       : Yes
    Offset         : 32
    Seq. Term.     : No
    Block0         : 0x00107070
    Downlink Mode  : default/fixed bit length
    Password Set   : Yes
    Password       : 00112233

[+] New configuration block 00107070 password 00112233
[+] Success, tag is locked

```

mwalker33, 2019-11-11:
So it worked with 00112233 but not FEEDBEEF mmmmmm
On my RDV4 it worked for both.

I think this needs to be split into two parts.

a) If the lf t55 detect is used just prior to the protect, does it stop "corrupting" the card ?
i.e. it may or may not set the protect, but should not corrupt the card.

I run some tests and in this exact test with the hid clone 2, and NO lf t55 detect before the protect, the read of the config block returned 0xFFFFFFFF.  If i write that back to block 0 (knowing it is invalid) then I got the exact results you saw.  i.e. lf seach - nothing, lf t55 det - nothing, but the wipe worked.  Based on my understanding of the t5577, that is what I expected.

b) Why does the FEEDBEEF password fail.

I think this may be a fsk2a  decoding challenge.  The detect worked (as you saw), then it wrote FEEDBEEF to the card, but for some reason when it read it back, it did not get FEEDBEEF (as it stated)
I have seen some issue in decoding fsk2a when the high bits are set, so while I cant explain why, It does match what I have seen in the past.
Based on my experience, I expect that the password was written ok, and if the password bit was then set it would work as expected.  BUT the safety check fails with an invalid read of block 7, this it does not continue.
 
Can you try this for me.
lf t55 wipe   : as needed to get a clean card
lf hid clone 2 
lf t55 det    : Should show the new block 0 ok.
lf t55 protect n FEEDBEEF

At this point, if your device is consistent and you get the fail again, then the following:
lf t55 write b 0 d 000880E0
lf t55 det
lf t55 dump
 and post the dump results.

iceman1001, 2019-11-11:
If protect command fails,  
```[!] Failed to validate the password write. aborting.```
the modified config block was never written to tag.   Only pwd block was written and it couldnt be verified and hence protect aborted.

So question is if your tag has some lock bits for block7 ?  

Lets see what happens in the different steps.   
```
lf t55xx wipe
lf t55xx detect
lf t55xx dump
lf hid clone 2
lf t55xx detect
lf t55xx protect n 11223344
lf t55xx detect
lf t55xx dump
```

swg0101, 2019-11-11:
Result 1:
```
[usb] pm3 --> lf t55 wipe

[=] Begin wiping T55x7 tag

[=] Default configation block 000880E0
[=] Writing page 0  block: 00  data: 0x000880E0
[=] Writing page 0  block: 01  data: 0x00000000
[=] Writing page 0  block: 02  data: 0x00000000
[=] Writing page 0  block: 03  data: 0x00000000
[=] Writing page 0  block: 04  data: 0x00000000
[=] Writing page 0  block: 05  data: 0x00000000
[=] Writing page 0  block: 06  data: 0x00000000
[=] Writing page 0  block: 07  data: 0x00000000
[usb] pm3 --> lf hid clone 2
[=] Preparing to clone HID tag with ID 000000002
[usb] pm3 --> lf t55 det
    Chip Type      : T55x7
    Modulation     : FSK2a
    Bit Rate       : 4 - RF/50
    Inverted       : Yes
    Offset         : 32
    Seq. Term.     : No
    Block0         : 0x00107060
    Downlink Mode  : default/fixed bit length
    Password Set   : No

[usb] pm3 --> lf t55 protect n FEEDBEEF
[=] Checking current configuration
[+] Wrote new password
[!] Failed to validate the password write. aborting.
[!] Command failed. Did you run `lf t55xx detect` before?
[usb] pm3 --> lf t55 write b 0 d 000880E0
[=] Writing page 0  block: 00  data: 0x000880E0
[usb] pm3 --> lf t55 det
    Chip Type      : T55x7
    Modulation     : ASK
    Bit Rate       : 2 - RF/32
    Inverted       : No
    Offset         : 33
    Seq. Term.     : Yes
    Block0         : 0x000880E0
    Downlink Mode  : default/fixed bit length
    Password Set   : No

[usb] pm3 --> lf t55 dump
[+] Reading Page 0:
[+] blk | hex data | binary                           | ascii
[+] ----+----------+----------------------------------+-------
[+]  00 | 000880E0 | 00000000000010001000000011100000 | ..▒▒
[+]  01 | 1D555555 | 00011101010101010101010101010101 | .UUU
[+]  02 | 55555555 | 01010101010101010101010101010101 | UUUU
[+]  03 | 55555559 | 01010101010101010101010101011001 | UUUY
[+]  04 | FFFFFFFF | 11111111111111111111111111111111 | ▒▒▒▒
[+]  05 | FFFFFFFF | 11111111111111111111111111111111 | ▒▒▒▒
[+]  06 | FFFFFFFF | 11111111111111111111111111111111 | ▒▒▒▒
[+]  07 | FDDB7DDF | 11111101110110110111110111011111 | ▒▒}▒
[+] Reading Page 1:
[+] blk | hex data | binary                           | ascii
[+] ----+----------+----------------------------------+-------
[+]  00 | 000880E0 | 00000000000010001000000011100000 | ..▒▒
[+]  01 | C02A4007 | 11000000001010100100000000000111 | ▒*@.
[+]  02 | 2A3C378C | 00101010001111000011011110001100 | *<7▒
[+]  03 | 00A00003 | 00000000101000000000000000000011 | .▒..
[+] saved 12 blocks to text file lf-t55xx-1D555555-55555555-55555559-data.eml
[+] saved 48 bytes to binary file lf-t55xx-1D555555-55555555-55555559-data.bin
```

Result 2:
[... 160 lines cut ...]
    Password       : 11223344

[usb] pm3 --> lf t55xx dump
[+] Reading Page 0:
[+] blk | hex data | binary                           | ascii
[+] ----+----------+----------------------------------+-------
[+]  00 | 55555555 | 01010101010101010101010101010101 | UUUU
[+]  01 | 55555555 | 01010101010101010101010101010101 | UUUU
[+]  02 | 55555555 | 01010101010101010101010101010101 | UUUU
[+]  03 | 55555555 | 01010101010101010101010101010101 | UUUU
[+]  04 | 55555555 | 01010101010101010101010101010101 | UUUU
[+]  05 | 55555555 | 01010101010101010101010101010101 | UUUU
[+]  06 | 55555555 | 01010101010101010101010101010101 | UUUU
[+]  07 | 55555555 | 01010101010101010101010101010101 | UUUU
[+] Reading Page 1:
[+] blk | hex data | binary                           | ascii
[+] ----+----------+----------------------------------+-------
[+]  00 | 951E1BC6 | 10010101000111100001101111000110 | ▒..▒
[+]  01 | 951E1BC6 | 10010101000111100001101111000110 | ▒..▒
[+]  02 | 951E1BC6 | 10010101000111100001101111000110 | ▒..▒
[+]  03 | 951E1BC6 | 10010101000111100001101111000110 | ▒..▒
[+] saved 12 blocks to text file lf-t55xx-55555555-55555555-55555555-55555555-55555555-55555555-55555555-data.eml
[+] saved 48 bytes to binary file lf-t55xx-55555555-55555555-55555555-55555555-55555555-55555555-55555555-data.bin

```

swg0101, 2019-11-11:
And after wiping the card, here's a dump:

```
[usb] pm3 --> lf t5 det
    Chip Type      : T55x7
    Modulation     : ASK
    Bit Rate       : 2 - RF/32
    Inverted       : No
    Offset         : 33
    Seq. Term.     : Yes
    Block0         : 0x000880E0
    Downlink Mode  : default/fixed bit length
    Password Set   : No

[usb] pm3 --> lf t5 dump
[+] Reading Page 0:
[+] blk | hex data | binary                           | ascii
[+] ----+----------+----------------------------------+-------
[+]  00 | 000880E0 | 00000000000010001000000011100000 | ..▒▒
[+]  01 | FFFFFFFF | 11111111111111111111111111111111 | ▒▒▒▒
[+]  02 | FFFFFFFF | 11111111111111111111111111111111 | ▒▒▒▒
[+]  03 | FFFFFFFF | 11111111111111111111111111111111 | ▒▒▒▒
[+]  04 | FFFFFFFF | 11111111111111111111111111111111 | ▒▒▒▒
[+]  05 | FFFFFFFF | 11111111111111111111111111111111 | ▒▒▒▒
[+]  06 | FFFFFFFF | 11111111111111111111111111111111 | ▒▒▒▒
[+]  07 | FFFFFFFF | 11111111111111111111111111111111 | ▒▒▒▒
[+] Reading Page 1:
[+] blk | hex data | binary                           | ascii
[+] ----+----------+----------------------------------+-------
[+]  00 | 000880E0 | 00000000000010001000000011100000 | ..▒▒
[+]  01 | C02A4007 | 11000000001010100100000000000111 | ▒*@.
[+]  02 | 2A3C378C | 00101010001111000011011110001100 | *<7▒
[+]  03 | 00A00003 | 00000000101000000000000000000011 | .▒..
[+] saved 12 blocks to text file lf-t55xx-data-1.eml
[+] saved 48 bytes to binary file lf-t55xx-data-1.bin

```

iceman1001, 2019-11-11:
Since protect worked,  you will need to use password to dump / detect afterwards.
I notice  `lf hid clone 2`   usually never get "Success,  " message.    Is it a keyfob you are using?
They are more sensitive to positioning over antenna.

And just to verify something,  the output from 
```
hw status
hw version
```

swg0101, 2019-11-11:
No, this was on a card.

```
[usb] pm3 --> hw status
#db# Memory
#db#   BIGBUF_SIZE.............40000
#db#   Available memory........40000
#db# Tracing
#db#   tracing ................1
#db#   traceLen ...............0
#db# Currently loaded FPGA image
#db#   mode.................... LF image built for 2s30vq100 on 2019-07-31 at 15:57:16
#db# LF Sampling config
#db#   [q] divisor.............95 ( 125.00 kHz )
#db#   [b] bps.................8
#db#   [d] decimation..........1
#db#   [a] averaging...........Yes
#db#   [t] trigger threshold...0
#db#   [s] samples to skip.....0
#db# LF T55XX config
#db#            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
#db#            mode            |start|write|write|write| read|write|write
#db#                            | gap | gap |  0  |  1  | gap |  2  |  3
#db# ---------------------------+-----+-----+-----+-----+-----+-----+------
#db# fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A |
#db#     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
#db#               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |
#db#    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 |
#db#
#db# Transfer Speed
#db#   Sending packets to client...
#db#   Time elapsed............517ms
#db#   Bytes transferred.......222720
#db#   Transfer Speed PM3 -> Client = 430793 bytes/s
#db# Various
#db#   DBGLEVEL................1
#db#   ToSendMax...............-1
#db#   ToSendBit...............0
#db#   ToSend BUFFERSIZE.......2308
#db#   Slow clock..............29588 Hz
#db# Installed StandAlone Mode
#db#   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman
  compiled with Clang/LLVM Clang 9.0.0 (tags/RELEASE_900/final) OS:Android ARCH:aarch64

 [ PROXMARK3 ]

 [ ARM ]
  bootrom: RRG/Iceman/master/bd13f72a 2019-10-14 23:23:05
       os: RRG/Iceman/master/ed0bbe45 2019-11-11 05:01:15
  compiled with GCC 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

 [ FPGA ]
  LF image built for 2s30vq100 on 2019-07-31 at 15:57:16
  HF image built for 2s30vq100 on 2018-09-03 at 21:40:23

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 242211 bytes (46%) Free: 282077 bytes (54%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

[... 56 lines cut ...]
[!] Safety check: Could not detect if PWD bit is set in config block. Exits.
[usb] pm3 --> lf t55xx dump p 11223344 o
[+] Reading Page 0:
[+] blk | hex data | binary                           | ascii
[+] ----+----------+----------------------------------+-------
[=] Safety check overridden - proceeding despite risk
[+]  00 | 00107070 | 00000000000100000111000001110000 | ..pp
[+]  01 | 1D555555 | 00011101010101010101010101010101 | .UUU
[+]  02 | 55555555 | 01010101010101010101010101010101 | UUUU
[+]  03 | 55555559 | 01010101010101010101010101011001 | UUUY
[+]  04 | 00000000 | 00000000000000000000000000000000 | ....
[+]  05 | 00000000 | 00000000000000000000000000000000 | ....
[+]  06 | 00000000 | 00000000000000000000000000000000 | ....
[+]  07 | 11223344 | 00010001001000100011001101000100 | ."3D
[+] Reading Page 1:
[+] blk | hex data | binary                           | ascii
[+] ----+----------+----------------------------------+-------
[+]  00 | 00107070 | 00000000000100000111000001110000 | ..pp
[+]  01 | F00A9001 | 11110000000010101001000000000001 | ▒.▒.
[+]  02 | 2A3C378C | 00101010001111000011011110001100 | *<7▒
[+]  03 | 00A00003 | 00000000101000000000000000000011 | .▒..
[+] saved 12 blocks to text file lf-t55xx-1D555555-55555555-55555559-data-1.eml
[+] saved 48 bytes to binary file lf-t55xx-1D555555-55555555-55555559-data-1.bin

```

mwalker33, 2019-11-11:
From your results for my test.

[+]  07 | FDDB7DDF | 11111101110110110111110111011111 
is FEEDBEEF out by 1 bit
```
FDDB7DDF    11111101110110110111110111011111 
FEEDBEEF   11111110111011011011111011101111
```
I am still thinking a modulation/decode offset when password is FEEDBEEF (high bits "1")
Card also looks like it might be the T5200 and not a real T5577 (i.e. Block 3 Page 1 "00A00003")
That said, I tested the same thing on a T5200 card and it worked for me on the RDV4.

At one point of trying to make something fail, I did manage to get some miss read data, then re positioned the card and it worked fine again, with no change to the data on the card.
e.g. with the card a bit too far from my antenna and block 7 FEEDBEEF
```
[usb] pm3 --> lf t55 dump
[+] Reading Page 0:
[+] blk | hex data | binary                           | ascii
[+] ----+----------+----------------------------------+-------
[+]  00 | 00107060 | 00000000000100000111000001100000 | ..p`
[+]  01 | 1D555555 | 00011101010101010101010101010101 | .UUU
[+]  02 | 55555555 | 01010101010101010101010101010101 | UUUU
[+]  03 | 55555559 | 01010101010101010101010101011001 | UUUY
[+]  04 | 00000000 | 00000000000000000000000000000000 | ....
[+]  05 | 00000000 | 00000000000000000000000000000000 | ....
[+]  06 | 00000000 | 00000000000000000000000000000000 | ....
[+]  07 | FF76DF77 | 11111111011101101101111101110111 | .v.w
```
I think the code is doing what it should and the "failed" read means something is not correct so play safe and not continue.  Then when it can verify correctly it works as expected.

what does the tune look like ?
hw tune

swg0101, 2019-11-11:
I think it looks good?

```
[usb] pm3 --> hw tune

[=] Measuring antenna characteristics, please wait...

[=] You can cancel this operation by pressing the pm3 button
..

[+] LF antenna: 47.44 V - 125.00 kHz
[+] LF antenna: 31.58 V - 134.83 kHz
[+] LF optimal: 49.00 V - 126.32 kHz
[+] LF antenna is OK

[+] HF antenna: 29.17 V - 13.56 MHz
[+] HF antenna is OK

[+] Displaying LF tuning graph. Divisor 88 is 134.83 kHz, 95 is 125.00 kHz.

No GUI in this build!
```

iceman1001, 2019-11-11:
I am thinking the device config timings is adjusted for RDV4 and OP says he has a Pm3 Easy.  
**r 1**  is kind of near offical repo timings.

Just to make sure, you compiled for  PM3OTHER ? 
 
Since I see these messages,  which shouldn't exist on latest source.  There should only be one message,  not multiple. 
```
[!] Safety check: Could not detect if PWD bit is set in config block. Exits.
[!] Safety check: Could not detect if PWD bit is set in config block. Exits.
[!] Safety check: Could not detect if PWD bit is set in config block. Exits.
```

swg0101, 2019-11-11:
Yep, I always have PLATFORM=PM3OTHER on my build script.

```
root@localhost:~/proxmark3# git log -1
commit ed0bbe45f375d81e7b4cd38e7e6907f568c2cd01 (HEAD -> master, origin/master, origin/HEAD)
Merge: 2c0a8bbd a0168441
Author: Iceman <iceman@iuse.se>
Date:   Fri Nov 8 20:23:34 2019 +0100

    Merge pull request #471 from bogiton/master

    read_pwd_mem lua scripts update
root@localhost:~/proxmark3# git status
Refresh index: 100% (1343/1343), done.
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
```

Also seeing this interesting behavior:
```
[usb] pm3 --> lf hid clone 2
[=] Preparing to clone HID tag with ID 000000002
[usb] pm3 --> lf sea
[=] NOTE: some demods output possible binary
[=] if it finds something that looks like a tag
[=] False Positives ARE possible
[=]
[=] Checking for known tags...
[=]
[+] HID Prox TAG ID: 000000002 (1) - Format Len: 37bit - OEM: 000 - FC: 0 - Card: 1

[+] Valid HID Prox ID found!

[+] Chipset detection: T55xx
[=] Hint: try `lf t55xx` commands
[usb] pm3 --> lf t5 det
    Chip Type      : T55x7
    Modulation     : FSK2a
    Bit Rate       : 4 - RF/50
    Inverted       : Yes
    Offset         : 32
    Seq. Term.     : No
    Block0         : 0x00107060
    Downlink Mode  : default/fixed bit length
    Password Set   : No

[usb] pm3 --> lf t5 pr n 90099009
[=] Checking current configuration
[+] Wrote new password
[!] Failed to validate the password write. aborting.
[!] Command failed. Did you run `lf t55xx detect` before?
[usb] pm3 --> lf t5 det
    Chip Type      : T55x7
    Modulation     : FSK2a
    Bit Rate       : 4 - RF/50
    Inverted       : Yes
    Offset         : 32
    Seq. Term.     : No
    Block0         : 0x00107060
    Downlink Mode  : default/fixed bit length
    Password Set   : No

[usb] pm3 --> lf t5 read b 7
[+] Reading Page 0:
[+] blk | hex data | binary                           | ascii
[+] ----+----------+----------------------------------+-------
[+]  07 | C804C804 | 11001000000001001100100000000100 | ▒.▒.
[usb] pm3 --> lf t5 pr n 10011001
[=] Checking current configuration
[+] Wrote new password
[+] Validated new password
[+] Wrote modified configuration block
[!] Safety check: Could not detect if PWD bit is set in config block. Exits.
[=] Block0 write detected, running `detect` to see if validation is possible
    Chip Type      : T55x7
    Modulation     : FSK2a
    Bit Rate       : 4 - RF/50
    Inverted       : Yes
    Offset         : 32
    Seq. Term.     : No
    Block0         : 0x00107070
    Downlink Mode  : default/fixed bit length
    Password Set   : Yes
    Password       : 10011001

[+] New configuration block 00107070 password 10011001
[+] Success, tag is locked
[usb] pm3 --> lf t5 det
[!] Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
[usb] pm3 --> lf t5 det p 10011001
    Chip Type      : T55x7
    Modulation     : FSK2a
    Bit Rate       : 4 - RF/50
    Inverted       : Yes
    Offset         : 32
    Seq. Term.     : No
    Block0         : 0x00107070
    Downlink Mode  : default/fixed bit length
    Password Set   : Yes
    Password       : 10011001

[usb] pm3 --> lf t5 read b 7 p 10011001 o
[+] Reading Page 0:
[+] blk | hex data | binary                           | ascii
[+] ----+----------+----------------------------------+-------
[=] Safety check overridden - proceeding despite risk
[+]  07 | 10011001 | 00010000000000010001000000000001 | ....
[usb] pm3 -->

```

mwalker33, 2019-11-11:
Dusted off my old pm3easy, it has an older version of the official repo on it.
reading the hid 2 clone card, I get the same bit offset on block 7 as you got with the feedbeef "password" data 
i.e. FDDB7DDF
I tried a few other passwords and while not exhaustive testing, I seem to get read offset errors when the high bit is set, others seem fine. 

I then moved the card off the lf antenna and sat it below that (over the hf antenna).  When there, the reads were perfect.

Can you:
clear the card
lf t55 det
lf hid clone 2
lf t55 det
lf t55 write b 7 d feedbeef

then with the card placed at differnet places around the antenna, 
lf t55 det
lf t55 read b 7

and see if any give the correct result.

swg0101, 2019-11-11:
Interestingly, it looks like all of it gave the correct result using this method unless I am very far off the antenna... I wonder if this is perhaps a "weak" card... :\

mwalker33, 2019-11-11:
To be honest, after all my playing with the T5577 and T5200 (5577 clones), while I feel have a good idea of whats going on, I tend to just put it down to the way things tend to work.  So the trick is to find a good place where put place your cards and fobs for the best results.  Once you find that sweet spot, for your antenna, work with that. (e.g. on the pm3 easy and the 4305 fobs I got best results by holding the fob at right angles to the antenna)
On the RDV4 they worked out that low Q antennas gives better read/write to the T5577, while High Q gives better results for reading a "real" LF card/clones (better read distance). 

Are you happy that you have a valid reason and fix for your issue ?

swg0101, 2019-11-11:
Experimenting with this a bit further, it looks like that when the password has the high bit set, it tends to be more difficult to write than one with zeros in front (e.g. 00112233). I did try various placements, and it seems like lifting the car in the air 1-2cm did result in a good write 80% of the time. Iceman mentioned there may be some timing issues with the PM3 easy on this build - are your timing values the same?

mwalker33, 2019-11-11:
As iceman said the long leading 0 values are very close (if not the same) os the original repo.
```
fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A |
    long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
```
So you could try setting the fixed bit (r 0) to the same as long leading.

swg0101, 2019-11-12:
Looking at the official repo, they define the timings as:

```
/* T5577C timing datasheet for Fixed-Bit-Length protocol (defualt):
 * Type                  |  MIN   | Typical |  Max   |
 * Start_Gap             |   8*8  |   15*8  |  50*8  |
 * Write_Gap Normal mode |   8*8  |   10*8  |  20*8  | 
 * Write_Gap Fast Mode   |   8*8  |   10*8  |  20*8  |
 * Write_0   Normal mode |  16*8  |   24*8  |  32*8  | 
 * Write_1   Normal mode |  48*8  |   56*8  |  64*8  | 
 * Write_0   Fast Mode   |   8*8  |   12*8  |  16*8  |
 * Write_1   Fast Mode   |  24*8  |   28*8  |  32*8  |
*/
```

The write 1 normal mode has a minimum of 48 cycles, while the timings defined in this repo for write 1 is 47 cycles. Perhaps this is potentially the reason why a write 1 would be interpreted as a shorter write 0  command?

iceman1001, 2019-11-12:
Since you are using the fixed bit length (default) setting on a Pm3 easy,  I would program the deviceconfig for r 0 ,  to match
`r 1 ==   long leading reference |  31 |  20 |  18 |  50 |  15 |`

Notice this change will not persist as possible on a rdv4,  but stay in memory until you power down your device.   If you want to make it more permenant ,  change the armsrc/lfsampling.c file. and flash.

To be honest,  we should have a macro guard for PM3_OTHER, to use old timings...  @mwalker33  up for it?

mwalker33, 2019-11-12:
@iceman Yep can do,  but will take a little time to get to it (working 7 days a week for the next 3 weeks)
then good to go (to catch up on a few items).

iceman1001, 2019-11-12:
No worries,  I sort it out :)

iceman1001, 2019-11-12:
@swg0101    pull latest and test.  I adjusted so when compiling with PM3OTHER,  it should have timings matching a non RDV4 device.  ref https://github.com/RfidResearchGroup/proxmark3/commit/93c9dc571ff30c45f7b6f9ab008fb80ae082e7f6

swg0101, 2019-11-12:
Looks good, thanks!

closed 2019-11-12 by swg0101

iceman1001, 2019-11-12:
What is your output from `hw status`  when you flashed latest?

swg0101, 2019-11-12:
[image]

Like this...

iceman1001, 2019-11-12:
Good, that would be correct timings for non rdv4.   Does your LF tag work better now?

swg0101, 2019-11-12:
It does, however, some cards still tend to puke when the first byte has the high bit set... I am guessing it's probably just the quality of these cards that are questionable...

iceman1001, 2019-11-12:
ok,  have you tried  with   `lf t55 read b 1 r 1` 
In order to see if long leading zero works better....

swg0101, 2019-11-12:
I think the main issue with this is on PSK2a. On ASK, it always work properly.

mwalker33, 2019-11-12:
> ok, have you tried with `lf t55 read b 1 r 1`
> In order to see if long leading zero works better....

Dont forget to get the card to try to use r1 you need to send the config to block 3 page 1 90000400
Based on one of the dumps the block 3 page 1 config, I think the card is a t5200 clone of the t5577 so may or may not support downlink modes.  From testing I got the impression r 1 did work but not r 2 or 3

swg0101, 2019-11-12:
@mwalker33 I think that's my experience as well... unfortunately, I don't think I have any real T5577s to test with (the only one I had was a Keysy but ended up returning that one)...

mwalker33, 2019-11-12:
If you are feeling techy.
Bring up the data plot
Then run the 
lf t55 read b 0 
then
 lf t55 read b 7
Take not if the start of modulation drifts much.
If you then move the card and repeat.

If the start drifts too much then the offset selected in the detect code will end up in the wrong spot.
Psk and fsk are a little harder to lock onto the start.
When the antenna voltage is higher the signal can tend to clip thus adding to the challenge.  Eg on my rdv4 high q was about 70 and t55xx had trouble. Then on low q I it's in the 30s with much better t5577 use.
On my pm3 easy i get high 30s.
By moving the card away from the reader a little the signal lowers and the graph looks better and t5577 use improves.

Last night I flashes my easy with the latest rrg and retested and could repeated both good and bad reads, with  both sets of timings.

mwalker33, 2019-11-12:
> @mwalker33 I think that's my experience as well... unfortunately, I don't think I have any real T5577s to test with (the only one I had was a Keysy but ended up returning that one)...

They should be ok.  In this issue I have been testing both with the same results.
When I first got the easy I had some issues with 55xx reads.  Then put the unit into a case such that the cards were held off the antenna a little.  While fsk still had some read issues it was much better.
Note. I feel that the write is ok, but the read is shifted a little.  Ie when the card starts to modulate the data.

mwalker33, 2019-11-12:
Side note.  If you interested in my view on the tech side, let me know your email.  It's a little too much and will drift to much for an issue here.

swg0101, 2019-11-12:
Love to, here's my email: softwareguy@gmail.com