RfidResearchGroup/proxmark3 #474: PM3 waiting to appear

pclever1, 2019-11-13:
**Describe the bug**
After installing this project on my Mac and flashing the device updates I am unable to use the command "pm3". I am able to continue flashing the bootloader but the pm3 command alone just sits there "Waiting for Proxmark3 to appear".

Can you tell me if I did anything wrong? I am using a pm3 v2.

**To Reproduce**
Steps to reproduce the behavior:
1. I followed this guide using the stable branch: [Here](https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Mac-OS-X-Homebrew-Installation-Instructions.md)
2. Ran the "pm3-flash-all" command
3. Ran command "pm3" then saw no response

iceman1001, 2019-11-13:
Did you compile the project for non rdv4 ?   
see  https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/4_Advanced-compilation-parameters.md

pclever1, 2019-11-13:
Thank you I did not try that. I read the documentation and tried it out but I am still having a problem. Can you advise please?

I navigated on my Mac to /usr/local/share/proxmark3
I created a file named "Makefile.platform" and added "PM3OTHER" to it.
I then ran "make clean && make install" and I got an error.

Am I in the wrong folder?

aotd1, 2019-11-13:
The same problem on 24638a832d0f6a79d9a44736b13cf2b15db37926
I have proxmark3 evo (with 512 from aliexpress). Official repo works fine.

- Compiled from brew: `HOMEBREW_PROXMARK3_PLATFORM=PM3OTHER brew install proxmark3 --HEAD`
- Flash new bootloader manually
- Use pm3-flash-all command successfully
- Device available only in bootloader mode. If I release button /dev/tty.* device disappears

I'll try to flash from new commits today or can provide some debug info if you send instructions)

iceman1001, 2019-11-13:
Regarding PM3 Evo,  see note inside the linked document in my previous post.  It will most likely not work on this repo because of strange pins for the leds.  

Regarding OP's mac issue,  you should do like @aotd1  did.   See https://github.com/RfidResearchGroup/homebrew-proxmark3/blob/master/README.md  for details.

And that is about it what I can answer when it comes to OS X.

closed 2019-11-13 by iceman1001

pclever1, 2019-11-13:
Thanks again for the help, I tried the following and am still experiencing the same issue:

1. "export HOMEBREW_PROXMARK3_PLATFORM=PM3OTHER"
2. "brew reinstall proxmark3"
 - Noticed the println "make all PLATFORM=PM3OTHER"
3. "pm3-flash-all"

Is there something else I need to do?

iceman1001, 2019-11-13:
try using the direct executable,  and not rely on the script to find a usb-enumeration.

pclever1, 2019-11-13:
What do you mean by direct executable, can I do that on my Mac? I am pretty sure the issue is with the device being flashed. The flash all completes with no errors but when I plug the device into my computer all the LEDs are off (from what I remember on the V2 one LED should be on when it is ready to use).

iceman1001, 2019-11-13:
What is your output from the flasher?  

running the client.
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Mac-OS-X-Homebrew-Installation-Instructions.md#run-the-client

mwalker33, 2019-11-13:
@iceman1001, do you know the there is a simple way on the mac to list usb devices and ID? The Id  may be different, thus new enumeratuon thus different port.  
Eg. This week when I changed the firmware on the easy clone from original repo to rrg, at one point the boot loader had a different usb port to the pm3 firmware.  Once everything was updated it was all on the same "new" comm port and all good.  But I did need to re-assign the port to the new id so it would have the same bootloader port and firmware port to get a clean flash.  Note this was on win10.

iceman1001, 2019-11-13:
on Win10 the new usb enumeration is just about ctrl-c  the script which is stuck,  and run the script again,  where it now correctly identifies the new com port.   
on linux,  you have to kill modem manager and use  lsusb to find new /dev/tty..
on mac,  well it should be very similar to linux but os x has  KEX problems / usb enumeration problems  since they use different comport drivers.  They have more than one.   

Normally you can search the forum / GH issues for OSX and find repeatingly these kinds of problems.

pclever1, 2019-11-13:
When running flash-all I see the following:
 - Waiting for Proxmark3 to appear on /dev/tty.usbmodemiceman1
 - Found
- Loading ELF file
- Loading ELF file
- Flashing...
- All done.

It seems to flash no problem but I can't tell if it is the right software for my device. As I mentioned before I re-compiled the build for PM3OTHER but im still thinking the issue is likely with the device itself not the computer. (again its a RV2) 

If I run "proxmark3 /dev/tty.usbmodemiceman1" I get this error:
- ERROR: invalid serial port /dev/tty.usbmodemiceman1

Please let me know if you have any other ideas for me to try!
Thanks for all the good ideas

iceman1001, 2019-11-13:
You have to find the new usb enumeration since your  usbmodemiceman1 doesn't exist...

pclever1, 2019-11-13:
I just ran the command "ioreg -p IOUSB" which lists active USB devices on a Mac and I didn't see anything resembling the PM3. If the problem is the usb port how is your software able to flash to the pm3?

iceman1001, 2019-11-14:
the flash-all updates your bootrom and fullimage,   if you were on offical repo firmware before, your device will now get a new enumeration.   
However you were on rrg/iceman or on iceman repo before? 

Anyway, I can't help you.  OSX and homebrew isn't my turf.

thomastweets, 2019-11-14:
Same problem here. Flashing all went fine ("All done.") and is repeatable when PM3 is put into boot loader mode again. However, after normal booting (just attaching PM3 to USB without holding a button) the blue 'D' LED is only on for a second and then turns off. The booting seems to crash.

The '/dev/tty.usbmodemiceman1' device only shows up when booting in boot loader mode (`ls /dev/tty*`). No serial device shows up when booting normally which leaves the client/pm3 program searching for the right device.

Thanks a lot for your help!

pclever1, 2019-11-14:
I don't remember what firmware was on my pm3 before I flashed this one (rrg) but that is a good idea. I will try to get the software installed on my windows box tonight and see if I can flash it differently. Thanks again, I will post my results later.

pclever1, 2019-11-14:
Also can someone tell me how to specify PM3OTHER in the windows installation?

iceman1001, 2019-11-14:
follow the instructions in the link previously posted.

pclever1, 2019-11-16:
Hey I am still trying to get this installed on my v2 PM3. I tried everything I could think of on OSX then today switched to try the install in Ubuntu (not using a VM). I tried following the install guide but ran into a new issue when running "make all" because **the package "gcc-arm-none-eabi" is not supported by Ubuntu 18.04.** 

For Linux:
- Do I need to use a specific version of Ubuntu?
- Or are there any additional steps to install that package?

For Mac:
- Is there anything else I can try?

thomastweets, 2019-11-16:
Just a short update: after cloning the repository and compiling in there, everything worked out for me (newest boot loader and newest image flashed fine; tty device shows up; client can connect). It seems that it was some homebrew hiccup using the `PM3OTHER` flag.

Again, thanks a lot for your work!

pclever1, 2019-11-16:
On my Mac everything looks to be working fine through the flash process. I set the `PM3OTHER` flag and flashed it again. The issue on the Mac is that when running the `pm3` command it doesn't find the device. 

Again I am wondering if the flash was correct or not and here's why:
- In Ubuntu if I run `sudo dmesg | grep -i usb` it only shows the proxmark if it is in flash mode.
- If I just let it power on normally I don't see the proxmark at all

Has anyone seen this happen before or have any advice on more things I can try?
Thank you!!

thomastweets, 2019-11-16:
@pclever1 that is the exact same behaviour I observed. My assumption is that something went wrong while flashing the newest firmware initially. I can only describe what worked for me without a good explanation why that is:

I cloned the repository and checked out the last firmware before the switch to the `PM3OTHER` flag system with `git checkout  3cf64f9` (https://github.com/RfidResearchGroup/proxmark3/commit/3cf64f9). In the `Makefile.platform` file I set `PLATFORM=PM3EASY512` and compiled. Flashing and connecting went fine. By then I thought that I would be stuck with `3cf64f9`. Out of curiosity I changed back to `HEAD`, compiled and flashed again and then everything worked fine although I had compiled and flashed the same code before.

Good luck, maybe this helps!

Btw. this seems to be a problem conceptually seen before, see also here https://github.com/RfidResearchGroup/proxmark3/issues/322

iceman1001, 2019-11-16:
Lets recap,
OSX users keep on having issues with their Proxmark3 devices (mostly non RDV4). Usually when going from older FW (official repo or iceman fork).   Once they manage to get flashed latest here they seem to have no issue any longer.

I suggest using the shell scripts (`pm3`) but the executables (`./client/proxmark3`) direct. 
This way you can specify which tty / serial port that got assigned to your device.  The shell scripts tries to detect and guess which seems to fail on OSX in these specific situations.

Right now its just a guessing game what actually is going on for OSX users.

iceman1001, 2019-11-17:
In #475   the pm3.sh shell script didn't identify devices with older official repo fw.   @doegox  just pushed a fix.  That fix applies for linux.   Question is if the detection inside pm3.sh for OS X has the same problems,  which might explain the problems experienced here.

pclever1, 2019-11-17:
Hello again,
After I flashed my PM3 in OSX I tried to use it in Ubuntu. If I run `sudo dmesg | grep -i usb` it only shows the proxmark if it is in flash mode. If I just let it power on normally I don't see the proxmark at all when using this command. 

Is this normal behavior or do you think the flash process in OSX is causing this issue?

aotd1, 2019-11-17:
Thanks to @thomastweets, it works! =) Looks like something broken with homebrew
@pclever1 it works for me on OS X:
```bash
git clone git@github.com:RfidResearchGroup/proxmark3.git
cd proxmark3
#add file Makefile.platform with `PLATFORM=PM3OTHER`
make clean && make all
sudo ./pm3-flash #In bootrom mode to flash new bootloader
#Poweroff proxmark3 and switch it on in boot mode
sudo ./pm3-flash-all #To reflash it all
./pm3
```
These steps are too redundant, but it works for me on the fresh master

pclever1, 2019-11-17:
THANK YOU!
I didn't think to try compiling it without homebrew. I just did exactly as you wrote and everything is working now! Thanks everyone for the help

doegox, 2019-11-18:
> In #475 the pm3.sh shell script didn't identify devices with older official repo fw. @doegox just pushed a fix. That fix applies for linux. Question is if the detection inside pm3.sh for OS X has the same problems, which might explain the problems experienced here.

Update: I pushed the same fix for OS X to be able to detect more easily devices with official firmwares < 2018.02

ersingencturk, 2022-02-23:
if you end up here.. `brew reinstall proxmark3 --with-generic`  is the way to fix this . this happens because you have a proxmark3 easy device  but you have installed proxmark3 rdv4 with brew. that's why your device is flashed with rdv4 firmware and not booting. after re installing correct version of your device, re flash your device and it will work this time.