RfidResearchGroup/proxmark3 #475: PROXMARK 3 Dev Kit 2 with factory firmware is not automatically detected by pm3

fishilico, 2019-11-16:
**Describe the bug**
I am using an Elechouse PROXMARK 3 Dev Kit 2. After building the project (with `PLATFORM=PM3OTHER`), no `pm3` commands (including `pm3-flash-fullimage`) detected this Proxmark:
```
pm3-flash-fullimage
[=] Waiting for Proxmark3 to appear...
```

After flashing the bootloader and the full image with `proxmark3 /dev/ttyACM0 --flash ...` commands (as described in the documentation), the issue disappeared and I am now able to use all `pm3` commands directly.

Therefore commands `pm3` seem to fail to detect the "official"/factory firmware which was initially present on the Elechouse PROXMARK 3 Dev Kit 2 even though it is compatible with command `proxmark3`.

**To Reproduce**
Steps to reproduce the behavior:
1. Build the project with `PLATFORM=PM3OTHER` and the latest master branch (commit e0d4e69cb9c2)
2. Buy a Elechouse PROXMARK 3 Dev Kit 2
3. Connect the Proxmark to USB port, following instructions in the documentation 
4. Run `pm3` or `pm3-flash-fullimage`
5. See error

**Expected behavior**
`pm3-flash-fullimage` invokes `proxmark3 /dev/ttyACM0 --flash ...`

**Desktop (please complete the following information):**
 - OS: Arch Linux on x86-64 CPU (Lenovo Thinkpad laptop)
 - Kernel logs before upgrading the firmware:
```
Nov 15 19:22:59 test kernel: usb 1-2: new full-speed USB device number 7 using xhci_hcd
Nov 15 19:23:04 test kernel: usb 1-2: New USB device found, idVendor=2d2d, idProduct=504d, bcdDevice= 0.01
Nov 15 19:23:04 test kernel: usb 1-2: New USB device strings: Mfr=1, Product=0, SerialNumber=0
Nov 15 19:23:04 test kernel: usb 1-2: Manufacturer: proxmark.org
Nov 15 19:23:04 test kernel: cdc_acm 1-2:1.0: ttyACM0: USB ACM device
```
- Kernel logs after upgrading the firmware:
```
Nov 15 19:30:31 test kernel: usb 1-2: new full-speed USB device number 13 using xhci_hcd
Nov 15 19:30:31 test kernel: usb 1-2: New USB device found, idVendor=9ac4, idProduct=4b8f, bcdDevice= 1.00
Nov 15 19:30:31 test kernel: usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Nov 15 19:30:31 test kernel: usb 1-2: Product: proxmark3
Nov 15 19:30:31 test kernel: usb 1-2: Manufacturer: proxmark.org
Nov 15 19:30:31 test kernel: usb 1-2: SerialNumber: iceman
Nov 15 19:30:31 test kernel: cdc_acm 1-2:1.0: ttyACM0: USB ACM device
```

Notice the new `usb 1-2: Product: proxmark3` line in these logs.
`pm3` seems to match this using udev properties: https://github.com/RfidResearchGroup/proxmark3/blob/e0d4e69cb9c263c6728ee7fb49ff1a5eeb113f5f/pm3#L24

These logs could help modifying `pm3` in order to detect a USB device with `Manufacturer: proxmark.org` without `Product: proxmark3`.

Moreover, when flashing the bootloader, some warning messages appeared:
```
proxmark3 /dev/ttyACM0 --flash --unlock-bootloader --image bootrom/obj/bootrom.elf                                                       15/11 19:29
[=] Session log /home/me/.proxmark3/log_20191115.txt
[+] About to use the following file:          
[+]     bootrom/obj/bootrom.elf          
[+] Waiting for Proxmark3 to appear on /dev/ttyACM0           
.Found           
[+] Entering bootloader...           
[+] (Press and release the button only to abort )          
[+] Waiting for Proxmark3 to appear on /dev/ttyACM0           
........... Found           
[!!] ====================== OBS ! ===========================================           
[!!] Note: Your bootloader does not understand the new CMD_BL_VERSION command            
[!!] It is recommended that you first update your bootloader alone,            
[!!] reboot the Proxmark3 then only update the main firmware 
          
[=] Available memory on this board: UNKNOWN 
          
[!!] ====================== OBS ! ======================================           
[!!] Note: Your bootloader does not understand the new CHIP_INFO command            
[!!] It is recommended that you first update your bootloader alone,            
[!!] reboot the Proxmark3 then only update the main firmware 
          
[=] Permitted flash range: 0x00100000-0x00140000          
[+] Loading ELF file bootrom/obj/bootrom.elf           
[+] Loading usable ELF segments:          
[+]    0 : V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94          
[+]    1 : V 0x00200000 P 0x00100200 (0x00000e28->0x00000e28) [R X] @0x298          
          
          
[+] Flashing... 
          
[+] Writing segments for file: bootrom/obj/bootrom.elf          
[+]  0x00100000..0x001001ff [0x200 / 1 blocks]          
. OK           
[+]  0x00100200..0x00101027 [0xe28 / 8 blocks]          
........ OK           

          
[+] All done.           

Have a nice day!          
```

iceman1001, 2019-11-16:
I am not sure what the issue is. 
You flashed bootrom and you during flashing you got warning messages correctly informing you that your device has the wrong bootrom  then the flashing went as it should.

fishilico, 2019-11-16:
> I am not sure what the issue is.
> You flashed bootrom and you during flashing you got warning messages correctly informing you that your device has the wrong bootrom then the flashing went as it should.

Flashing with `proxmark3 /dev/ttyACM0 --flash ...` worked even though `pm3`, `pm3-flash-fullimage`, etc. were all stuck in `[=] Waiting for Proxmark3 to appear...`. I added the log at the end of my message in order to be clear that the issue I am reporting is not about command `proxmark3` but about `pm3`.

doegox, 2019-11-16:
Thanks Nicolas!
I recognize the issue we've seen yesterday during the workshop.
I'll have a look tomorrow.

doegox, 2019-11-17:
Fixed, actually pm3 works also on official fw but you had an issue because very probably your official fw was older than 2018.02.

doegox, 2019-11-17:
BTW the warning messages are there on purpose, to make sure ppl change their bootloader

closed 2019-11-17 by doegox

fishilico, 2019-11-17:
Thanks for the quick fix! It was indeed the issue that I got during your workshop at GreHack.

commit 5dc2c7c3 (2019-11-17): pm3 serial detection on Linux more robust. Fixes #475

Official firmwares < 2018.02 had a different USB Model string (PM3<>proxmark3).
Better to detect based on the vendor string (proxmark.org) which is more stable,
to ease reflashing devices with old firmwares to the RDV4 firmware.
Thanks @fishilico for having pointed it out.