RfidResearchGroup/proxmark3 #482: Annoying error messages running the client on Linux

evazzoler, 2019-11-26:
[image]

Some GTK errors appearing when I start the client. I searched the internet and found little documentation. It seems the code try to load a not existing theme (why?).
The plot command works apparently good.

doegox, 2019-11-26:
Nothing we can do about it, it's due to missing stuff in your desktop theme.
See e.g. #386

closed 2019-11-26 by doegox

evazzoler, 2019-11-26:
And why the executable are looking for GUI stuff when it is a CLI?
Just notice that the executable coult search for the wrong stuff...
Stuff, in example, tipically not existing on a Ubuntu 18.04 desktop environment.

doegox, 2019-11-26:
the CLI has GUI components, e.g. when running "data plot".
> Stuff, in example, tipically not existing on a Ubuntu 18.04 desktop environment.
can you elaborate?

evazzoler, 2019-11-26:
It seems the code try to force a specific theme instead "current theme".
It happens on a clean install where other GUI applications are working well.

evazzoler, 2019-11-26:
Data plot is working perfectly despite these errors.

doegox, 2019-11-26:
what is your desktop environment?

evazzoler, 2019-12-09:
My desktop environment is Gnome on Ubuntu 18.04.3. The theme is Ambiance.
The same issue happens with other themes.
The issue don't happens if I launch proxmark client as root (sudo).

Maybe the client search the theme in the location where root user theme is located, not the user one, and has no rights there.

I found some similar issues on google pointing to github issue tracker of other projects and developer closed them writing "fixed for dark theme" or "fixed". This means it was a code issue and not user environment issue / user side issue...