RfidResearchGroup/proxmark3 #486: Can't connect the BT with phone or linux laptop

evazzoler, 2019-12-05:
Hi,
I have the BT support enabled and compiled
```
 [ PROXMARK3 RDV4 Extras ]          
  FPC USART for BT add-on support: present  
```  

but the problem is before this...

With my Huawei P30 Pro (I tried with a P9 Pro, too) I scan the BT and find the "PM3_RDV4.0", it asks for the PIN and I enter 1234, then the PM3 is moved in the paired device list but it is disconnected and there is no way for connecting it. The led is always blinking fast.

The same with my Laptop (Thinkpad T-460) running Ubuntu Linux 18.04.03 LTE: it goes on the pared list but disconnected.

I don't know ho to get low level logs on the PM3 side. If someone tells me how, I can provide them.
Maybe the BT module is defective?

iceman1001, 2019-12-05:
When it comes to bt connection on the phone,   
as stated in the [manual](https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/bt_manual_v10.md#1-connecting-your-rdv40-with-bluetooth-on-mobile-phone-or-computer)

the leds doesn't got solid until you start the RFID app.

evazzoler, 2019-12-05:
Both phone and laptop have "disconnected" label near the PM3 entrty no tty is created and on the phone the Walrus app has no devices to open.

iceman1001, 2019-12-05:
You would need to post more information, like outputs from what you have done and which commands.

evazzoler, 2019-12-05:
There are no command I can issue on a smartphone... Only a UI in the Android OS for managing the BT.
If needed I can post here some screenshots, but they probably won't add informations to the initial explanation.

Is there a way for obtaining a log from the BT stack PM3 side?

iceman1001, 2019-12-05:
Make a video?

evazzoler, 2019-12-06:
Ok, [this](https://youtu.be/oaLLpmR-W4A) is the viedeo. (Sorry for the external link but Github don't permit the upload of videos)
After the pairing the PM3 is disconnected. At that point I pair a BT earphone (you can hear the command I issue for put it in pairing mode). After the pairing you see "connected" (plus information about the type of the connection) near the name of the earphone.
If "connected" is missing, that BT device is paired but not connected.
In fact the blue led remains in fast blinking state.

evazzoler, 2019-12-06:
And this is about Linux PC
```
enrico@thinkpad-t460:~$ sudo hcitool scan
[sudo] password di enrico: 
Scanning ...
	20:19:05:06:23:20	n/a
enrico@thinkpad-t460:~$ sudo rfcomm bind rfcomm0 20:19:05:06:23:20
enrico@thinkpad-t460:~$ cd /opt/pm3/proxmark3/client/
enrico@thinkpad-t460:/opt/pm3/proxmark3/client$ ./proxmark3 /dev/rfcomm0 

██████╗ ███╗   ███╗ ████╗      ...iceman fork          
██╔══██╗████╗ ████║   ══█║       ...dedicated to RDV40           
██████╔╝██╔████╔██║ ████╔╝           
██╔═══╝ ██║╚██╔╝██║   ══█║     iceman@icesql.net          
██║     ██║ ╚═╝ ██║ ████╔╝    https://github.com/rfidresearchgroup/proxmark3/          
╚═╝     ╚═╝     ╚═╝ ╚═══╝  pre-release v4.0          

Support iceman on patreon - https://www.patreon.com/iceman1001/          
                 on paypal - https://www.paypal.me/iceman1001          

          
[=] Session log /home/enrico/.proxmark3/log_20191206.txt
[=] Using UART port /dev/rfcomm0           
[!] Communicating with Proxmark3 device failed           
[!!] ERROR: cannot communicate with the Proxmark
```
And a couple of screenshots about the BT GUI Manager.

[image]
[image]

evazzoler, 2019-12-06:
Ok, here you can look at the "manual pairing" procedure I tried for debug purposes.
You can see that the device connects for a half second then disconnects by itself.

(I don't know why the prompt of bluetoothctl is the name of my keyboard, but I verified that this not affects the commands and other devices can pair and connect without problems)

```
enrico@thinkpad-t460:~$ bluetoothctl
[NEW] Controller 44:85:00:A1:E4:78 ThinkPad-T460 [default]
[NEW] Device 20:19:05:06:23:20 PM3_RDV4.0
[NEW] Device C8:07:D1:14:37:BC Expert_C807D11437BC
[NEW] Device 00:1F:20:EA:EF:3C Logitech K810
[NEW] Device 20:20:00:06:A9:C7 Bluetooth 3.0 Keyboard
[NEW] Device 00:07:61:96:46:5A Bluetooth Laser Travel Mouse
[NEW] Device E1:D0:0D:4B:18:59 MX Anywhere 2
[NEW] Device 38:18:4C:11:6F:95 WH-1000XM3
Agent registered
[Logitech K810]# pair 20:19:05:06:23:20
Attempting to pair with 20:19:05:06:23:20
[CHG] Device 20:19:05:06:23:20 Connected: yes
Request PIN code
[Logi1m[agent] Enter PIN code: 1234
[CHG] Device 20:19:05:06:23:20 UUIDs: 00001101-0000-1000-8000-00805f9b34fb
[CHG] Device 20:19:05:06:23:20 ServicesResolved: yes
[CHG] Device 20:19:05:06:23:20 Paired: yes
Pairing successful
[CHG] Device 20:19:05:06:23:20 ServicesResolved: no
[CHG] Device 20:19:05:06:23:20 Connected: no
[Logitech K810]# info 20:19:05:06:23:20
Device 20:19:05:06:23:20 (public)
	Name: PM3_RDV4.0
	Alias: PM3_RDV4.0
	Class: 0x00001f00
	Paired: yes
	Trusted: no
	Blocked: no
	Connected: no
	LegacyPairing: yes
	UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)
[Logitech K810]# trust 20:19:05:06:23:20
[CHG] Device 20:19:05:06:23:20 Trusted: yes
Changing 20:19:05:06:23:20 trust succeeded
[Logitech K810]# unblock 20:19:05:06:23:20
Changing 20:19:05:06:23:20 unblock succeeded
[Logitech K810]# connect 20:19:05:06:23:20
Attempting to connect to 20:19:05:06:23:20
[CHG] Device 20:19:05:06:23:20 Connected: yes
[CHG] Device 20:19:05:06:23:20 ServicesResolved: yes
Failed to connect: org.bluez.Error.InProgress
[CHG] Device 20:19:05:06:23:20 ServicesResolved: no
[CHG] Device 20:19:05:06:23:20 Connected: no
[Logitech K810]# connect 20:19:05:06:23:20
Attempting to connect to 20:19:05:06:23:20
Failed to connect: org.bluez.Error.Failed
[Logitech K810]# 
```

hiviah, 2019-12-06:
I encountered similar issue. I can make Blueshark pair, but for some reason the host sends disconnect to blueshark (seen in wireshark).

As you can see  from the commands below, it's paired, but after about 3 seconds since connect it disconnects. I even put the trusted flag on it.

    [bluetooth]# pair 20:19:05:06:27:09
    Attempting to pair with 20:19:05:06:27:09
    [CHG] Device 20:19:05:06:27:09 Connected: yes
    [CHG] Device 20:19:05:06:27:09 UUIDs: 00001101-0000-1000-8000-00805f9b34fb
    [CHG] Device 20:19:05:06:27:09 Paired: yes
    Pairing successful

    [bluetooth]# paired-devices
    Device 20:19:05:06:27:09 PM3_RDV4.0
    [bluetooth]# trust 20:19:05:06:27:09
    [CHG] Device 20:19:05:06:27:09 Trusted: yes

    [bluetooth]# connect 20:19:05:06:27:09
    Attempting to connect to 20:19:05:06:27:09
    [CHG] Device 20:19:05:06:27:09 Connected: yes
    Failed to connect: org.bluez.Error.NotAvailable
    [CHG] Device 20:19:05:06:27:09 Connected: no

[PCAPNG packet capture of bluetooth for the connect-disconnect](https://qabs.cz/o/blueshark-proxmark-disconnect2.pcapng)

hiviah, 2019-12-06:
Could it be possibly caused that the flat wire between Blueshark and Proxmark doesn't sit correctly maybe? Is there any way to test it? Or it shouldn't affect the connecting?

hiviah, 2019-12-06:
Full rebuild (`make clean && make`), reflash at least now makes it work with Nexus 5X phone, result in Linux is still the same (disconnect immediately after connect). Could be something like ModemManager ruining this? Though I don't see any serial ports created in dmesg.

Sdptool returns empty profile (works on other BT devices):

    # sdptool browse 20:19:05:06:27:09
    Browsing 20:19:05:06:27:09 ...

hiviah, 2019-12-06:
Seems that `rfcomm bind rfcomm0 20:19:05:06:27:09` bind makes it work on Linux, but the messages in `dmesg` are confusing:

    Bluetooth: TIOCGSERIAL is not supported

It was few years since I read the BT specs, but shouldn't the device declare serial profile by itself? It would probably make it work out-of-the-box without rfcomm. Also since we'd like to survive reboot, this might be better (but untested) in `/etc/bluetooth/rfcomm.conf`:

    rfcomm0 {
    bind yes;
    
    device 20:19:05:06:27:09;
    
    channel 1;
    # Name of this device
    comment "Proxmark Blueshark"
    }

evazzoler, 2019-12-07:
It seems we have different problems:

1. You get "Failed to connect: org.bluez.Error.NotAvailable" and I get "Failed to connect: org.bluez.Error.Failed"
2. I never connected the phone (tried 4 different phones by 4 different vendors)
3. I can't connect with a windows machine, too

hiviah, 2019-12-07:
@evazzoler  The error is different, but it seems to connect for you, and then disconnect, as it did for me. This was the most confusing thing in the whole process. 

What the bluetoothctl says about connection status did not seem to matter for the connection (the explicit connect is not useful). I needed to create the `rfcomm0` device, then run proxmark client and only then it connected. Then bluetoothctl and the KDE/Plasma icon showed the device is connected once the client was run. Maybe compare the captured packets from bluetooth I linked above to what your bluetooth communication looks like.

(But it's possible that you have a fault unit.)

evazzoler, 2020-01-15:
My HW was totally replaced.
BT is working but only with a one year old firmware and "RFID tools".
Walrus is not working and PM3 Client on Laptop is not working.
Maybe the previous issue was a flat cable issue, but with the new HW it is definitely a firmware issue.

iceman1001, 2020-01-15:
- Blueshark / android  / rfid app == custom fw from 2019-08.  (Not one year old)
- Blueshark / android / termux == compile and go?
- Blueshark / android / walrus ==  suppose to be supported in a branch of walrus but nothing is going on.
- Blueshar / laptop ==  latest source from repo,  shouldn't be any issue.

evazzoler, 2020-01-16:
Thank you for the reply. I verified all and worked on deep-understanding.

> * Blueshark / android  / rfid app == custom fw from 2019-08.  (Not one year old)

Ok, info on the net about this were no clear. I git checked out a version of 08-2019 without success with BT then I figured out that the requested version was a custom fw of 2019-08, not an old version of the official RRD firmware. Honestly I don't understand the reason why a client for a serial connection needs custom code "server" side. That way we are forced to use an obsolete fw... Not so good.

> * Blueshark / android / termux == compile and go?

Not exactly. Termux has no support to BT, so you'll never be able to connect the blue shark (I hope they will develop it in the API as they did for accessing all the other hw components of the smartphone!).
The only way is to use a OTG cable with the [dedicated USB Bluetooth adapter](https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/bt_manual_v10.md#2-fast-connection-using-dedicated-usb-bluetooth-adapter-under-windows), but I really can't find where to buy it and I think is very dispersive to use an external BT adapter for connecting a BT device to a BT enabled smartphone. You can use another HC-06 with usb-to-serial adapter and a OTG cable, but you have to program it the right way and with all the wires it becomes cumbersome and not usable on the field because you appear suspicious.
Anyway for connecting the ttyACM port (bt is transparent in that configuration) from termux you need a rooted device and this enhances the difficulty (not all smartphones are rootable).

> * Blueshark / android / walrus ==  suppose to be supported in a branch of walrus but nothing is going on.

The walrus published apk sees the device via OTG (in truth I connected it but I was not able to make it working), not via BT. Honestly I have'nt set up a complete android sdk and building environment for testing this. It could take days. Should be a good gift to have an apk of the experimental branch on the github repo, but the project seems to be abandouned.

> * Blueshar / laptop ==  latest source from repo,  shouldn't be any issue.

Correct. But it has some issues on linux. It don't provides the minimal informations needed to linux (current distros) for creating a serial binding automatically. With some commands it is possible to connect but it is not practical. The way is:

`1. pair normally using code 1234, then the device will disconnect an remain disconnected`

`2. open a terminal an launch "hcitool scan" a lot of times (why???) until the BT address will be found. Sometimes you will read "PM3_RDV4.0", sometimes the PM3 will not give this parameter and you'll read "n/a" (why???)`

`3. use the rfcomm for binding the device with a virtual comm with "rfcomm bind 0 11:22:33:44:55:66 1"`

`4. connect the pm3 with "proxmark3 /dev/rfcomm0"`

If I use `sdptool browse 11:22:33:44:55:66`, I get nothing instead a lot of informations of the device.
This means that blue shark is not compliant with bluetooth protocol. It has a piece of the stack, the communication made by HC-06, but the identification (type of the device, descriptors, etc.) is still missing and the minimal requirements are this way not reached.
So let me tell that blue shark is not exactly the device that a person intend when he watch the video or read the tech specs... I hope that the develop will not stop here or other people will be crazy driven by missing things thinking something is wrong or broken.

closed 2020-01-16 by evazzoler

doegox, 2020-01-16:
Blueshark is "just" an off-the-shelve HC-06 providing an UART bridge to the Proxmark, identification details are not under control unless HC-06 is replaced by something better...

iceman1001, 2020-01-16:
Regarding custom fw,  yeah,  we have nothing to do with the development of the RFID app nor their design decision,  so you be better off venting frustration in the right repo instead of here.
Termux, same thing,  
Walrus, same thing,

The only thing left is your experience with *nix and how to have it connect / pair with a BT.
Not really this repo's purpose either, since its like @doegox wrote,  a UART bridge. 

If you think the current documentation regarding Blueshark needs a revision,  feel free to do so.
The most relevant docs will be these two.
- https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/bt_manual_v10.md
- https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/uart_notes.md

evazzoler, 2020-01-16:
Ok, thank you.
I'm not venting frustration but pointing to who will read in the future why what is expected to work doesn't work, just in case he suspects is the hw.
I don't have write permissions on the doc, so I cannot edit it. It is substantially correct except the line

`sudo rfcomm bind rfcomm0 aa:bb:cc:dd:ee:ff`
that should be
`sudo rfcomm bind 0 aa:bb:cc:dd:ee:ff 1` ("1" is the channel and cannot be omitted)

@doegox from what I remember, it is the µC that must provide identification at higher layer of the protocol. But maybe I remember badly.

iceman1001, 2020-01-16:
You would fork, edit, commit and make a PR with the changes.   They are not wiki-pages.

doegox, 2020-01-16:
The only control over HC-06 we have is what its AT command set offers, which is very little, cf http://www.martyncurrey.com/hc-06-hc01-comv2-0/

About rfcomm, maybe it depends on the version, I never had to use this "1".
From my man page:

       bind <dev> [bdaddr] [channel]
              This binds the RFCOMM device to a remote Bluetooth  device.  The
              command does not establish a connection to the remote device, it
              only creates the binding. The  connection  will  be  established
              right  after  an application tries to open the RFCOMM device. 
              **If no channel number is specified, it uses the channel number 1.**