RfidResearchGroup/proxmark3 #489: hf mf chk -- block and sector count, interface issues

mashrum, 2019-12-09:
Example card:

```
|---|----------------|---|----------------|---|          
|sec|key A           |res|key B           |res|          
|---|----------------|---|----------------|---|          
|000|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|001|  ------------  | 0 |  ------------  | 0 |          
|002|  ------------  | 0 |  ------------  | 0 |          
|003|  ------------  | 0 |  ------------  | 0 |          
|004|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|005|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|006|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|007|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|008|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|009|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|010|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|011|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|012|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|013|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|014|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|015|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|---|----------------|---|----------------|---|   
``` 

Let's check
sec 0, block 0, key B -- looks Ok

```
[usb] pm3 --> hf mf chk 0 B ffffffffffff
[ 0] key FF FF FF FF FF FF           
.          
[+] Time in checkkeys: 0 seconds
          
|---|----------------|---|----------------|---|          
|sec|key A           |res|key B           |res|          
|---|----------------|---|----------------|---|          
|000|  ------------  | 0 |  ffffffffffff  | 1 |          
|---|----------------|---|----------------|---|  
```   

sec 0, block 0, key A
- why Reading block 3 ?
- why testing to read key B... ?
```
[usb] pm3 --> hf mf chk 0 A ffffffffffff
[ 0] key FF FF FF FF FF FF           
.          
[+] Time in checkkeys: 0 seconds
          
[=] testing to read key B...          
Reading block 3          
Data:FF FF FF FF FF FF           
|---|----------------|---|----------------|---|          
|sec|key A           |res|key B           |res|          
|---|----------------|---|----------------|---|          
|000|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|---|----------------|---|----------------|---|
```

sec 10, block 40, key A
- why Reading block 3 ?
- why testing to read key B... ?
- why sec 000 ?

```
[usb] pm3 --> hf mf chk 40 A ffffffffffff
[ 0] key FF FF FF FF FF FF           
.          
[+] Time in checkkeys: 0 seconds
          
[=] testing to read key B...          
Reading block 3          
Data:FF FF FF FF FF FF           
|---|----------------|---|----------------|---|          
|sec|key A           |res|key B           |res|          
|---|----------------|---|----------------|---|          
|000|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|---|----------------|---|----------------|---|  
```

if chk B, not A
no Reading block 3 will appear :)

Better to check, and fix this interface bug

iceman1001, 2019-12-25:
That was a sneaky logic bug.  Lets see if I can sort it out today.
You only see it when testing individual blocks instead of whole tag.

iceman1001, 2019-12-28:
Try latest code.  It should have fixed that particular bug...

closed 2019-12-28 by iceman1001