RfidResearchGroup/proxmark3 #490: wrong dictionary path for autopwn

mashrum, 2019-12-09:
[usb] pm3 --> hf mf fchk 1 mfc_default_keys.dic
[+] Loaded 865 keys from mfc_default_keys.dic
[+] Running strategy 1
....

---- but ----

[usb] pm3 --> hf mf autopwn f mfc_default_keys.dic
[!] no known key was supplied, key recovery might fail
[+] loaded 0 keys from dictionary file /usr/local/Cellar/proxmark3/HEAD-5773919/bin/../share/proxmark3/dictionaries/mfc_default_keys.dic
[-] An error occurred while loading the dictionary! (we will use the default keys now)
[+] loaded 23 keys from hardcoded default array
[=] running strategy 1

MacOS, Brew

iceman1001, 2019-12-11:
@doegox ,  this is one for you faviourites :)

msoose, 2020-03-01:
This exact scenario works for me on latest master without issues, can you update and check, and if still fails provide more information about your setup? Could simply be that a working directory needs to be set properly.

```
hf mf fchk 1 mfc_default_keys.dic
[+] Loaded 906 keys from mfc_default_keys.dic           
[+] Running strategy 1 
```
```
[usb] pm3 --> hf mf autopwn f mfc_default_keys.dic
[!] no known key was supplied, key recovery might fail          
[+] loaded 906 keys from dictionary file /home/user/tools/proxmark3/client/dictionaries/mfc_default_keys.dic           
[=] running strategy 1 
```

iceman1001, 2020-03-01:
Looks like OSX / Homebrew was used when it went wrong for @mashrum 
```
[+] loaded 0 keys from dictionary file /usr/local/Cellar/proxmark3/HEAD-5773919/bin/../share/proxmark3/dictionaries/mfc_default_keys.dic
```

And @msoose  you seem to use *nix / Proxspace where  it works.

mashrum, 2020-03-02:
will check and update today

doegox, 2020-05-20:
@mashrum what's the status ?

iceman1001, 2020-06-02:
closing because of lack of response from OP

closed 2020-06-02 by iceman1001