RfidResearchGroup/proxmark3 #491: hf search line-end interface issue

mashrum, 2019-12-10:
small interface issue
command: hf search
line end symbol missed before [+]  UID : 

```
[usb] pm3 --> hf search
[=] Checking for known tags...
          
[-] Searching for ISO14443-A tag...[+]  UID : 79 85 DF 4D           
[+] ATQA : 00 04          
[+]  SAK : 08 [2]          
[+] TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1 | 1k Ev1          
[=] proprietary non iso14443-4 card found, RATS not supported          
[+] Prng detection: WEAK     

[+] Valid ISO14443-A tag  found
```      
the same time, there is no such bug on
command:  hf 14a info

```
[usb] pm3 --> hf 14a info
[+]  UID : 79 85 DF 4D           
[+] ATQA : 00 04          
[+]  SAK : 08 [2]          
[+] TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1 | 1k Ev1          
[=] proprietary non iso14443-4 card found, RATS not supported          
[+] Prng detection: WEAK
```

Interface bug detected on
```
[ CLIENT ]          
  client: RRG/Iceman          
  compiled with Clang/LLVM 4.2.1 Compatible Apple LLVM 11.0.0 (clang-1100.0.33.12) OS:OSX ARCH:x86_64          

 [ PROXMARK3 ]          
          
 [ ARM ]
  bootrom: RRG/Iceman/master/b748a79 2019-12-10 12:32:27
       os: RRG/Iceman/master/b748a79 2019-12-10 12:32:37
  compiled with GCC 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
```

iceman1001, 2019-12-10:
Yup.  I know.

iceman1001, 2019-12-10:
https://github.com/RfidResearchGroup/proxmark3/commit/3dd380e9850219b4485d2c0e988edcdd84e0f688

closed 2019-12-10 by iceman1001