RfidResearchGroup/proxmark3 #492: RDV4 OSX flash with --with-bluetooth not working

anson338, 2019-12-15:
Hi all,
I tried to flash the pm3 with blueshark add-on but pm3 shows absent on load.
Might take a look for my situation? 

─   anson@AnsondeMacBook-Pro   ~                              16:50 
╰─ brew install --HEAD  --with-blueshark proxmark3
Updating Homebrew...
==> Installing proxmark3 from rfidresearchgroup/proxmark3
==> Cloning https://github.com/RfidResearchGroup/proxmark3.git
Updating /Users/anson/Library/Caches/Homebrew/proxmark3--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at 8e45e0e5 textual
==> make clean
==> make all PLATFORM=PM3RDV4 PLATFORM_EXTRAS=BTADDON
==> make install PREFIX=/usr/local/Cellar/proxmark3/HEAD-8e45e0e
==> Install success!
==> The latest bootloader and firmware binaries are ready and waiting in the cur
🍺  /usr/local/Cellar/proxmark3/HEAD-8e45e0e: 563 files, 15.9MB, built in 1 minute 41 seconds

╭─   anson@AnsondeMacBook-Pro   ~                    02:07   16:53 
╰─ pm3-flash-all
[=] Waiting for Proxmark3 to appear...
[=] Session log /Users/anson/.proxmark3/log_20191215.txt
[+] About to use the following files:
[+]     /usr/local/Cellar/proxmark3/HEAD-8e45e0e/bin/../share/proxmark3/firmware/bootrom.elf
[+]     /usr/local/Cellar/proxmark3/HEAD-8e45e0e/bin/../share/proxmark3/firmware/fullimage.elf
[+] Waiting for Proxmark3 to appear on /dev/tty.usbmodemiceman1
.Found
[+] Entering bootloader...
[+] (Press and release the button only to abort )
[+] Waiting for Proxmark3 to appear on /dev/tty.usbmodemiceman1
.. Found
[=] Available memory on this board: 512K bytes

[=] Permitted flash range: 0x00100000-0x00180000
[+] Loading ELF file /usr/local/Cellar/proxmark3/HEAD-8e45e0e/bin/../share/proxmark3/firmware/bootrom.elf
[+] Loading usable ELF segments:
[+]    0 : V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
[+]    1 : V 0x00200000 P 0x00100200 (0x00000e08->0x00000e08) [R X] @0x298

[+] Loading ELF file /usr/local/Cellar/proxmark3/HEAD-8e45e0e/bin/../share/proxmark3/firmware/fullimage.elf
[+] Loading usable ELF segments:
[+]    0 : V 0x00102000 P 0x00102000 (0x000414b8->0x000414b8) [R X] @0x94
[+]    1 : V 0x00200000 P 0x001434b8 (0x00001528->0x00001528) [RW ] @0x4154c
[=] Note: Extending previous segment from 0x414b8 to 0x429e0 bytes

[+] Flashing...

[+] Writing segments for file: /usr/local/Cellar/proxmark3/HEAD-8e45e0e/bin/../share/proxmark3/firmware/bootrom.elf
[+]  0x00100000..0x001001ff [0x200 / 1 blocks]
. OK
[+]  0x00100200..0x00101007 [0xe08 / 8 blocks]
........ OK

[+] Writing segments for file: /usr/local/Cellar/proxmark3/HEAD-8e45e0e/bin/../share/proxmark3/firmware/fullimage.elf
[+]  0x00102000..0x001449df [0x429e0 / 533 blocks]
..................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................... OK

[+] All done.

Have a nice day!

╭─   anson@AnsondeMacBook-Pro   ~                    19.17   16:53 
╰─ pm3

[=] Waiting for Proxmark3 to appear...

  ██████╗ ███╗   ███╗ ████╗
  ██╔══██╗████╗ ████║   ══█║
  ██████╔╝██╔████╔██║ ████╔╝
  ██╔═══╝ ██║╚██╔╝██║   ══█║     iceman@icesql.net
  ██║     ██║ ╚═╝ ██║ ████╔╝    https://github.com/rfidresearchgroup/proxmark3/
  ╚═╝     ╚═╝     ╚═╝ ╚═══╝  pre-release v4.0

[=] Session log /Users/anson/.proxmark3/log_20191215.txt
[=] Using UART port /dev/tty.usbmodemiceman1
[=] Communicating with PM3 over USB-CDC

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman
  compiled with Clang/LLVM 4.2.1 Compatible Apple LLVM 11.0.0 (clang-1100.0.33.16) OS:OSX ARCH:x86_64

 [ PROXMARK3 RDV4 ]
  external flash:                  present
  smartcard reader:                present

 [ PROXMARK3 RDV4 Extras ]
  FPC USART for BT add-on support: absent

 [ ARM ]
  bootrom: RRG/Iceman/master/8e45e0e5 2019-12-15 16:52:49
       os: RRG/Iceman/master/8e45e0e5 2019-12-15 16:53:02
  compiled with GCC 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]

 [ FPGA ]
  LF image built for 2s30vq100 on 2019-07-31 at 15:57:16
  HF image built for 2s30vq100 on 2018-09-03 at 21:40:23

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 281055 bytes (54%) Free: 243233 bytes (46%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

[usb] pm3 -->
[usb] pm3 -->

iceman1001, 2019-12-15:
This is more of a homebrew issue where it seems not to respect the `make all PLATFORM=PM3RDV4 PLATFORM_EXTRAS=BTADDON`

see [#326](https://github.com/RfidResearchGroup/proxmark3/issues/326), [#294](https://github.com/RfidResearchGroup/proxmark3/issues/294)

Some ppl seem to have solved it with git clone and manually compile the project with blueshark enabled.

anson338, 2019-12-19:
I tried to build with PLATFORM=PM3OTHER, it's ok.
However it leads

>  [ CLIENT ]
>   client: RRG/Iceman
>   compiled with Clang/LLVM 4.2.1 Compatible Apple LLVM 11.0.0 (clang-1100.0.33.16) OS:OSX ARCH:x86_64
> 
>  [ PROXMARK3 RDV4 ]
>   external flash:                  absent
>   smartcard reader:                absent
> 

>  [ PROXMARK3 RDV4 Extras ]
>   FPC USART for BT add-on support: present

But if i compiled with PM3RDV4. The flasher on windows will show (other platform defined is ok)
Error : PHDR is not contained in Flash

iceman1001, 2019-12-19:
????  That would be very untested,  usart for a non-rdv4....

anson338, 2019-12-19:
nvm, was my fault. I tried to use some Chinese guys built flasher.exe and flash the BTaddon rdv4 version into the rdv4. 

It don't have any issue if no btaddon on rdv4, or another platform version with btaddon.

Just used OSX pm3-flash-image and it's ok now.

Sorry for this nonsense issue i created

closed 2019-12-19 by anson338

iceman1001, 2019-12-19:
We strongly recommend using the provided software from this repo.  Unknown sources is on your own risk.