RfidResearchGroup/proxmark3 #497: fixed nonce message

mashrum, 2019-12-20:
Add message about fixed nonce.
Related commands:
hf search
hf 14a info

[usb] pm3 --> hf sear
[=] Checking for known tags...          
[-] Searching for ISO14443-A tag...           
[+]  UID : 79 85 DF 4D           
[+] ATQA : 00 04          
[+]  SAK : 08 [2]          
[+] TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1 | 1k Ev1          
[=] proprietary non iso14443-4 card found, RATS not supported          
[+] Prng detection: WEAK           
          
[+] Valid ISO14443-A tag  found
          
[usb] pm3 --> hf 14a info
          
[+]  UID : 79 85 DF 4D           
[+] ATQA : 00 04          
[+]  SAK : 08 [2]          
[+] TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1 | 1k Ev1          
[=] proprietary non iso14443-4 card found, RATS not supported          
[+] Prng detection: WEAK           
[usb] pm3 -->

iceman1001, 2019-12-23:
Try latest,  I have added static/fixed nonce detection to **hf 14a info** command

mashrum, 2019-12-23:
Still not work for me :(
```
  bootrom: RRG/Iceman/master/07b3b6ed 2019-12-23 16:36:56
       os: RRG/Iceman/master/07b3b6ed 2019-12-23 16:37:06
  compiled with GCC 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
```

```
[+]  UID : 79 85 DF 4D           
[+] ATQA : 00 04          
[+]  SAK : 08 [2]          
[+] TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1 | 1k Ev1          
[=] proprietary non iso14443-4 card found, RATS not supported          
[+] Prng detection: WEAK           
[usb] pm3 --> 
```

but nonces file 
shows only
00000000:  01 20 01 45 01 20 01 45 01 20 01 45 01 20 01 45  :. .E. .E. .E. .E

[usb] pm3 --> hf mf rdbl 24 A FFFFFFFFFFFF
[usb] pm3 --> trace list

```
      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation          
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------          
          0 |        992 | Rdr |52                                                                       |     | WUPA          
       2244 |       4612 | Tag |04  00                                                                   |     |           
       7040 |       9504 | Rdr |93  20                                                                   |     | ANTICOLL          
      10692 |      16580 | Tag |79  85  df  4d  6e                                                       |     |           
      19072 |      29600 | Rdr |93  70  79  85  df  4d  6e  9c  67                                       |  ok | SELECT_UID          
      30788 |      34308 | Tag |08  b6  dd                                                               |     |           
      36352 |      41056 | Rdr |60  18  3c  e7                                                           |  ok | AUTH-A(24)          
      43076 |      47812 | Tag |01  20  01  45                                                           |     |           
      56832 |      66144 | Rdr |21  52  15  29! e7! fc! 50  8c!                                          | !crc|           
      67396 |      72068 | Tag |70  6a! 52  44                                                           |     |           
      77824 |      82592 | Rdr |0a  3e! 37! 57                                                           | !crc|           
      83780 |     104580 | Tag |7a  71  00  48  82! c7! 2a! 88! 37! aa! 17  dc! ee! 87  b0  07  fc! cf   | !crc|           
     117248 |     121952 | Rdr |3d  5d  58! 48     
```                                                      | !crc|

mashrum, 2019-12-23:
Also on
```
[ ARM ]
  bootrom: RRG/Iceman/master/a6a48f0e 2019-12-23 18:46:49
       os: RRG/Iceman/master/a6a48f0e 2019-12-23 18:47:00
  compiled with GCC 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
```

Additional new bug only for fixed nonce (for normal cards there is no bug, all works good)
if to run
```
[usb] pm3 --> hf mf nested 1 0 A FFFFFFFFFFFF d
[+] Testing known keys. Sector count=16          
.          
[+] Chunk: 2.7s | found 26/32 keys (24)
          
[+] Time to check 23 known keys: 3 seconds
          
[+] enter nested attack          
[!] button pressed. Aborted.
```

PM3 Easy
LEDs
blue - ON
orange - ON
red - OFF
Green - ON

Device still response for commands

Hardnested - runs, but with no result

iceman1001, 2019-12-23:
try again.

iceman1001, 2019-12-23:
original issue solve.   make new issue for nested,  or feel free to solve it.

closed 2019-12-23 by iceman1001