RfidResearchGroup/proxmark3 #498: Blue Shark no connect

joelsernamoreno, 2019-12-21:
**Describe the bug**
I have blueshark, but when I connect my PC via bluetooth and enter the pin (1234), blueshark is connected but disconnected

**To Reproduce**
1. Turn on the blueshark battery and bluetooth modules
2. Launch bluetooth on your laptop
3. Search for Blueshark in the bluetooth menu of your laptop
4. Connect to blueshark and enter the pin (1234)

Blueshark is connected, but after 2 seconds it disconnects

**Expected behavior**
Hello!

I have blueshark, but when I connect my PC via bluetooth and enter the pin (1234), blueshark is connected but disconnected

I tried Android and the RFID tool application, it works!

How do I properly connect my laptop with blueshark? (I'm using my laptop's internal bluetooth module) do I need a more powerful bluetooth adapter?

**Desktop (please complete the following information):**
 - OS: Kali Linux or Ubuntu 18.04
 - Proxmark installation is correct, Bluetooth addon: present

joelsernamoreno, 2019-12-21:
I solved it.

You can remove this issue

Thank you :D

iceman1001, 2019-12-21:
what was your solution?

closed 2019-12-21 by iceman1001

joelsernamoreno, 2019-12-21:
Connect via terminal. Bluetooth GUI does not work properly with Blueshark

KamicDemon, 2020-02-02:
I think I'm having a similar issue, but it occurs when I try to launch the client the blue LED turns solid momentarily but then times apparently times out and I get:

[!!] ERROR: cannot communicate with the Proxmark

Is it related somehow?
Is this an appropriate place to ask this question?

iceman1001, 2020-02-04:
it comes down to how you are using your Blueshark.
If you are using the rfid app,  then it uses an older custom firmware which you need on your Proxmark3.
If you are using laptop bt to connect via pm3 client,  you will only get solid leds when you connect the client.  Not duing paring etc.

Needless to say, you will need to provide detailed information for any chance of real debugging / understanding your problem

joelsernamoreno, 2020-02-04:
Iceman's right.

I had to go back to an older version of Proxmark3 to use the RFID (android) application. I updated my proxmark3 3 days ago and have not been able to use the application.

To use blue shark with the laptop BT, I did this:

https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/bt_manual_v10.md

Tested with the latest version of Proxmark3 on Kali Linux and Ubuntu 18.04