RfidResearchGroup/proxmark3 #499: Using "lf config" without parameters breaks LF data acquistion (possibe data corruption inside ARM OS)

hiviah, 2019-12-22:
**Describe the bug**
Using `lf config` without parameters and then read a tag with e.g. `lf em 410x_read` makes the data acquistion timeout or have all-zero data buffer (as can be seen by `data plot`).

Until the `lf config` is used, I can request the tag read as many times as I want. Often `lf config` does not write out the current config as it should.

It seems that using `lf config` without parameters corrupts the configuration itself, as can be seen under `hw status` ouput below

I am using current master (66426cee1ec60698348f46f9c8dcb22a6ab4a51c) on 1st gen proxmark, but I observed same behavior with older revisions on RDV4.

**To Reproduce**
Steps to reproduce the behavior:
1. Issue `lf config`
2. Read a tag, e.g. `lf em 410x_read`
3. Goto 1. Repeat several times. I can usually get it to trigger the bug under 10 attempts.
4. Eventually the tag read will start to timeout and data displayed with `data plot` will be all zeros

**Expected behavior**
Issuing `lf config` to read LF configuration should not break data acquistion and thus reading LF tags.

**GDB trace**
I tried to find out where the ARM CPU is stuck after client says timeout, seems the CPU is still long stuck in `DoAcquistion`:

    ─── Stack ───────────────────────────────────────────────────────────
    [0] from 0x00105e08 in DoAcquisition+368 at lfsampling.c:158
    [1] from 0x00105f0c in DoAcquisition_config+72 at lfsampling.c:257
    [2] from 0x00105f6c in ReadLF+52 at lfsampling.c:275
    [3] from 0x00105f96 in SampleLF+22 at lfsampling.c:286
    [4] from 0x00110ff4 in PacketReceived+2380 at appmain.c:737
    [5] from 0x00110ff4 in AppMain+2588 at appmain.c:2064
    [6] from 0x0010206a in Vector+106 at start.c:80
    ─── Source ──────────────────────────────────────────────────────────
     153      uint32_t cancel_counter = 0;
     154  
     155      uint16_t checked = 0;
     156  
     157      while (true) {
     158          if (checked == 1000) {                           <-----
     159              if (BUTTON_PRESS() || data_available())
     160                  break;
     161              else
     162                  checked = 0;

**Desktop (please complete the following information):**
 - OS: Linux (Ubuntu 16.04 LTS)
 - hw version

     [ CLIENT ]           
      client: RRG/Iceman          
      compiled with GCC 5.4.0 20160609 OS:Linux ARCH:x86_64          

     [ PROXMARK3 ]           
              
     [ ARM ]
      bootrom: RRG/Iceman/master/ed0bbe4 2019-11-09 22:08:49
           os: RRG/Iceman/master/66426ce-dirty-unclean 2019-12-22 17:05:54
      compiled with GCC 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

     [ FPGA ]
      LF image built for 2s30vq100 on 2019-07-31 at 15:57:16
      HF image built for 2s30vq100 on 2018-09-03 at 21:40:23          

     [ Hardware ]           
      --= uC: AT91SAM7S256 Rev A          
      --= Embedded Processor: ARM7TDMI          
      --= Nonvolatile Program Memory Size: 256K bytes, Used: 241822 bytes (92%) Free: 20322 bytes ( 8%)          
      --= Second Nonvolatile Program Memory Size: None          
      --= Internal SRAM Size: 256K bytes          
      --= Architecture Identifier: AT91SAM7Sxx Series          
      --= Nonvolatile Program Memory Type: Embedded Flash Memory          

 - hw status - the values seem wrong (threshold and divisor for example)

    #db# Memory           
    #db#   BIGBUF_SIZE.............40000          
    #db#   Available memory........40000          
    #db# Tracing           
    #db#   tracing ................1          
    #db#   traceLen ...............0          
    #db# Currently loaded FPGA image           
    #db#   mode.................... HF image built for 2s30vq100 on 2018-09-03 at 21:40:23          
    #db# LF Sampling config           
    #db#   [q] divisor.............861995008 ( 0.00 kHz )          
    #db#   [b] bps.................8          
    #db#   [d] decimation..........1          
    #db#   [a] averaging...........Yes          
    #db#   [t] trigger threshold...3145728          
    #db#   [s] samples to skip.....0           
    #db# LF T55XX config           
    #db#            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]          
    #db#            mode            |start|write|write|write| read|write|write          
    #db#                            | gap | gap |  0  |  1  | gap |  2  |  3          
    #db# ---------------------------+-----+-----+-----+-----+-----+-----+------          
    #db# fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | N/A | N/A |           
    #db#     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |           
    #db#               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |           
    #db#    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 |           
    #db#           
    #db# Transfer Speed           
    #db#   Sending packets to client...          
    #db#   Time elapsed............500ms          
    #db#   Bytes transferred.......325120          
    #db#   Transfer Speed PM3 -> Client = 650240 bytes/s          
    #db# Various           
    #db#   DBGLEVEL................1          
    #db#   ToSendMax...............-1          
    #db#   ToSendBit...............0          
    #db#   ToSend BUFFERSIZE.......2308          
    #db#   Slow clock..............31015 Hz          
    #db# Installed StandAlone Mode           
    #db#   LF HID26 standalone - aka SamyRun (Samy Kamkar)          

 - data tune

    [+] LF antenna: 12.32 V - 125.00 kHz          
    [+] LF antenna: 19.40 V - 134.83 kHz          
    [+] LF optimal: 20.25 V - 133.33 kHz          
    [+] LF antenna is OK

iceman1001, 2019-12-22:
I think the `lf config`  call without params should print the current config, not change anything.
Definitly a bug.

iceman1001, 2020-01-09:
Fix:  https://github.com/RfidResearchGroup/proxmark3/commit/0869cbc6ad4a6c493b8fe934c581f1abb88fcd0a

closed 2020-01-09 by iceman1001