RfidResearchGroup/proxmark3 #503: hf mf nested on static nonce

mashrum, 2019-12-24:
related with
```
[ ARM ]
  bootrom: RRG/Iceman/master/aeb3f74c 2019-12-24 11:36:02
       os: RRG/Iceman/master/aeb3f74c 2019-12-24 11:36:12
  compiled with GCC 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]

```
only for fixed nonce (for normal cards there is no bug, all works good)
if to run

```
[usb] pm3 --> hf mf nested 1 0 A FFFFFFFFFFFF d
[+] Testing known keys. Sector count=16          
.          
[+] Chunk: 2.7s | found 26/32 keys (24)
          
[+] Time to check 23 known keys: 3 seconds
          
[+] enter nested attack          
[!] button pressed. Aborted.
```

Hardware: PM3 Easy
LEDs
blue - ON
orange - ON
red - OFF
Green - ON

Device still accept and response for commands.

If the exit called by static/fixed nonce 
- better to have check it ASAP, before Testing known keys
- Provide exit text message "Static/fixed nonce detected. Nested not applicable."

iceman1001, 2019-12-24:
As you noticed that nested doesn't work with no original tags.
To my opinion, nested shoulded,  Nor should darkside.
There is an idea of solution going on in the offical repo,   https://github.com/Proxmark/proxmark3/pull/900

That kind of  solution will not be implemented in this repo.  The prefered solution is a new dedicated command.

iceman1001, 2019-12-25:
With latest changes, nested command checks for static nonce and quits if found.

closed 2019-12-25 by iceman1001

mashrum, 2019-12-26:
Just let me confirm
that fixed nonce cards really readable with nested attack released Proxmark/proxmark3
It works.

Will wait this feature release in RfidResearchGroup/proxmark3 :)
and autopwn command

big thanks for your help and your work

iceman1001, 2019-12-26:
Feel free to contribute with adding the check to autopwn.

APOFISAN, 2023-03-26:
So far so good, the case is that I got my hands on a new card from this company and what was my surprise that when I try to read it with the proxmark3 and does not let me read sectors 1 and 2. This company has changed the passwords of sectors 0,3,4,5,6,7,8,9,9,10,11,12,13,14,15 and has put default keys FFFFFFFFFFFFFFFFFFFF. In the first one you only knew the password A of sector 0, which was A0A1A2A3A4A5.

if I use hf mf keycheck, it comes out empty, it does not find any key.
if I use hf mf fchk, I get all keys except for sector 1 and 2.
if I use hf mf autopwn, it only gets the FFFFFFFFFFFFFFFFFFFFF and at the end it says: nested: 00000000 vs 00000000. error: no response from proxmark3.

if use hf mf darkside pone runing darkside…- card is not vulnerable to darkside attack, doesn’t send NACK on authentication request.

Another change that I have seen and I had not noticed is that the header 0 of sector 0, has also changed, that is to say, this the uid and other numbers, that in the old cards except for the uid, were all the same. In this new change in each card are not the same.

[usb] pm3 → hf mf chk
[=] Start check for keys…
[=] …
[=] time in checkkeys 3 seconds

[=] testing to read key B…

[+] found keys:

[+] -----±----±-------------±–±-------------±—
[+] Sec | Blk | key A |res| key B |res
[+] -----±----±-------------±–±-------------±—
[+] 000 | 003 | FFFFFFFFFFFF | 1 | FFFFFFFFFFFF | 1
[+] 001 | 007 | ------------ | 0 | ------------ | 0
[+] 002 | 011 | ------------ | 0 | ------------ | 0
[+] 003 | 015 | FFFFFFFFFFFF | 1 | FFFFFFFFFFFF | 1
[+] 004 | 019 | FFFFFFFFFFFF | 1 | FFFFFFFFFFFF | 1
[+] 005 | 023 | FFFFFFFFFFFF | 1 | FFFFFFFFFFFF | 1
[+] 006 | 027 | FFFFFFFFFFFF | 1 | FFFFFFFFFFFF | 1
[+] 007 | 031 | FFFFFFFFFFFF | 1 | FFFFFFFFFFFF | 1
[+] 008 | 035 | FFFFFFFFFFFF | 1 | FFFFFFFFFFFF | 1
[+] 009 | 039 | FFFFFFFFFFFF | 1 | FFFFFFFFFFFF | 1
[+] 010 | 043 | FFFFFFFFFFFF | 1 | FFFFFFFFFFFF | 1
[+] 011 | 047 | FFFFFFFFFFFF | 1 | FFFFFFFFFFFF | 1
[+] 012 | 051 | FFFFFFFFFFFF | 1 | FFFFFFFFFFFF | 1
[+] 013 | 055 | FFFFFFFFFFFF | 1 | FFFFFFFFFFFF | 1
[+] 014 | 059 | FFFFFFFFFFFF | 1 | FFFFFFFFFFFF | 1
[+] 015 | 063 | FFFFFFFFFFFF | 1 | FFFFFFFFFFFF | 1
[+] -----±----±-------------±–±-------------±—
[+] ( 0:Failed / 1:Success )

[usb] pm3 → hf mf autopwn
[!] no known key was supplied, key recovery might fail
[+] loaded 45 keys from hardcoded default array
[=] running strategy 1
[=] Chunk 1.2s | found 28/32 keys (45)
[=] running strategy 2
[=] Chunk 1.2s | found 28/32 keys (45)
[+] target sector 0 key type A – found valid key [ FFFFFFFFFFFF ] (used for nested / hardnested attack)
[+] target sector 0 key type B – found valid key [ FFFFFFFFFFFF ]
[+] target sector 3 key type A – found valid key [ FFFFFFFFFFFF ]
[+] target sector 3 key type B – found valid key [ FFFFFFFFFFFF ]
[+] target sector 4 key type A – found valid key [ FFFFFFFFFFFF ]
[+] target sector 4 key type B – found valid key [ FFFFFFFFFFFF ]
[+] target sector 5 key type A – found valid key [ FFFFFFFFFFFF ]
[+] target sector 5 key type B – found valid key [ FFFFFFFFFFFF ]
[+] target sector 6 key type A – found valid key [ FFFFFFFFFFFF ]
[+] target sector 6 key type B – found valid key [ FFFFFFFFFFFF ]
[+] target sector 7 key type A – found valid key [ FFFFFFFFFFFF ]
[+] target sector 7 key type B – found valid key [ FFFFFFFFFFFF ]
[+] target sector 8 key type A – found valid key [ FFFFFFFFFFFF ]
[+] target sector 8 key type B – found valid key [ FFFFFFFFFFFF ]
[+] target sector 9 key type A – found valid key [ FFFFFFFFFFFF ]
[+] target sector 9 key type B – found valid key [ FFFFFFFFFFFF ]
[+] target sector 10 key type A – found valid key [ FFFFFFFFFFFF ]
[+] target sector 10 key type B – found valid key [ FFFFFFFFFFFF ]
[+] target sector 11 key type A – found valid key [ FFFFFFFFFFFF ]
[+] target sector 11 key type B – found valid key [ FFFFFFFFFFFF ]
[+] target sector 12 key type A – found valid key [ FFFFFFFFFFFF ]
[+] target sector 12 key type B – found valid key [ FFFFFFFFFFFF ]
[... 76 lines cut ...]
006 ------------ 0 ------------ 0
007 ------------ 0 ------------ 0
008 ------------ 0 ------------ 0
009 ------------ 0 ------------ 0
010 ------------ 0 ------------ 0
011 ------------ 0 ------------ 0
012 ------------ 0 ------------ 0
013 ------------ 0 ------------ 0
014 ------------ 0 ------------ 0
015 ------------ 0 ------------ 0
— ---------------- — ---------------- —
Do you wish to save the keys to dumpfile? [y/n] ?
[usb] pm3 → hf mf nested --1k --blk 0 -a -k FFFFFFFFFFFF
[+] Testing known keys. Sector count 16
[=] Chunk 1.3s | found 28/32 keys (46)
[+] Time to check 45 known keys: 1 seconds

[+] enter nested key recovery
[#] Nested: 00000000 vs 00000000
[!!] Command execute timeout

[usb] pm3 --> hf mf nested --1k --blk 0 -a -k ffffffffffff --tblk 8 --ta
[-] Tag isn't vulnerable to Nested Attack (PRNG is not predictable).

I think it is a static encrypted nonces. Could it be?

iceman1001, 2023-03-26:
Yes,  it is a static encrypted nonce.   
You can only sniff to get keys from those so far.   Well,  if its not a magic tag where you can read the memory regardless.

APOFISAN, 2023-03-26:
and how is it done?

Hmvgit, 2023-03-28:
@iceman1001 what about this implementation? https://github.com/RfidResearchGroup/proxmark3/commit/b37a4c14eb497b431f7443b9f685d7f2e222bfa0
was this not supposed to be a fix for the same issue?

iceman1001, 2023-03-28:
Don't be confused with  `static nested`  which solves static nonces,   vs  static encrypted nonce

APOFISAN, 2023-03-29:
And what do you think is the best way to sniff the card and thus try to get the key. A mini crash course could be? please. I would like to learn

APOFISAN, 2023-03-29:
I put my card over the Proxmark and put hf 14a sniff.

I understand that with this command is put to sniff and then use an android mobile and with the application raid nfc tool, I put the sector and the key and sent his reading in sector 0 because I know his key and in sector 2 that I do not know his key and gave me wrong key. But then I do not know what to do with that data you get and I do not know if they are worth.

Translated with www.DeepL.com/Translator (free version)